Koenkk/zigbee2mqtt #3895: Bulbs from China

maxpd1, 2020-07-12:
Hi,

I want to buy a bunch (at least 20) of RGBW bulbs from China with Zigbee compatibility. Some sellers say these bulbs are compatible with HUE or Tuya Gateways.

How can I be sure that the bulb will work with my iobroker Zigbee Adapter?

This description looks a little bit as it is only for developers? https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

Regards
maxpd

Koenkk, 2020-07-12:
If these devices are not on https://www.zigbee2mqtt.io/information/supported_devices.html they are not supported currently.

However if they are compatible with Hue or TuYa, adding Zigbee2mqtt support shouldn't be a problem (no guarantees). If you post the Zigbee2mqtt log after pairing them we can add them.

Wuppi68, 2020-07-30:
Most Gledopto devices works well ...

adopting my MR16 bulbs in herdsman took about 30 minutes by copying another bulb and modify the names

Buy one bulb for testing and if it works buy the rest ;-)

maxpd1, 2020-09-01:
Bulbs receives. Unfortunately they aren't pairing with my CC2530

maxpd1, 2020-09-01:
@Koenkk How do I get the mqtt log from iobroker? Do I have to acitvate zigbee debug mode?
@Wuppi68 Do you have a tutorial how you acomplished that?

Koenkk, 2020-09-01:
Can you provide a link to this device? (e.g. aliexpress?)

maxpd1, 2020-09-01:
Difficult. Got it via alibaba on basis of this

https://app.alibaba.com/dynamiclink?touchId=1600064220375&type=product&schema=enalibaba%3A%2F%2Fdetail%3Fid%3D1600064220375%26ck%3Dshare_detail&ck=share_detail&shareScene=buyer

Koenkk, 2020-09-01:
Link doesn't work for me, can you post a picture of the device?

maxpd1, 2020-09-02:
Strange, worked for me on another device. Nevertheless, here it is. Thank you so much!
[image]

maxpd1, 2020-09-02:
I jumpeg again over your tutorial what you need. 
@Koenkk: Where the infos above enough or should I try to get through your manual?

Koenkk, 2020-09-02:
Can you try with the following added to `devices.js`?

```js
{
    fingerprint: [{modelID: 'TS0505A', manufacturerName: '_TZ3000_evag0pvn'}],
    model: 'TS0505A',
    description: 'E27 RGBW bulb',
    vendor: 'TuYa',
    extend: generic.light_onoff_brightness_colortemp_colorxy,
},
```

See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to edit `devices.js`.

maxpd1, 2020-09-03:
Ok, so far:
I added these lines into devices.js

I didn't found configuration.yaml

I restarted iobroker.

I still see unknown device in instance configuration.
I don't see the device in my objects

I put the device into a group.
If I change brightness of group the bulb is getting darker.

maxpd1, 2020-09-04:
oh, one thing I did not mention. During saving devices.js I got error message that file was saved but without a new time stamp. Don't know if this is relevant.

And in your code first line is "fingerprint" whereas normally its called "zigbeeModel"

maxpd1, 2020-09-04:
PS: Am I in the right repository or do I have to speak with someone of them, as I think I do have this adapter in iobroker: https://github.com/ioBroker/ioBroker.zigbee

Koenkk, 2020-09-04:
yes, please continue there. I dont know how to add new devices for testing in iobroker.

closed 2020-09-04 by Koenkk