Koenkk/zigbee2mqtt #3897: Can't re-pair aqara presence, z2m keep sending Leave message

wassfila, 2020-07-12:
# Important
I'm aware this is a known issue as detaching the router for 10 sec resolves it, yet I figured out the root cause of the issue.

- remove device
- remove device fails (for any reason)
- For each attempt to repair the device, the coordinator keeps sending zigbee message "Remove Device".

So the issue is that it is not possible to clear the temporary memory that requests a device to be removed. A fix could be to limit removal request in time. Not sure if this should not be addressed to firmware rather, but I post it here for info.

[image]

# Bug Report
## What happened
Problem :
Can't re pair aqara presence sensor
    "0x00158d0002b45474":{
        "type":"Aqara presence",
        "reference":"RTCGQ11LM",
        "id":"lumi.sensor_motion.aq2"
    }
Under the category "Device does not pair or interview fails?"
- https://www.zigbee2mqtt.io/information/FAQ.html#why-does-my-device-not-or-fail-to-pair

# Workaround
C2531/CC2530 coordinator only:
    Stop Zigbee2mqtt, unplug the coordinator, wait 10 seconds, plug the coordinator, start Zigbee2mqtt and try to pair the device again.

## How to reproduce it (minimal and precise)
## Debug Info
Zigbee2mqtt version:
- Coordinator CC2530 
"version":"1.14.0"
"commit":"9009de2"
"coordinator":{"type":"zStack12","meta": "transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}

Koenkk, 2020-07-14:
I've tried to reproduce it but failed to.

Setup:
- CC2531 coordinator with ZStack 3 20190425  and ZStack 1.2 20190608
- Xiaomi RTCGQ11LM

Procedure:
- Pair device
- Disconnect battery 
- Send removal request to `zigbee2mqtt/bridge/config/remove`
- Wait until log shows `Failed to remove 0x00158d000204aef4 (Error: AREQ - ZDO - mgmtLeaveRsp after 10000ms)`
- Connect battery
- Device started to work again (as if nothing happened) -> expected because it didn't receive the request
- Hold pair button on RTCGQ11LM
- Device sends Leave request and almost directly after this joins the network successfully again.
- Device works as normal

Sniffs for reference:  [sniffs.zip](https://github.com/Koenkk/zigbee2mqtt/files/4920309/sniffs.zip)

Could you provide me the minimal setup and procedure to reproduce this issue?

wassfila, 2020-07-15:
I'm unfortunately not able to reproduce it the way it happened. The difference between your test sequence is that the device had its battery and was alive when I got the error "Failed to remove". So was probably not paired at the first place, otherwise I would not have tried to pair it again. I did not use that device for a very long time so it had an undefined state either paired with a coordinator that is no longer available or somehow out of sync.

I thought there were enough evidence to try to find the bug by reviewing the remove process on the firmware side, if it has any undefinitely pending state until power reset. But I understand that no need to venture modifying that without possibility to test. 

I'll try to play aruond again by holding the pair button while z2m not in pairing mode, that might bring the device in that weired state that, or pair it with another instance first, that might trigger the rest of the issue. If I manage that I would reopen this ticket, but we could close it for now.

Koenkk, 2020-07-15:
No problem! Let me know when you have been able to reproduce the problem then we can re-open the issue.

closed 2020-07-15 by Koenkk

wassfila, 2020-07-16:
Let's keep this ticket closed, I just post a comment to keep track here :

This time I reproduced it with a brand new device (ikea symfonisk) first time unpacked, put a battery and pressed four times, it got already a remove message, I could pair two devices before this happens, but for that I had to disconnect the coordinator and replug it.

[image]