Koenkk/zigbee2mqtt #3900: Danalock V3 - connection issues

otakon87, 2020-07-13:
Hi, 

I'm using a Danalock V3 Zigbee with a **CC2530+CC2592 Coordinator**, zigbee2mqtt version **1.14.1**. 
I also have several osram plug as routers. 

The problem I have is that the lock receive the command to open or lock two times every time. I really don't know how to fix that.

This is the specific errore I see on the logs when I send an open/close command.  https://pastebin.com/iDuh4nsE

And here is the log of my zigbee2mqtt instance: https://pastebin.com/VBGV5D46

This is a map of my network atm: https://ibb.co/rf4ccwR
In the higher section of the image you can see the danalock v3.

Hope you can help me since is a very annoying problem. Thank you, guys.

nickcj931, 2020-07-14:
I get the same problem as you @otakon87 
**Error when hitting Unlock**
```
zigbee2mqtt:error 2020-07-14 13:09:39: Publish 'set' 'state' to 'lock_backdoor' failed: 'Error: Command 0x000b57fffe59ce62/1 closuresDoorLock.lockDoor({"pincodevalue":""}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 43889 - 1 - 220 - 257 - 0 after 10000ms)'
zigbee2mqtt:info  2020-07-14 13:09:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'lock_backdoor' failed: 'Error: Command 0x000b57fffe59ce62/1 closuresDoorLock.lockDoor({\"pincodevalue\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 43889 - 1 - 220 - 257 - 0 after 10000ms)'","meta":{"friendly_name":"lock_backdoor"}}'
```
**Error when hitting lock**
```
zigbee2mqtt:error 2020-07-14 13:10:03: Publish 'set' 'state' to 'lock_backdoor' failed: 'Error: Command 0x000b57fffe59ce62/1 closuresDoorLock.unlockDoor({"pincodevalue":""}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Error: Timeout - 43889 - 1 - 221 - 257 - 1 after 10000ms)'
zigbee2mqtt:info  2020-07-14 13:10:03: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'lock_backdoor' failed: 'Error: Command 0x000b57fffe59ce62/1 closuresDoorLock.unlockDoor({\"pincodevalue\":\"\"}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":true,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Error: Timeout - 43889 - 1 - 221 - 257 - 1 after 10000ms)'","meta":{"friendly_name":"lock_backdoor"}}'
```

The reported state is always unlocked even if it is locked.
I am basically not getting any states reported on the lock so no battery/signal/lock states are being recieved.

However when you issue to the lock/unlock state it does as it should.

otakon87, 2020-07-14:
Does your lock connects directly to the coordinator? I think the errors might be caused by the routers, the osram plugs! I removed them from the network and everything worked as expected...

nickcj931, 2020-07-14:
Yep it does in deed. I couldnt pair it until I took the lock to the coordinator, after hours of faffing it paired when I had it within 10cm of the coordinator. 

I then moved it to ... well the door.. and after a few days it re-routed via a zigbee bulb ... not 100% if thats when it started with its issues.

So are we saying that basically this lock is useless unless it has direct connection with the Coordinator?

otakon87, 2020-07-14:
I suspect it might be the problem! I hope @Koenkk can suggest us what to do.
I have a CC2531 I used as coordinator before, I can flash it with the router firmware and place near the lock and hope for the best! I don't know what else to do here....

nickcj931, 2020-07-14:
Seems like its the same issue as 
Duplicate of #3835

otakon87, 2020-07-14:
I'll close this so we can continue on the other open issue

closed 2020-07-14 by otakon87