Koenkk/zigbee2mqtt #3902: Konke 2AJZ4KPBS channels HA

matyasdelcampo, 2020-07-13:
Hi I have 
Konke 2AJZ4KPBS
zigbee2mqtt: Current version: 1.14.1
Home Assistant

Im running on channel: 11 and I know its running on channels:  15/20/25

But somehow I have paired one Konke before. So I have bought couple of more.. My questions are:.

1-Can I add them manualy(but I need to stay on channel 11(Aquara devices a IKEA)...
2-Can I Use just one channel at time?
3-Can I Switch to 15 and then roll back to 11?

Thanks Matyas

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version:

Koenkk, 2020-07-14:
1. I don't know what you mean with this.
2. Yes
3. No (required repairing of all devices)

matyasdelcampo, 2020-07-16:
Thanks. 

What Channel should I use for ikea/aqara and Konke? To make it all works

Thanks

Matyas

Koenkk, 2020-07-16:
15/20/25 (as Konke only works on that, Ikea/Aqara works on all).

closed 2020-07-19 by matyasdelcampo