Koenkk/zigbee2mqtt #3904: S31ZB - SONOFF Zigbee smart plug (US version) - multiple messages generated per event

yevgenb, 2020-07-13:
# Bug Report
## What happened
Multiple msgs received in response on state change of S31ZB - SONOFF Zigbee smart plug (US version) 
## What did you expect to happen
Single message should be generated per each topic
## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2530 + CC (E18-MS1-PCB)
Adapter firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}

Here is a debug log:

zigbee2mqtt:debug 2020-07-13 21:07:37: Received MQTT message on 'zigbee2mqtt/0x00124b001e7200a0/set' with data 'ON'
zigbee2mqtt:debug 2020-07-13 21:07:37: Publishing 'set' 'state' to '0x00124b001e7200a0'
zigbee2mqtt:debug 2020-07-13 21:07:37: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-13 21:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:debug 2020-07-13 21:07:37: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-13 21:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:debug 2020-07-13 21:07:37: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-13 21:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:debug 2020-07-13 21:07:37: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-13 21:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:debug 2020-07-13 21:07:37: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-13 21:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:debug 2020-07-13 21:07:37: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-13 21:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:info  2020-07-13 21:07:37: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'

Koenkk, 2020-07-14:
Fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-14 by Koenkk

yevgenb, 2020-07-14:
Thank you very much!

yevgenb, 2020-07-14:
I tried latest dev branch and the issue is still reproducible:

-----
{
  "data_path": "/share/zigbee2mqtt",
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": "mqtt",
    "password": "mqtt"
  },
  "serial": {
    "port": "/dev/ttyUSB0"
  },
  "advanced": {
    "log_level": "debug",
    "pan_id": 6754,
    "channel": 15,
    "network_key": [
      11,
      3,
      5,
      7,
      9,
      1,
      13,
      15,
      1,
      2,
      4,
      6,
      5,
      10,
      12,
      13
    ],
    "availability_blacklist": []
  },
  "ban": [],
  "whitelist": [],
  "queue": {},
  "experimental": {},
  "socat": {
    "enabled": false,
    "master": "pty,raw,echo=0,link=/dev/ttyZ2M,mode=777",
    "slave": "tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5",
    "restartdelay": 1.0,
    "initialdelay": 1.0,
    "options": "-d -d",
    "log": false
  }
}Debug information end
./run.sh: line 24: [Info] Configuration backup found in /share/zigbee2mqtt/.configuration.yaml.bk. Skipping config backup.: No such file or directory
[Info] Socat is DISABLED and not started
2020-07-14T20:58:32: PM2 log: Launching in no daemon mode
2020-07-14T20:58:32: PM2 log: App [npm:0] starting in -fork mode-
2020-07-14T20:58:32: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.1-dev start /app
> node index.js
zigbee2mqtt:info  2020-07-14 20:58:41: Logging to console and directory: '/share/zigbee2mqtt/log/2020-07-14.20-58-41' filename: log.txt
zigbee2mqtt:debug 2020-07-14 20:58:41: Removing old log directory '/share/zigbee2mqtt/log/2020-07-13.09-19-20'
zigbee2mqtt:debug 2020-07-14 20:58:43: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2020-07-14 20:58:43: Starting zigbee2mqtt version 1.14.1-dev (commit #)
zigbee2mqtt:info  2020-07-14 20:58:43: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-07-14 20:58:43: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[15],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"path":"/dev/ttyUSB0"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-07-14 20:58:46: zigbee-herdsman started
[... 7 lines cut ...]
zigbee2mqtt:info  2020-07-14 20:58:49: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-07-14 20:58:49: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":68}'
zigbee2mqtt:info  2020-07-14 20:58:49: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.1-dev","commit":"","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}'
zigbee2mqtt:info  2020-07-14 20:58:49: MQTT publish: topic 'homeassistant/switch/0x00124b001e7200a0/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/0x00124b001e7200a0/set","state_topic":"zigbee2mqtt/0x00124b001e7200a0","json_attributes_topic":"zigbee2mqtt/0x00124b001e7200a0","name":"0x00124b001e7200a0_switch","unique_id":"0x00124b001e7200a0_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001e7200a0"],"name":"0x00124b001e7200a0","sw_version":"Zigbee2mqtt 1.14.1-dev","model":"Zigbee smart plug (US version) (S31ZB)","manufacturer":"SONOFF"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-07-14 20:58:49: MQTT publish: topic 'homeassistant/sensor/0x00124b001e7200a0/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00124b001e7200a0","json_attributes_topic":"zigbee2mqtt/0x00124b001e7200a0","name":"0x00124b001e7200a0_linkquality","unique_id":"0x00124b001e7200a0_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00124b001e7200a0"],"name":"0x00124b001e7200a0","sw_version":"Zigbee2mqtt 1.14.1-dev","model":"Zigbee smart plug (US version) (S31ZB)","manufacturer":"SONOFF"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:debug 2020-07-14 20:59:09: Received MQTT message on 'zigbee2mqtt/0x00124b001e7200a0/set' with data 'OFF'
zigbee2mqtt:debug 2020-07-14 20:59:09: Publishing 'set' 'state' to '0x00124b001e7200a0'
zigbee2mqtt:debug 2020-07-14 20:59:09: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-14 20:59:09: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"OFF","linkquality":70}'
zigbee2mqtt:debug 2020-07-14 20:59:09: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-14 20:59:09: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-14 20:59:09: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":0}' from endpoint 1 with groupID 0
[... cut ...]
zigbee2mqtt:debug 2020-07-14 20:59:32: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-14 20:59:32: Received Zigbee message from '0x00124b001e7200a0', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-14 20:59:32: MQTT publish: topic 'zigbee2mqtt/0x00124b001e7200a0', payload '{"state":"ON","linkquality":65}'

Koenkk, 2020-07-15:
The device indeed still sends multiple zigbee messages (we cannot prevent this), but only 1 MQTT message is send.