Koenkk/zigbee2mqtt #3906: Crash of zigbee2mqtt container

valepe, 2020-07-14:
# Bug Report
## What happened

Sometime zigbee2mqtt crash and restart

## What did you expect to happen

Normal run

## How to reproduce it (minimal and precise)

Don't know

## Debug Info
Zigbee2mqtt version: 1.14.1 but also happened with 1.13.1
Adapter hardware: CC1352P-2
Adapter firmware version: 20200417

Zigbee2mqtt is running in a docker container on a Synology NAS.
I include an extract of the log. I noticed that every time that the container crash the last recorded item was from an aqara motion and illuminance sensor.

Never noticed this behavior with the CC2531. 

[https://pastebin.com/wdqazY5R](url)

Koenkk, 2020-07-14:
It seems that the adapter disconnects for some reason: `No such device or address, cannot open /dev/ttyACM0`, can you try connecting the adapter via a powered usb hub?

valepe, 2020-07-14:
> It seems that the adapter disconnects for some reason: `No such device or address, cannot open /dev/ttyACM0`, can you try connecting the adapter via a powered usb hub?

The adapter is connected with a 10m USB cable with signal amplifier. Maybe CC1353P-2 is more power hungry than CC2531. I'll try with a powered hub

valepe, 2020-07-28:
After more than a week with no crashes, I close the issue.
Thank you for your help

closed 2020-07-28 by valepe