Koenkk/zigbee2mqtt #3907: Add "power_outage_memory" for Aqara QBCZ11LM

denisov-vlad, 2020-07-14:
# Bug Report
## What happened
QBCZ11LM doesn't have power outage memory option but [ZNCZ02LM has](https://github.com/Koenkk/zigbee2mqtt.io/blob/develop/docs/devices/ZNCZ02LM.md#power-outage-memory). It's possible to remember power state in MiHome.

## What did you expect to happen
Add option to set power_outage_memory for this device.

## How to reproduce it (minimal and precise)

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: Conbee 2
Adapter firmware version:

Koenkk, 2020-07-14:
With a bit of luck the required command is identical, try adding `tz.ZNCZ02LM_power_outage_memory` to https://github.com/Koenkk/zigbee-herdsman-converters/blob/c5f407faa8e3ba736b93101155c6b686bb1479a2/devices.js#L862 and see if it works. See https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html on how to modify devices.js

denisov-vlad, 2020-07-15:
It works! Thanks!

closed 2020-07-15 by denisov-vlad

aladin2000, 2022-01-12:
Hi 
Is there a way to test if such a fix 

>>  toZigbee: [tz.on_off, tz.ZNCZ04LM_power_outage_memory],

Could it work for other devices like Bulb from Lidl ? 

regards