Koenkk/zigbee2mqtt #3908: NET2GRID N2G-SP Always reports Energy 0 Power 0

markheloking, 2020-07-14:
# Bug Report
## What happened

1. Pair N2G to zigbee2mqtt
1. Plug in device drawing significant power (>100W)
1. Observe power and energy reported being 0

## What did you expect to happen
The power and/or energy reported to not be 0 but the actual measurement value

## How to reproduce it (minimal and precise)
1. Pair N2G to zigbee2mqtt
1. Plug in device drawing significant power (>100W)
1. Observe power and energy reported being 0

## Debug Info
Zigbee2mqtt version: 1.14.1 Hass.io Add-on
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

## Discussion
It's not really a surprise to see that in the logs from other issues regarding this device, all measurements are also 0: #3556 

I think the real question becomes whether there's something funky going on trying to interpret the measurement values from the device correctly, or whether the device doesn't properly report the measurements at all.
Using the Net2Grid Zigbee bridge it does however provide proper measurement values for both energy and for power, so at least the device is capabile of it.

Koenkk, 2020-07-15:
Can you provide debug logging when attaching a load?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

markheloking, 2020-07-15:
For some reason after I enabled debugging, it worked. Now going back to no debugging it still works... Very weird. It feels almost as if it's a bug in the device's firmware where it doesn't switch to actively reporting power when it senses >0W power draw.

I guess this can be closed for now.

Things to try for other people running into the same issue:

- Restart zigbee2mqtt after the N2G device has previously been pared
- Press the button to disable and then enable the load, while the load is drawing power
- Enable debugging and see if reporting now changes (if this does change things, maybe we should start sharing logs on here to figure out what causes this issue)

closed 2020-07-15 by markheloking