Koenkk/zigbee2mqtt #3909: NET2GRID N2G-SP Reports energy but this is never added as an entity to homeassistant

markheloking, 2020-07-15:
# Bug Report
## What happened

1. Set `homeassistant: true` in config
1. Pair N2G device to zigbee2mqtt
1. Device shows up as device in homeassistant
1. Device has `state`, `linkquality` and `power` entities, but no `energy` entity

## What did you expect to happen
I expect the reported energy by the device to also appear as an entity in homeassistant through autodiscovery

## How to reproduce it (minimal and precise)
1. Set `homeassistant: true` in config
1. Pair N2G device to zigbee2mqtt
1. Device shows up as device in homeassistant
1. Device has `state`, `linkquality` and `power` entities, but no `energy` entity

## Debug Info
Zigbee2mqtt version: 1.14.1 Hass.io Add-on
Adapter hardware: CC2531
Adapter firmware version: CC2531_DEFAULT_20190608

## Logs
Normal logging, it seems that the mqtt message does seem to contain the energy parameter:
```
zigbee2mqtt:info  2020-07-15 22:40:25: MQTT publish: topic 'zigbee2mqtt/N2G_PowerSwitchMeter', payload '{"state":"ON","linkquality":13,"power":149,"energy":0.12}'
```
Debug loggin:
```
zigbee2mqtt@1.14.1 start /zigbee2mqtt-1.14.1
> node index.js
zigbee2mqtt:info  2020-07-15 22:41:41: Logging to console and directory: '/share/zigbee2mqtt/log/2020-07-15.22-41-40' filename: log.txt
zigbee2mqtt:debug 2020-07-15 22:41:41: Removing old log directory '/share/zigbee2mqtt/log/2020-07-14.00-54-25'
zigbee2mqtt:debug 2020-07-15 22:41:41: Loaded state from file /share/zigbee2mqtt/state.json
zigbee2mqtt:info  2020-07-15 22:41:41: Starting zigbee2mqtt version 1.14.1 (commit #unknown)
zigbee2mqtt:info  2020-07-15 22:41:41: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-07-15 22:41:41: Using zigbee-herdsman with settings: '{"network":{"panID":PANID,"extendedPanID":[IDLIST],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/share/zigbee2mqtt/database.db","databaseBackupPath":"/share/zigbee2mqtt/database.db.backup","backupPath":"/share/zigbee2mqtt/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"tcp://ip:port"},"adapter":{"concurrent":null}}'
zigbee2mqtt:info  2020-07-15 22:41:42: zigbee-herdsman started
zigbee2mqtt:info  2020-07-15 22:41:42: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:debug 2020-07-15 22:41:42: Zigbee network parameters: {"panID":PANID,"extendedPanID":"EXTENDEDID","channel":11}
zigbee2mqtt:info  2020-07-15 22:41:42: Currently 12 devices are joined:
zigbee2mqtt:info  2020-07-15 22:41:42: N2G_PowerSwitchMeter (0x84df0c0010000b12): N2G-SP - NET2GRID White Net2Grid power outlet switch with power meter (Router)
*snip*
zigbee2mqtt:info  2020-07-15 22:41:42: Zigbee: disabling joining new devices.
zigbee2mqtt:info  2020-07-15 22:41:43: Connecting to MQTT server at mqtt://core-mosquitto
zigbee2mqtt:debug 2020-07-15 22:41:43: Received Zigbee message from 'N2G_PowerSwitchMeter', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'zigbee2mqtt/N2G_PowerSwitchMeter', payload '{"state":"ON","linkquality":2,"power":149,"energy":0.12}'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'homeassistant/sensor/0x84df0c0010000b12/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:flash","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","json_attributes_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","name":"N2G_PowerSwitchMeter_power","unique_id":"0x84df0c0010000b12_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x84df0c0010000b12"],"name":"N2G_PowerSwitchMeter","sw_version":"Zigbee2mqtt 1.14.1","model":"White Net2Grid power outlet switch with power meter (N2G-SP)","manufacturer":"NET2GRID"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'homeassistant/switch/0x84df0c0010000b12/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/N2G_PowerSwitchMeter/set","state_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","json_attributes_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","name":"N2G_PowerSwitchMeter_switch","unique_id":"0x84df0c0010000b12_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x84df0c0010000b12"],"name":"N2G_PowerSwitchMeter","sw_version":"Zigbee2mqtt 1.14.1","model":"White Net2Grid power outlet switch with power meter (N2G-SP)","manufacturer":"NET2GRID"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'homeassistant/sensor/0x84df0c0010000b12/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","json_attributes_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","name":"N2G_PowerSwitchMeter_linkquality","unique_id":"0x84df0c0010000b12_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x84df0c0010000b12"],"name":"N2G_PowerSwitchMeter","sw_version":"Zigbee2mqtt 1.14.1","model":"White Net2Grid power outlet switch with power meter (N2G-SP)","manufacturer":"NET2GRID"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:debug 2020-07-15 22:41:43: Received Zigbee message from 'N2G_PowerSwitchMeter', type 'attributeReport', cluster 'seMetering', data '{"instantaneousDemand":151,"currentSummDelivered":[0,122],"currentSummReceived":[0,0]}' from endpoint 10 with groupID 0
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'zigbee2mqtt/N2G_PowerSwitchMeter', payload '{"state":"ON","linkquality":0,"power":151,"energy":0.12}'
zigbee2mqtt:info  2020-07-15 22:41:43: Connected to MQTT server
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'zigbee2mqtt/N2G_PowerSwitchMeter', payload '{"state":"ON","linkquality":0,"power":151,"energy":0.12}'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.1","commit":"unknown","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'homeassistant/sensor/0x84df0c0010000b12/power/config', payload '{"unit_of_measurement":"W","icon":"mdi:flash","value_template":"{{ value_json.power }}","state_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","json_attributes_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","name":"N2G_PowerSwitchMeter_power","unique_id":"0x84df0c0010000b12_power_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x84df0c0010000b12"],"name":"N2G_PowerSwitchMeter","sw_version":"Zigbee2mqtt 1.14.1","model":"White Net2Grid power outlet switch with power meter (N2G-SP)","manufacturer":"NET2GRID"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'homeassistant/switch/0x84df0c0010000b12/switch/config', payload '{"payload_off":"OFF","payload_on":"ON","value_template":"{{ value_json.state }}","command_topic":"zigbee2mqtt/N2G_PowerSwitchMeter/set","state_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","json_attributes_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","name":"N2G_PowerSwitchMeter_switch","unique_id":"0x84df0c0010000b12_switch_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x84df0c0010000b12"],"name":"N2G_PowerSwitchMeter","sw_version":"Zigbee2mqtt 1.14.1","model":"White Net2Grid power outlet switch with power meter (N2G-SP)","manufacturer":"NET2GRID"},"availability_topic":"zigbee2mqtt/bridge/state"}'
zigbee2mqtt:info  2020-07-15 22:41:43: MQTT publish: topic 'homeassistant/sensor/0x84df0c0010000b12/linkquality/config', payload '{"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","json_attributes_topic":"zigbee2mqtt/N2G_PowerSwitchMeter","name":"N2G_PowerSwitchMeter_linkquality","unique_id":"0x84df0c0010000b12_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x84df0c0010000b12"],"name":"N2G_PowerSwitchMeter","sw_version":"Zigbee2mqtt 1.14.1","model":"White Net2Grid power outlet switch with power meter (N2G-SP)","manufacturer":"NET2GRID"},"availability_topic":"zigbee2mqtt/bridge/state"}'
*snip*
[... cut ...] :"zigbee2mqtt/N2G_PowerSwitchMeter","name":"N2G_PowerSwitchMeter_linkquality","unique_id":"0x84df0c0010000b12_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x84df0c0010000b12"],"name":"N2G_PowerSwitchMeter","sw_version":"Zigbee2mqtt 1.14.1","model":"White Net2Grid power outlet switch with power meter (N2G-SP)","manufacturer":"NET2GRID"},"availability_topic":"zigbee2mqtt/bridge/state"}'
```

But this is suspiciously missing any energy reporting (or I'm not looking at this right).

Koenkk, 2020-07-15:
Fixed!

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-15 by Koenkk

markheloking, 2020-07-15:
What's the typical/average/ballpark turnover from develop to stable on this git? (would prefer not to switch over to dev branch for live environment if I can avoid it)

Koenkk, 2020-07-15:
It will be included in the next release (22 July).