Koenkk/zigbee2mqtt #3910: MAC no ack (233) error sending commands to an Osram smart plug connected via routers

sniper-dan, 2020-07-16:
I am getting an error 'MAC no ack' (233), when trying to send commands to an Osram smart plug. This only seems to occur when the connection to the network is via routers rather than the co-ordinator. I can pair and control the smart plug fine when right next to the co-ordinator, and where there is a direct connection to the co-ordinator from the plug showing on the network map. However when I move it to the next room it starts to get this error and no longer responds to commands. In the same and adjacent rooms I have 3 Samsung smartplugs which are connected and responding fine. The network map shows the Osram plug connected to all the Samsung plugs as routers with a strong signal  but not directly to the co-ordinator. In zigbee2mqttassistant the device shows online and connected. It does not appear to be a network connectivity issue in that case, but rather an issue with routing commands to the device from the co-ordinator via the routers. See also issues #2561 and #362 which is the same device type again with the same error.

See https://pasteboard.co/JhT9tGj.png for my network map. The device ending 2a8c is the Osram plug, the other unamed ones are the Samsung plugs.

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531 (with extension cable)
Adapter firmware version: 20190608

IVI053, 2020-07-16:
These devices have very very bad antenna design! Trust ne it's a shame that these things come from Osram!

I would advise you to mod the antenna with a short wire of the correct length dir 2,4GHz. If you dont feel comfortable opening up a mains conneteced device you should consider switching to another brand.

And trust ne, it is indeed a connection issue. I have five as well and a lot of problems. After antenna mod, no issues at all.

sniper-dan, 2020-07-16:
Even with a poorly designed antenna, the device is currently sat 5cm away from one of the routers so range should not be an issue. You can see on my network map that is connected to 3 routers and reporting good signal strength. This appears to be an issue specifically related to the processing of messages that are sent via a router rather than directly from the co-ordinator.

Koenkk, 2020-07-16:
Can you provide the herdsman debug logging of the problem?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

sniper-dan, 2020-07-17:
> Can you provide the herdsman debug logging of the problem?
> 
> To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Yes, pasted here: https://pastebin.com/CWH35EWN

Koenkk, 2020-07-18:
I've added some more debug logging, can you provide the logging again, please provide from the moment when you execute the command via MQTT until the error + 20 seconds after that.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

sniper-dan, 2020-07-19:
I think I caught it but there's so much log info it's hard to tell!

https://pastebin.com/KfJT8uE3

Koenkk, 2020-07-20:
I see zigbee2mqtt requests a route but there is no answer from the routers. To dive further into this we need to sniff the traffic : https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

sniper-dan, 2020-07-21:
Unfortunately I dont have a CC debugger in order to reflash my CC2531 stick. Before I go and buy one, is it likely that what we find is likely to lead to a solution, or just identification of one of the hardware devices needing to be replaced?

Koenkk, 2020-07-21:
I honestly don't know, this issue has already been reported a lot of times but never well investigated with a sniff:
- https://github.com/Koenkk/zigbee2mqtt/issues/2561
- https://github.com/Koenkk/zigbee2mqtt/issues/1755
- https://github.com/Koenkk/zigbee2mqtt/issues/762
- https://github.com/Koenkk/zigbee2mqtt/issues/2950
- https://github.com/Koenkk/zigbee2mqtt/issues/2871

In https://github.com/Koenkk/zigbee2mqtt/issues/2561#issuecomment-576855062 it was fixed when switching to the source routing firmware: https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

sniper-dan, 2020-07-21:
OK, presume one of these kits will give me everything I need? 
https://www.aliexpress.com/item/4001095299084.html?spm=a2g0o.productlist.0.0.34b147b56TN80q&algo_pvid=d975e6b9-f69f-4fd3-9627-f41bdece8607&algo_expid=d975e6b9-f69f-4fd3-9627-f41bdece8607-24&btsid=0b0a050b15953239540491405e8383&ws_ab_test=searchweb0_0,searchweb201602_,searchweb201603_

Koenkk, 2020-07-21:
Yes!

sniper-dan, 2020-07-22:
OK, I've ordered one but it'll probably take at least a month to arrive...

sniper-dan, 2020-07-28:
OK, all arrived and installed following your instructions, what am I looking for in the wireshark capture?

Koenkk, 2020-07-29:
Now sniff at the location between the osram plug and the coordinator (location is important, otherwise signals can be missed). Send a command which results in an error. Try to keep the sniff as short as possible, start right before you send the command, wait 60 seconds and stop the sniff (makes it easier to investigate)

sniper-dan, 2020-07-29:
Ok, done 2 captures:

https://www.dropbox.com/s/pokh0bzl6803wmu/capture%201.pcapng?dl=0
https://www.dropbox.com/s/dxo64g0yib0eoaf/capture2.pcapng?dl=0

Koenkk, 2020-07-29:
Thanks for the good sniffs, I don't fully understand what is happening yet.
- To what router is the OSRAM connected in this case? (please provide me that networkAddress of this device, can be found in `data/database.db`)
- Can you provide a same sniff of the working situation?
- I see that there is a Philips device communication on the same channel but does not seem to be in the network? (as it has a different network key, see e.g capture_2 no. 39). Does this make sense?

sniper-dan, 2020-07-29:
The OSRAM socket is connected to multiple routers, here's a screengrab of the network map at present with the network addresses:
https://www.dropbox.com/s/a0uu3ecycrklksj/Map.png?dl=0

I'll move the socket closer to the coordinator and take another sniff in the morning once it's had a chance to pair with it directly.
I also have some Hue bulbs paired to a Hue Hub on their own network, which is likely what the Philips device is.

Koenkk, 2020-07-29:
@sniper-dan I would recommend running the Hue bridge on another channel (you can change it easily in the Hue app), this to avoid interference from being the cause of this issue.

sniper-dan, 2020-07-29:
Done, but it doesn't seem to have made any immediate difference.

sniper-dan, 2020-07-30:
New sniffs of the socket working (as it's now directly connected to the coordinator) :
https://www.dropbox.com/s/lbkg895u330z88k/capture3.pcapng?dl=0
https://www.dropbox.com/s/5443j01ko33mins/capture4.pcapng?dl=0

New map:
https://www.dropbox.com/s/ed2u4hogkumz9oc/Map2.png?dl=0

I switched the socket on and off a couple of times at the beginning of each sniff with no issues.

Koenkk, 2020-07-31:
Thanks, can you:

- Update to latest dev (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)
- Replace `/app/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js` with https://github.com/Koenkk/zigbee-herdsman/blob/issue/z2m_3910/zStackAdapter.js
- Start zigbee2mqtt, sniff and try to control the plug again when it's connected via a router

sniper-dan, 2020-08-01:
Where do I replace that .js file? I am running HASSIO on an RPi so don't see the path you are referring to.

Koenkk, 2020-08-01:
You have to enter the Zigbee2MQTT container, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html#hassio-addon

sniper-dan, 2020-08-01:
My bash and vi skills are a little rusty, so hopefully I've done it right. 

New sniffs:
https://www.dropbox.com/s/sf8c6vkmovzgeze/capture5.pcapng?dl=0
https://www.dropbox.com/s/44ot7cjrsfq4j5h/capture6.pcapng?dl=0

Latest map:
https://www.dropbox.com/s/2jw2zebtba83i7q/map3.JPG?dl=0

Koenkk, 2020-08-02:
@sniper-dan doesn't seem to fix it unfortunately, one thing we should also test is to see if it works with the source routing firmware, can you try switching to that and see if things improve (doesn't require repairing)? https://github.com/Koenkk/Z-Stack-firmware/tree/master/coordinator/Z-Stack_Home_1.2/bin/source_routing

sniper-dan, 2020-08-02:
Done, but that seems to have caused another issue:

https://www.dropbox.com/s/nkb70z8iikhtz0b/map4.JPG?dl=0

Only the devices now showing a connection to the coordinator are reporting back LQI

sniper-dan, 2020-08-03:
OK, so weirdly despite the network map showing none of my sensors connected, and the Zigbee2MQTT log showing errors getting the LQI, they still seem to be connected and reporting back status changes. The Osram plug seems to have stopped communicating altogether though.

Koenkk, 2020-08-03:
@sniper-dan could you provide a sniff again when controlling the osram plug?

sniper-dan, 2020-08-04:
New sniff: https://www.dropbox.com/s/kypazmm00tko6fz/capture7.pcapng?dl=0

It's the outdoor version of that socket, but I expect they have the same guts on the inside.

[image]

Koenkk, 2020-08-05:
Thanks, at No. 16 I see the coordinator sending the on/off message to 0xb63a indirectly via 0x3b07. However I don't see 0x3b07 forwarding the message after this. Can you take the sniff at the location of this device to make sure we didn't miss it in the sniff? 0x3b07 is the device having networkAddress `15111` (you can look it up in the `data/database.db`.

sniper-dan, 2020-08-05:
https://www.dropbox.com/s/am5ch37gnpcgtwe/capture8.pcapng?dl=0

For info that router is situated right next to the Osram plug, so I'm taking the sniff next to both of them.

Things went a little differently this time. After looking up the router in the database.db I decided to rename it before taking the sniff to something more meaningful. After I renamed it I restarted Zigbee2MQTT to force it to push new discovery entries. Then I went and took the sniff. You'll see I sent 3 power on/off commands at the start of this sniff. The first failed, the second worked, and then the third failed. But as I type this I'm sending the same commands now and they seem to be working every time, albeit sometimes slightly delayed. I also just checked the network map and that has changed too:

https://www.dropbox.com/s/85a6xlwdnlg6cpu/map5.JPG?dl=0

Koenkk, 2020-08-05:
@sniper-dan that's interesting, does it keep working reliable now?

sniper-dan, 2020-08-05:
It appears so, yes. I'll keep an eye over it for the next 24 hours and report back.

sniper-dan, 2020-08-06:
Everything seems to be working fine now. I've even managed to move the socket outside and it's still responding normally. Is it OK for me to switch back to the stable branch from dev?

Koenkk, 2020-08-07:
Yes you can safely switch back

sniper-dan, 2020-08-07:
Still working fine - is our conclusion that this error can only be resolved by switching to source routing?

Koenkk, 2020-08-08:
@sniper-dan does the issue immediately reappear when switching to the default firmware? If yes, does it disappear again when switching to source routing?

sniper-dan, 2020-09-10:
Apologies for the late reply, to be honest I am reluctant to mess with the configuration now that this is working. Though I appreciate it would be useful to get a conclusion on whether the source routing is the answer, there is not much value for me in attempting to prove otherwise!

Koenkk, 2020-09-10:
@sniper-dan in case you want to confirm, follow https://github.com/Koenkk/zigbee2mqtt/issues/3910#issuecomment-670936861 , otherwise please close this.

closed 2020-09-10 by sniper-dan