Koenkk/zigbee2mqtt #3911: Ubisys S1 and 'decoupled mode'

sjorge, 2020-07-16:
I've been looking into the Ubisys S1 and I'm trying to figure out if it supports decoupled mode.
(input form the attached switch -> send to group of bulbs, leaving the power state unchanged, while being able to toggle the power state via mqtt)

https://www.ubisys.de/wp-content/uploads/ubisys-s1-technical-reference.pdf `7.3.4.On/off Cluster (Client)`

Mentions:
```
Notice: This cluster uses the binding table for managing command targets.When factory fresh, this cluster is bound to endpoint #1 to enable local control. This works even when the device is not commissioned yet.In addition, the device setup cluster on the Device Management endpoint can be used to configure the behavior of thecluster, i.e. which of the commands mentioned below will actually be sent, when the input state changes.Using the device setup cluster, you can configure this endpoint either to be used with a push-button (momentary switch, one stable position) or a rocker switch (two stable positions).
```

So from this I gather the on/off client cluster is bound to the on/off server cluster by default.
Does this means one can use the binding support to unbind the genOnOff cluster from itself, then rebind the genOnOff cluster to a group of bulbs?

Under `7.2.5.On/off Cluster (Server)` it mentions:

````
Notice: This cluster uses the binding table for managing reporting targets. Currently, there is no default reporting configuration in place, so you need to configure reporting, if you want to receive feedback on state changes.
```

Looking at the devices.js we do not bind to that cluster or configure reporting, so does this mean we are currently not getting state changes when something externally toggles the state?

@Koenkk does this sound correct? I kind of one to pick one up but at 99 EUR it's a rather expensive device.

Koenkk, 2020-07-16:
That indeed indicates that rebinding (and thus decoupling) would be possible. @felixstorm do you know more about this?

sjorge, 2020-07-16:
Good, that means I was not totally misreading the tech doc. Lets see what Felix has to say and I might pick one up to play around with. I am still looking for 'smart switches' for my bulbs but not finding anything that is not Aqara or TuYa.

Having these as an option would be great as they seem rather flexible on a lot of front (config wise)

felixstorm, 2020-07-17:
@sjorge Using them ‚decoupled‘ in the way you mention should indeed be possible. Although I never explicitly tried this myself with an S1, I have been using an ubisys C4 to control two ZigBee groups of two smart bulbs each for a few months now without a single issue (C4 inputs configured for one-button-dimming and bound to respective group on ZigBee level).  
If you are primarily looking for button inputs for your bulbs, then the C4s might be an even better fit - essential they contain the input side of 4 S1s and/or 2 J1s and have 4 input lines instead of only 2 on the S1 but no outputs.  
I am currently on vacation, so I cannot try it out myself, but if you go with an S1 then I would also try to unbind the device from itself first and then bind it to a different device. Just make sure you specify the endpoint hosting the _client_ cluster as the binding source - it probably is endpoint 2 for a S1 (definitely not endpoint 1). Unfortunately the internet connection is pretty poor here so I cannot look up the correct endpoint number from the ubisys docs at the moment.

And about reporting external state changes: When you enable reporting globally, the server `genOnOff` cluster on endpoint 1 will (should?) automatically be bound to the coordinator and be setup to report state changes, so z2m also gets notified when somebody presses the button to turn on the light. I do not use the inputs on the S1s in my setup (I only use them to be able to monitor if the stove is really switched off and to potential force it off in case somebody forgot to do so), but I know for sure that it does work well this way with a D1.

I am not personally affiliated with ubisys (apart from getting some rebate on their modules through the company I work for, but I work in a completely different area at my full-time-job), but my experiences are very good so far. I have been using 15-20 J1s (to control venetian blinds and some shutters around the house) and a few S1s, D1s and C4s for 2-3 years now and the only issues I have are that I loose a device from the network every 2-3 months or so, but this could very well be a problem on the coordinator side. All in all they seem to be pretty well-thought-out and professional devices.

sjorge, 2020-07-17:
I was looking at the S1 specifically so I can turn on/off the bulbs on a physical level if I need to, but generally they should be powered and toggled using a zigbee group, fitting a simple on/off + a C4 into the existing spots won't work as there is only room for one of those little devices.

Good to know they are working well or you. I'll look around for a local supplied and pick one up to test.
If I get it to work, I'll write some docs for it.

sjorge, 2020-08-09:
Looks like they ship to where I live, one S1 is on the way for testing

sjorge, 2020-08-12:
OK, I have one installed and by default it indeed hardcuts the power to the light when using the switch. I can also change the state over mqtt... however un(binding) doesn't seem to work.

```
mosquitto_pub -t zigbee2mqtt/bridge/unbind/0x001fee0000005e6c -m 0x001fee0000005e6c
```

```
Zigbee2MQTT:error 2020-08-12 15:02:53: Nothing to unbind from '0x001fee0000005e6c' to '0x001fee0000005e6c'
```

```
mosquitto_pub -t zigbee2mqtt/bridge/bind/0x001fee0000005e6c -m office/lights
```

```
Zigbee2MQTT:error 2020-08-12 15:00:40: Nothing to bind from '0x001fee0000005e6c' to 'office/lights'
```

@Koenkk looking at devices.js, I think some bits that indicate what 'endpoint 1' seems to be missing.

It doesn't look like there are any endpoints that we interact with aside from ep3 for saMetering.

sjorge, 2020-08-12:
I'm not sure on the reporting for genOnOff... toggling the switch does seem to change the state... but perhaps not in decoupled mode...

Koenkk, 2020-08-12:
Can you share the `data/database.db` entry of this device?

sjorge, 2020-08-12:
Already left so copied via phone, sorry for the formatting

```
{"id":65,"type":"Router","ieeeAddr":"0x001fee0000005e6c","nwkAddr":44429,"manufId":4338,"manufName":"ubisys","powerSource":"Mains (single phase)","modelId":"S1 (5501)","epList":[1,2,200,232,242,3],"endpoints":{"1":{"profId":260,"epId":1,"devId":266,"inClusterList":[0,3,4,5,6],"outClusterList":[],"clusters":{"genBasic":{"attributes":{"modelId":"S1 (5501)","manufacturerName":"ubisys","powerSource":1,"zclVersion":2,"appVersion":1,"stackVersion":1,"hwVersion":6,"dateCode":"20200304-DE-FB0","swBuildId":""}},"genOnOff":{"attributes":{"onOff":1}}},"binds":[{"cluster":6,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]},"2":{"profId":260,"epId":2,"devId":1,"inClusterList":[0,3],"outClusterList":[5,6,8],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"3":{"profId":260,"epId":3,"devId":1281,"inClusterList":[0,1794,2820],"outClusterList":[],"clusters":{"genBasic":{"attributes":{}},"seMetering":{"attributes":{"multiplier":1,"divisor":1000,"instantaneousDemand":1}}},"binds":[{"cluster":1794,"type":"endpoint","deviceIeeeAddress":"0x00124b001938a7e5","endpointID":1}]},"200":{"profId":49155,"epId":200,"devId":0,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"232":{"profId":260,"epId":232,"devId":1287,"inClusterList":[0,21,64512],"outClusterList":[3,25],"clusters":{"genBasic":{"attributes":{}}},"binds":[]},"242":{"profId":41440,"epId":242,"devId":102,"inClusterList":[33],"outClusterList":[33],"clusters":{},"binds":[]}},"appVersion":1,"stackVersion":1,"hwVersion":6,"dateCode":"20200304-DE-FB0","zclVersion":2,"interviewCompleted":true,"meta":{"reporting":1,"configured":3},"lastSeen":1597243900640}
```

felixstorm, 2020-08-12:
Not in front of my computer, but as stated before IMHO you need to unbind and bind endpoint 2 (as this is the client endpoint of the inputs sending out the commands on the S1) to make things work, e.g. something like this:
```
mosquitto_pub -t zigbee2mqtt/bridge/unbind/0x001fee0000005e6c/2 -m 0x001fee0000005e6c
mosquitto_pub -t zigbee2mqtt/bridge/bind/0x001fee0000005e6c/2 -m office/lights
```
As long as you just use the endpoint numbers, they do not need to be explicitly defined in`devices.js`.

And reporting should also work in decoupled mode as the report is being sent out by endpoint 1 (which manages the output). It does not matter where the command to switch the output on or off comes from (local input or remote zigbee command) - the report always covers the actual state of the output.

Felix

sjorge, 2020-08-12:
> Not in front of my computer, but as stated before IMHO you need to unbind and bind endpoint 2 (as this is the client endpoint of the inputs sending out the commands on the S1) to make things work, [...]

Ah excellent! I had what to unbind/bind right but couldn't figure out how to address the endpoints, this just works!

I'll write some docs for this tomorrow or later tonight if the heat dies down.  Not cheap but works install was a breeze and works great!

sjorge, 2020-08-12:
Looks like I need to do more digging,

Of the 4 ikea panels in the group only one updates the status via reporting. The other 3 do a check in a few minutes late with state=on, but are not on.  If I use a trådfri remote or via mqtt to the to the group it works fine.

~ sjorge

sjorge, 2020-08-12:
I stand corrected, the behavior is the same between the tradfri remote and the s1!
Looks like the FLOAT panels do not get reporting configured on the latest dev branch.

After closing inspection, none of the IKEA bulbs do! I am sure it they were reporting fine before, but now all lights i hart cut power and bring back... don’t report anymore.

Last I remember it working fine was last week after a cut the power to all bulbs. Did anything change on dev since then @koenkk ?

~ sjorge

sjorge, 2020-08-12:
I wonder if it is related to people seeing 0 brightness for some bulbs that got reported recently.

As some lamps after power cut, not only a GU10 only report genOnOff but no more brightness or color.

~ sjorge

sjorge, 2020-08-12:
Updated to today’s dev and reporting is working again now! Also constantly seeing the brightness 0 now when externally toggling. But it auto correct a second later by attReport.

So sorry for the unrelated noise, will close this once I have updated the docs.

felixstorm, 2020-08-13:
Good to hear that it does work now.

If you want to connect a push-button switch (momentary, one stable position, "Taster" in German) to also control the brightness of your bound lights (Ikea LED panels?), you could reconfigure your input on the S1 as follows (basically taken from the first half of the default configuration of the D1 and documented in it's technical reference):
```
pub -t zigbee2mqtt/0x001fee0000005e6c/set -m '{"configure_device_setup": {"inputActions": [[0, 7, 2, 6, 0, 2], [0, 134, 2, 8, 0, 5, 0, 50], [0, 198, 2, 8, 0, 5, 1, 50], [0, 11, 2, 8, 0, 7]]}}'
````
And to get back to the default it should be (documented in the S1's technical reference):
```
pub -t zigbee2mqtt/0x001fee0000005e6c/set -m '{"configure_device_setup": {"inputActions": [[0, 13, 2, 6, 0, 2], [0, 3, 2, 6, 0, 2]]}}'
````

(I use MQTT CLI in shell mode, so you might have to fiddle with the quotes a bit to get it to work with your client.)

Felix

sjorge, 2020-08-13:
@felixstorm currently it has this switch on top: https://ecataleg.be/nl/product/741540

One I get S1 (maybe S2 for some of my double switches) I might replace the switches, but it will take a few months as the S1's are not cheap. (They look very very well build though!)

I was looking at replacing it with https://ecataleg.be/nl/product/741547, not sure if that is the type of switch you are describing. if it is, how would those work for brightness? tap toggle, tap and hold to cycle brightness?

felixstorm, 2020-08-14:
> I was looking at replacing it with https://ecataleg.be/nl/product/741547, not sure if that is the type of switch you are describing. if it is, how would those work for brightness? tap toggle, tap and hold to cycle brightness?

Yes, that switch looks like one I was talking about and operation should work exactly as you say.

closed 2020-08-14 by Koenkk

sjorge, 2020-08-19:
> Good to hear that it does work now.
>
> If you want to connect a push-button switch (momentary, one stable position, "Taster" in German) to also control the brightness of your bound lights (Ikea LED [...]

@felixstorm so I picked up one of those new switch buttons at the local shop, because they were cheapish and needed some more wire.

z2m logs 2 warnings when setting this though, is this normal? 
```
Zigbee2MQTT:warn  2020-08-19 13:16:41: ubisys: Device setup read for 'bathroom/lights/switch': {"0":[0]}
Zigbee2MQTT:warn  2020-08-19 13:16:41: ubisys: Device setup read for 'bathroom/lights/switch': {"1":[{"type":"Buffer","data":[0,7,2,6,0,2]},{"type":"Buffer","data":[0,134,2,8,0,5,0,50]},{"type":"Buffer","data":[0,198,2,8,0,5,1,50]},{"type":"Buffer","data":[0,11,2,8,0,7]}]}
```

It does seem to work though, clicking the button toggles the lights. Hold raises or lowers the brightness (alternating)

If this is normal, I'll update the docs (probably best under the C4 one) with an example to switch between momentum and rocker switches.

sjorge, 2020-08-19:
It also looks like this sets the default switch state when restoring power to the S1 to off, I'll need to dig into the manual a bit I think

sjorge, 2020-08-19:
Ok, the default state seems to always be off, I probably toggled it on physically the first time before unbinding/binding.
I wonder if there is a configuration option too somewhere to set the initial state.

felixstorm, 2020-08-20:
> z2m logs 2 warnings when setting this though, is this normal?

Yes, this is normal. I just wanted to make sure users are able to see the values in the log (primarily when reading them out from the devices) even when the log level is set to the default.

> I'll update the docs (probably best under the C4 one) with an example to switch between momentum and rocker switches.

Yes, this would be great. To document the different input configurations with samples has been on my to-do list for quite some time now, but I never found the time to actually do it. So if you could start, this would surely be of great help.  
And documenting it for the C4 would also have been my way to do it (I think the C4 also has a section about it, but rather abstract without any samples and the other ubisys devices should already have links to it).

> I wonder if there is a configuration option too somewhere to set the initial state.

Unfortunately I am not aware of any option to set the initial state. Although on the other hand I cannot remember having to explicitly turn on my S1s (in line with the stove to be able to cut off it's power remotely) after a power outage either...