Koenkk/zigbee2mqtt #3921: IKEA Tradfri LED1836G9 not accepting brightness value 192

Imaginous, 2020-07-17:
# Bug Report
## What happened
When sending a brightness value of 192 to a IKEA Tradfri LED1836G9 the bulb does not react to the value.
It even seems the command is not send by z2m. Sending any other value works without a problem.
The bulb has the latest firmware.

## What did you expect to happen
Bulb turning to the set value.

## How to reproduce it (minimal and precise)
Send a JSON to the bulb with brightness value 192 (example):
zigbee2mqtt/Woonkamer_Schemerlamp_Hoek/set
{ "state": "on",  "brightness": 192,  "transition": 0 }

Sending any other brightness value works without a problem.

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: Conbee II
Adapter firmware version: 0x26580700

Koenkk, 2020-07-18:
Can you post the herdsman debug logging of this issue?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

Imaginous, 2020-07-19:
## Sending the not working brightness value 192

 zigbee-herdsman:deconz:frameParser device state: 10100010 +40s
  zigbee-herdsman:controller:endpoint Command 0xec1bbdfffe9ec52c/1 genLevelCtrl.moveToLevelWithOnOff({"level":192,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +1m
  zigbee-herdsman:deconz:adapter zclFrame.payload: +1m
  zigbee-herdsman:deconz:adapter { level: 192, transtime: 0 } +0ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xa41 EP:1 SeqNr. 38 request id: 3 +7s
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 39 +3s
  zigbee-herdsman:deconz:frameParser device state: 10100010 +10s

## Sending a working brightness value 190

 zigbee-herdsman:controller:endpoint Command 0xec1bbdfffe9ec52c/1 genLevelCtrl.moveToLevelWithOnOff({"level":190,"transtime":0}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) +0ms
  zigbee-herdsman:deconz:adapter zclFrame.payload: +3m
  zigbee-herdsman:deconz:adapter { level: 190, transtime: 0 } +0ms
  zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0xa41 EP:1 SeqNr. 28 request id: 2 +9s
  zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 2 status: 0 +9s
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110 +8ms
  zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 29 +50ms
  zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0xa41 request id: 2 confirm status: 0 +39ms
  zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send +73ms
  zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010 +39ms
  zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 30 +50ms
  zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 30 srcAddr: 0xa41 destAddr: 0x0 profile id: 0x104 cluster id: 0x8 +12ms
  zigbee-herdsman:deconz:frameParser response payload: 8,3,11,4,0 +0ms
  zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":3,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":4,"statusCode":0}},"address":2625,"endpoint":1,"linkquality":159,"groupID":null}' +2m
  zigbee-herdsman:deconz:adapter response received +54ms
zigbee2mqtt:info  2020-07-19 18:38:47: MQTT publish: topic 'zigbee2mqtt/Woonkamer_Schemerlamp_Hoek', payload '{"update_available":false,"state":"ON","linkquality":157,"brightness":190}'
  zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 31 +903ms
  zigbee-herdsman:deconz:frameParser device state: 10100010 +904ms

Koenkk, 2020-07-20:
This looks to be a deconz adapter specific issue, @ChrisHae could you take a look here?

ChrisHae, 2020-07-21:
Yes this was a bug in the deconz adapter that was fixed in the latest commit 5 Days ago:
https://github.com/Koenkk/zigbee-herdsman/commit/6061c41ecb772f7f0a592dfe8b1a87da0cff36b1

Koenkk, 2020-07-21:
@ChrisHae  thanks!

@Imaginous please check out latest dev branch (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html) or wait for new zigbee2mqtt release (which will be released tomorrow).

Imaginous, 2020-07-21:
@Koenkk, @ChrisHae thanks for the quick support and the fact that it was solved with another bug fix.

No time to update today, so I wil go for the stable version tomorrow.

Thanks again.

closed 2020-08-28 by github-actions[bot]