Koenkk/zigbee2mqtt #3923: Tradfri LED1624G9 doesn't report color

elfhack, 2020-07-17:
# Bug Report
## What happened
When changing color on the bulb via a bound E1524 remote (5-button puck), the colors are not getting reported back to homeassistant. According to [the docs](https://www.zigbee2mqtt.io/information/report.html) this should work, the relevant portion of my configuration.yaml is as follows
```yaml
advanced:
  report: true
devices:
  '0x000b57fffe39eb73':
    friendly_name: Remote
  '0xd0cf5efffe2357cd':
    friendly_name: Couch_1
    transition: 0.5
  '0xd0cf5efffe2d3b55':
    friendly_name: Couch_2
    transition: 0.5
  '0xd0cf5efffe292755':
    friendly_name: Couch_3
    transition: 0.5
groups:
  '901':
    friendly_name: Couch
    retain: false
    optimistic: true
    transition: 0.5
    devices:
      - 0xd0cf5efffe2357cd/1
      - 0xd0cf5efffe2d3b55/1
      - 0xd0cf5efffe292755/1
```
## What did you expect to happen
The bulb state would reflect x/y color coords. The bulb state updates power state, brightness, but not x/y color coords
## How to reproduce it (minimal and precise)
Press left or right on the puck to change group color
## Debug Info
Zigbee2mqtt version: 1.14.1 (tried 1.14.1-dev, no difference)
Adapter hardware: CC2531
Adapter firmware version: 20190608

Koenkk, 2020-07-18:
I think the bulb doesn't provide color updates because the remote changes it using scenes and not direct color commands.

To verify this, can you provide the zigbee2mqtt debug logging when changing the color from zigbee2mqtt?

MattWestb, 2020-07-19:
Also trying updating the bulb firmware. Its the old RGBWW LL and have not getting any Z30 update but an refresh in the 29th June 2020 update:
"CWS1.0 (1.3.013).
Fixed On Off State after OTA Upgrade."
I have one but using it stand alone without the remote.

elfhack, 2020-07-19:
@Koenkk you're absolutely right, and it actually goes deeper than that. Let me first explain my setup. Before getting a cc2531 I've used a Hue bridge combined with a 3 round remotes, a set of different tradfri bulbs (dimmable, white color, and rgb), 3 innr RGBs and 2 Hue dimmables. Now, one can pair these round remotes to the Hue bridge, then to the bulb, but the remote doesn't show up on the hue bridge as a device, and the arrow buttons do not work. 

Now, I was hoping that when switching to z2mqtt, i would go around that problem. First I was planning to just handle all the buttons with node-red, but then I read about binding, so I did that. Minding the fact that those remotes are hard to cooperate with binding, it turns out that the side buttons work for tradfri color whites (but when you do that, they report back to the coordinator in a way that home assistant doesn't understand, changing x/y values, but not the white colour that HA uses), but not for RGB bulbs - because as you say, they use scenes. UNLESS one decides to make a group 0, then bind the remote to that group. Then it works perfectly. One can also change the bulbs to the original group that the remote binds itself to (in my case 903). However, the bulbs then loose the ability to report altogether, although control is still possible.

All in all, while binding gives one the possibility to use the bulbs with the remote without the presence of the coordinator, it comes with too many gotchas, so I'll be content to just use node-red to program in my own functionality instead. Also, i think this is the command you're looking for

```
debug 2020-07-17 22:03:43: Received Zigbee message from 'tradfri-switch-round0', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":256,"value2":13}' from endpoint 1 with groupID 0
debug 2020-07-17 22:03:49: Received Zigbee message from 'tradfri-switch-round0', type 'commandTradfriArrowSingle', cluster 'genScenes', data '{"value":257,"value2":13}' from endpoint 1 with groupID 0
```

@MattWestb I've updated the bulbs when I paired them to zigbee2mqtt. No change.

MattWestb, 2020-07-19:
Very understanding but i think better then having the older fw. The LED1624G9 one of the oldest bulb Ikea still producing and its still Zigbee LL 2.X and not getting one Z30 update as the most other have getting. Sorry for not getting the binding working all the way but the E1524 its famous for the sean selection problem. You can doing and have doing very much in node red but the not so good thing its that all things must being online for it to working opposite to zigbee binding.
Good work done !!

elfhack, 2020-07-19:
@MattWestb You misunderstood me. The bulbs are on 1.3.013, and the remote is on 2.3.014

MattWestb, 2020-07-19:
No misunderstanding the both is its at the latest version as my is.

basszt, 2020-08-14:
Hi there,
is it somehow possible to trigger these scenes without the Tradfri remote via zigbee2mqtt?

closed 2020-09-14 by elfhack