Koenkk/zigbee2mqtt #3925: Support for MOES ZigBee Switch Module

MikeVRT, 2020-07-17:
Can be used from Smart Life and Tuya (from description).
[Item](https://aliexpress.com/item/4001065019839.html) in official MOES' store on AliExpress.
It's like Xiaomi relay (LLKZMK11LM) but with one relay and switch support. Can be placed behind switch.

[image]

MikeVRT, 2020-07-19:
I bought a new one. If need help for add to device list, please give me advice what to do.

Koenkk, 2020-07-20:
Please follow: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html

closed 2020-08-27 by github-actions[bot]

sorryusernameisalreadytaken, 2021-03-23:
> 
> 
> I bought a new one. If need help for add to device list, please give me advice what to do.

Did you got your item? Can it measure something like watt or voltage?