Koenkk/zigbee2mqtt #3926: Xiaomi WXCJKG13LM is __automatically__ bound to __all__ zigbee lights; how to unbind?

prankousky, 2020-07-18:
Hi everybody,

tl;dr: I connected two `WXCJKG13LM` ([ziggbee2mqtt.io link](https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html)) to my coordinator. The **both** control **all** my zigbee2mqtt lights. I did **not** follow [this step](https://www.zigbee2mqtt.io/devices/WXCJKG13LM.html#binding), so should the not automatically be bound to the coordinator, not to any lights..?

This happens with both switches that I added recently. I did already own a third switch, that has paired fine and can be used to control things without this issue.

# Bug Report
## What happened

I connected each switch to my coordinator. By running `tail -f /opt/zigbee2mqtt/data/configuration.yaml`, I was able to observe that those switches had both actually connected to my coordinator and registered by `zigbee2mqtt`. 

I have ***not*** written any automations regarding these switches in `Home Assistant` (which I use to control all my zigbee2mqtt devices) yet. However, when I press buttons, they act like this:

```
Button 1 | Button 2
Button 3 | Button 4
Button 5 | Button 6
```

Click Button 1: Lights Off
Click Button 2: Lights On
Click Button 3: Decrese Brightness
Click Button 4: Increase Brightness
Click Button 5 & 6: nothing as far as I can tell
Hold *Any* Button: nothing as far as I can tell

## What did you expect to happen

I expected *nothing* to happen except for `zigbee2mqtt` reporting (and displaying in `Home Assistant`), that the appropriate button I pressed had, indeed, been pressed. 

## How to reproduce it (minimal and precise)

I didn't do anything besides pairing these switches.

## Debug Info
Zigbee2mqtt version: `1.14.1`
Zigbee commit: `a6fff1e`
Adapter hardware: `CC1352P-2`
Adapter firmware version: `CC1352P_2_20191106.hex`

How can I fix this? Thank you for your help :)

Koenkk, 2020-07-18:
Did both configure successfully? You can trigger this manually via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , log shows if its OK or not, make sure to wakeup the switches right before executing this (otherwise it will always fail).

prankousky, 2020-07-24:
Thank you. After configuring one of the switches, then re-pairing the other, both seem to work fine now :)

closed 2020-07-24 by prankousky