Koenkk/zigbee2mqtt #3927: IKEA E1757 (ikea fyrtur blind) state

dezral, 2020-07-18:
# Bug Report
## What happened
when using {"position":X} with fyrtur blind, it don't send a current state..

##What did you expect to happen
i expected it would have an payload with the state like  opening, closing or stop.

## How to reproduce it (minimal and precise)
normal usage

## Debug Info
Zigbee2mqtt version: 
info  2020-07-18 15:21:25: Starting zigbee2mqtt version 1.11.0 (commit #e140cf5)

Adapter hardware: CC2531

Adapter firmware version: 
info  2020-07-18 15:21:27: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

`{"position":74,"linkquality":44,"update_available":false,"battery":98}
`

reason for wanting this, is that homebridge can't show the state of the blind correctly when set.. sometimes homekit don't update it's status unless it get a stop value. 

mqttthing:
https://github.com/arachnetech/homebridge-mqttthing/blob/master/docs/Accessories.md#window-covering

>    "positionStateValues": [ "decreasing-value", "increasing-value", "stopped-value" ] <

Koenkk, 2020-07-18:
I think we cannot determine if the state is `opening` or `closing` (as we don't know if the blinds are moving). However with the code below we will get `OPEN` and `CLOSED`. You can change this in `fromZigbee`.js, see https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html how to edit.

```js
cover_position_tilt: {
    cluster: 'closuresWindowCovering',
    type: ['attributeReport', 'readResponse'],
    convert: (model, msg, publish, options, meta) => {
        const result = {};
        // ZigBee officially expects 'open' to be 0 and 'closed' to be 100 whereas
        // HomeAssistant etc. work the other way round.
        // ubisys J1 will report 255 if lift or tilt positions are not known.
        if (msg.data.hasOwnProperty('currentPositionLiftPercentage')) {
            const liftPercentage = msg.data['currentPositionLiftPercentage'];
            result.position = liftPercentage <= 100 ? (100 - liftPercentage) : null;
            result.state = result.position === 0 ? 'CLOSED' : 'OPEN';
        }
        if (msg.data.hasOwnProperty('currentPositionTiltPercentage')) {
            const tiltPercentage = msg.data['currentPositionTiltPercentage'];
            result.tilt = tiltPercentage <= 100 ? (100 - tiltPercentage) : null;
        }
        return result;
    },
},
```

dezral, 2020-07-19:
thanks i made some node-red to fix this instead ;-)

Koenkk, 2020-07-19:
Good! I assume this can be closed now

closed 2020-07-19 by Koenkk