Koenkk/zigbee2mqtt #3929: Philips Hue ST64 no brightness

emaras, 2020-07-18:
Can someone help with set brightness Philips 929002241201.

I can turn ON/OFF bulb with command: zigbee2mqtt/Hue1/set On/OFF
but I can't set brightness. I try command: zigbee2mqtt/Hue1/set 50, zigbee2mqtt/Hue1/brightness 50, zigbee2mqtt/Hue1/brightness/set 50, 

When I send command by Mqtt.FX zigbee2mqtt/Hue1/brightness "brightness":"50" bulb has dimming and turn off.

interesting is status mqtt message zigbee2mqtt/Hue1/brightness is 0 or 28 all time

Koenkk, 2020-07-18:
See https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#without-json (`zigbee2mqtt/Hue1/set/brightness` with payload `50`)

closed 2020-07-18 by Koenkk

emaras, 2020-07-21:
Answer is: 
zigbee2mqtt/HUE1/set {"brightness"50}

For OH
Type dimmer : HUE1 "HUE1"      [ stateTopic="zigbee2mqtt/HUE1/brightness", commandTopic="zigbee2mqtt/HUE1/set",  min=0, max=254, step=1, formatBeforePublish="{\"brightness\":%s}" ]