Koenkk/zigbee2mqtt #3930: Error pairing E1734 Ikea Button

Dragon0501, 2020-07-18:
I tried to pair an Ikea device. It is an E1734. Everytime I push the pairing button an error occured. It doesn't matter at which accesspoint I try to pair.

[image]

Dragon0501, 2020-07-18:
Problem solved. Waited for three or four hours. Retried and paired at first try...

closed 2020-07-18 by Dragon0501

MattWestb, 2020-07-19:
Good that you get the E1734 pared.
Its can being that you have being hitted of the famous paring bug.
The Z2M team it's on the hunt and have have made some progress but still not implanted in the code / firmware.

Dragon0501, 2020-07-19:
It still shows an error code when pairing but it works.

MattWestb, 2020-07-19:
So both good and bad for you :-)  :-(
I'm not sure if its the paring bug described here https://github.com/Koenkk/zigbee2mqtt/issues/3177 and i have all my E1743 its in production state in the Ikea GW and deCONZ so i cant testing the paring in Z2M for the moment.
By the way: have you checking if the firmware its at the latest version in the E1743 ?
It was little crappy in the first Z30 releases.

Dragon0501, 2020-07-19:
Firmware check failed,... device didn't respond to OTA request. I tried But impossible...

MattWestb, 2020-07-19:
Did you waking up the E1743 and keeping it awake then sending the request to it ?
Sorry i was forgetting saying that :-(
Normally its enough pressing the off key with some second between repeating it for keeping it awake.

Dragon0501, 2020-07-19:
Firmware checked. no update available.

MattWestb, 2020-07-19:
That was good.
Then you continuing working with battery powered devices = end devices (and some exoitc mains powered) that is normally sleeping they must being waked up for listening on command the system its sending.
One example: if having one pared motion sensor and like to binding it to a group then its necessary triggering some motion for waking it up one short moment before sending the configuration command or its timing out = not working. Main powered devices = routers (the most of them) devices its normally not sleeping and is always online so no need to do anything then sending one command to them.