Koenkk/zigbee2mqtt #3934: Zigbee lights dont restore brightness when turned on through homeassistant homekit

SalkinDK, 2020-07-19:
# Bug Report
When turning on my zigbee lights through the homeassistant homekit component thy turn on at full brightness, but if i turn them on tin the lovelace interface they restore the old brightness just fine.
it only happens with my zigbee light running with zigbee2mqtt, my other lights restore just fine.
## What happened
Lights turn on at full brightness when turned on in homekit, but restores old brightness just fine when turned on in homeassistant lovelace.
## What did you expect to happen
restore old brightness in homekit
## How to reproduce it (minimal and precise)
turn on light in homekit
## Debug Info
Zigbee2mqtt version: hassio add-on 1.14.1 
Adapter hardware: electrolama zzh (cc26x2r1)
Adapter firmware version: CC26X2R1_20200417

Hass issue here: https://github.com/home-assistant/core/issues/37955
Reddit post here: https://www.reddit.com/r/homeassistant/comments/ht38id/lights_always_turn_on_to_full_brightness/

Koenkk, 2020-07-19:
Can you post logging while turning on via lovelace vs homekit?

SalkinDK, 2020-07-19:
of course sorry for not posting this first.

turning on then off in lovelace and then on and off in homekit, without debug.

`zigbee2mqtt:info  2020-07-19 13:06:22: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"ON","linkquality":63,"brightness":127,"update_available":false}'
zigbee2mqtt:info  2020-07-19 13:06:24: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"OFF","linkquality":63,"brightness":0,"update_available":false}'
zigbee2mqtt:info  2020-07-19 13:06:31: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"ON","linkquality":63,"brightness":254,"update_available":false}'
zigbee2mqtt:info  2020-07-19 13:06:37: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"OFF","linkquality":63,"brightness":0,"update_available":false}'
`

turning on then off in lovelace and then on and off in homekit, with debug.

`zigbee2mqtt:debug 2020-07-19 13:12:23: Received MQTT message on 'zigbee2mqtt/Sofa_light/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-07-19 13:12:23: Publishing 'set' 'state' to 'Sofa_light'
zigbee2mqtt:info  2020-07-19 13:12:23: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"ON","linkquality":69,"brightness":138,"update_available":false}'
zigbee2mqtt:debug 2020-07-19 13:12:24: Received MQTT message on 'zigbee2mqtt/Sofa_light/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-07-19 13:12:24: Publishing 'set' 'state' to 'Sofa_light'
zigbee2mqtt:info  2020-07-19 13:12:25: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"OFF","linkquality":69,"brightness":0,"update_available":false}'
zigbee2mqtt:debug 2020-07-19 13:15:33: Received MQTT message on 'zigbee2mqtt/Sofa_light/set' with data '{"state": "ON", "brightness": 254}'
zigbee2mqtt:debug 2020-07-19 13:15:33: Publishing 'set' 'brightness' to 'Sofa_light'
zigbee2mqtt:info  2020-07-19 13:15:33: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"ON","linkquality":69,"brightness":254,"update_available":false}'
zigbee2mqtt:debug 2020-07-19 13:15:36: Received MQTT message on 'zigbee2mqtt/Sofa_light/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-07-19 13:15:36: Publishing 'set' 'state' to 'Sofa_light'
zigbee2mqtt:info  2020-07-19 13:15:36: MQTT publish: topic 'zigbee2mqtt/Sofa_light', payload '{"state":"OFF","linkquality":69,"brightness":0,"update_available":false}'`

jschonenberg, 2020-07-20:
I'm experiencing the exact same issue. 
When toggling the light via LoveLace the previous brightness is remember when turning the light back on. 
When toggling via Homekit, the light always turns on to 100%. 
Furthermore the issue only occurs with Zigbee-connected lights.

Koenkk, 2020-07-20:
Can you both post your zigbee2mqtt `configuration.yaml`? Remove any sensitive data, I'm only interested wether a default `transition` is specified.

jschonenberg, 2020-07-20:
`data_path: /share/zigbee2mqtt
devices: devices.yaml
groups: groups.yaml
homeassistant: true
permit_join: false
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://core-mosquitto'
  user: <omitted>
  password: <omitted>
serial:
  disable_led: true
  port: >-
    /dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E1A183-if00
advanced:
  pan_id: 6754
  channel: 11
  network_key: <omitted>
  availability_timeout: 30
  availability_blacklist: []
ban: []
whitelist: []
queue: {}
socat:
  enabled: false
  master: 'pty,raw,echo=0,link=/dev/ttyZ2M,mode=777'
  slave: 'tcp-listen:8485,keepalive,nodelay,reuseaddr,keepidle=1,keepintvl=1,keepcnt=5'
  restartdelay: 1
  initialdelay: 1
  options: '-d -d'
  log: false
`

Koenkk, 2020-07-20:
@jschonenberg and your `devices.yml`?

jschonenberg, 2020-07-20:
Nothing special afaic;
````
'0x000d6ffffe1e52fd':
  friendly_name: Tradfri Remote Control
  retain: false
'0x90fd9ffffe14997a':
  friendly_name: Tradfri E27 bulb
  retain: false
'0x00158d00036bd78b':
  friendly_name: Aqara Temperature Sensor
  retain: false
'0x00158d0003132895':
  friendly_name: Aqara Vibration Sensor
  retain: false
'0x000b3cfffef0b5ba':
  friendly_name: Tradfri Outlet
'0x00178801062b3b4d':
  friendly_name: Hue Welcome
'0x00158d0003f24ca3':
  friendly_name: Aqara Motion Sensor
  occupancy_timeout: 60
'0x00158d00045c1483':
  friendly_name: Aqara Door Sensor
  retain: false
'0x00158d00044fba0c':
  friendly_name: '0x00158d00044fba0c'
'0xec1bbdfffee383f7':
  friendly_name: '0xec1bbdfffee383f7'
````

Koenkk, 2020-07-20:
Do the received `ON` mqtt messages from homekit contain brightness?

From https://github.com/Koenkk/zigbee2mqtt/issues/3934#issuecomment-660628085, the log shows the following for the on from homekit:

```
zigbee2mqtt:debug 2020-07-19 13:15:33: Received MQTT message on 'zigbee2mqtt/Sofa_light/set' with data '{"state": "ON", "brightness": 254}'
```

Here you see a brightness of 254 is requested.

jschonenberg, 2020-07-20:
Given the following scenario:
1. Turn on to +- 50%
2. Turn off
3. Turn on
4. Turn off

The log is as follows:
````
zigbee2mqtt:info  2020-07-20 21:09:43: MQTT publish: topic 'zigbee2mqtt/Tradfri E27 bulb', payload '{"state":"ON","brightness":144,"linkquality":39,"update_available":false}'
zigbee2mqtt:info  2020-07-20 21:09:50: MQTT publish: topic 'zigbee2mqtt/Tradfri E27 bulb', payload '{"state":"OFF","brightness":0,"linkquality":39,"update_available":false}'
zigbee2mqtt:info  2020-07-20 21:09:53: MQTT publish: topic 'zigbee2mqtt/Tradfri E27 bulb', payload '{"state":"ON","brightness":254,"linkquality":39,"update_available":false}'
zigbee2mqtt:info  2020-07-20 21:09:56: MQTT publish: topic 'zigbee2mqtt/Tradfri E27 bulb', payload '{"state":"OFF","brightness":0,"linkquality":39,"update_available":false}'
````

As you can see the second time the lamp is turned on, the brightness of 144 is not restored, but rather a brightness of 254 is used.

Koenkk, 2020-07-20:
You need to look at the `Received MQTT message on` log lines (that shows what is coming from homekit). This is only shown in the debug log.

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

jschonenberg, 2020-07-20:
HomeKit:
````
zigbee2mqtt:debug 2020-07-20 21:19:58: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "ON", "brightness": 112}'
zigbee2mqtt:debug 2020-07-20 21:20:00: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-07-20 21:20:01: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "ON", "brightness": 254}'
zigbee2mqtt:debug 2020-07-20 21:20:02: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "OFF"}'
````

vs

Lovelace:
````
zigbee2mqtt:debug 2020-07-20 21:21:57: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "ON", "brightness": 105}'
zigbee2mqtt:debug 2020-07-20 21:21:58: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "OFF"}'
zigbee2mqtt:debug 2020-07-20 21:22:00: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "ON"}'
zigbee2mqtt:debug 2020-07-20 21:22:00: Received MQTT message on 'zigbee2mqtt/Tradfri E27 bulb/set' with data '{"state": "OFF"}'
````

Key difference being the third line. While LoveLace just puts `"state": "ON"`, Homekit somehow also adds `"brightness": 254`. Seeing this I think the conclusion that this problem does not occur in Zigbee2Mqtt is correct.

Koenkk, 2020-07-20:
Good, I see you already updated https://github.com/home-assistant/core/issues/37955, I will close this.

closed 2020-07-20 by Koenkk

SalkinDK, 2020-07-20:
Thank you so much for answering and helping out, while i was away! @jschonenberg.
And of course @Koenkk