Koenkk/zigbee2mqtt #3935: MAC channel access failure and failed to ping after system disk replace

iStitch07, 2020-07-19:
Hello.
I replaced my system disk where installed z2m via docker (dev:latest). All persistent files of z2m was fully copied. After replace all worked like normal, but sometime not by first command. In the log I see a lot of messages like

```

zigbee2mqtt:error 2020-07-19 11:35:32: Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))',
zigbee2mqtt:error 2020-07-19 11:52:46: Failed to ping 'Living Room Main Light 4',
zigbee2mqtt:error 2020-07-19 11:56:37: Failed to read state of 'Living Room Main Light 7' after reconnect,
zigbee2mqtt:info  2020-07-19 11:35:32: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"Aqara Relay"}}',
zigbee2mqtt:error 2020-07-19 11:37:30: Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))',
zigbee2mqtt:info  2020-07-19 11:37:30: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"Aqara Relay"}}',
zigbee2mqtt:error 2020-07-19 11:41:19: Failed to read state of 'Living Room Main Light 7' after reconnect,
zigbee2mqtt:error 2020-07-19 11:39:44: Failed to ping 'Living Room Main Light 5',
zigbee2mqtt:error 2020-07-19 11:43:59: Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))',
zigbee2mqtt:info  2020-07-19 11:43:59: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"Aqara Relay"}}',
zigbee2mqtt:error 2020-07-19 11:45:24: Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))',
zigbee2mqtt:info  2020-07-19 11:45:24: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"Aqara Relay"}}',
zigbee2mqtt:error 2020-07-19 11:46:41: Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225))',
zigbee2mqtt:info  2020-07-19 11:46:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"zigbee_publish_error","message":"Publish 'set' 'state' to 'Aqara Relay' failed: 'Error: Command 0x00158d0003e82ee5/1 genOnOff.on({}, {\"timeout\":10000,\"disableResponse\":false,\"disableDefaultResponse\":false,\"direction\":0,\"srcEndpoint\":null,\"reservedBits\":0,\"manufacturerCode\":null,\"transactionSequenceNumber\":null}) failed (Data request failed with error: 'MAC channel access failure' (225))'","meta":{"friendly_name":"Aqara Relay"}}',
zigbee2mqtt:error 2020-07-19 11:48:43: Failed to read state of 'Living Room Main Light 3' after reconnect,
zigbee2mqtt:error 2020-07-19 11:51:25: Failed to ping 'Living Room Main Light 7'
```

How you think, it possible to fix without reconnect all my devices?

iStitch07, 2020-07-19:
additional error from logs:
```

zigbee2mqtt:info  2020-07-19 12:30:21: MQTT publish: topic 'zigbee2mqtt/Living Room Main Light 4/availability', payload 'online',
zigbee2mqtt:error 2020-07-19 12:30:21: Failed to setup reporting for '0xec1bbdfffe38ed03' - Error: Bind 0xec1bbdfffe38ed03/1 genOnOff from '0x00124b001caabbf8/1' failed (AREQ - ZDO - bindRsp after 10000ms),
    at Timeout._onTimeout (/app/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35),
    at listOnTimeout (internal/timers.js:549:17),
    at processTimers (internal/timers.js:492:7),
zigbee2mqtt:error 2020-07-19 12:30:27: Failed to setup reporting for '0x086bd7fffe0385a2' - Error: ConfigureReporting 0x086bd7fffe0385a2/1 genOnOff([{"attribute":"onOff","minimumReportInterval":0,"maximumReportInterval":300,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Data request failed with error: 'MAC channel access failure' (225)),
    at ZStackAdapter.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:655:27),
    at Generator.next (<anonymous>),
    at fulfilled (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/zStackAdapter.js:24:58)
```

VOVAiS, 2020-07-19:
turn off the power and turn on all zigbee devices. The network will be rebuilt

iStitch07, 2020-07-19:
I tried it. But its not help

Koenkk, 2020-07-20:
Is your adapter (CC2531?) connected through a USB extension cable?

iStitch07, 2020-07-20:
Yes, CC26X2R1 connected via extending cable. No more changes except system disk was made. 
Now I revert back my old disk and all now working fine

Koenkk, 2020-07-20:
Is it also position far away from the pc? (could it be that the new disk generates more interference?)

iStitch07, 2020-07-21:
Yes. At next weekends I will try again and try to reposition new disk far away from adapter

iStitch07, 2020-07-25:
Yes, it's like you were right. I parted my new disk and adapter as far as possible and now I do not have any errors.

Koenkk, 2020-07-25:
Great! I assume this can be closed now.

closed 2020-07-25 by Koenkk

hwacookie, 2021-02-17:
> Is your adapter (CC2531?) connected through a USB extension cable? This error happens when there is too much interference on the wireless channel

That fixed (at least one of) my problems. Thank you!

radix50, 2021-10-05:
It would be good to know the old and new disk types and models - especially if the new disk is a SSD so as to help others if the disk turns out to be the cause of the issue