Koenkk/zigbee2mqtt #3936: Problem to start npm

stefano099, 2020-07-19:
Hi community,
can someone tell me what this error is all about?
Is this error known and resolved?
first time it works but after rebooting nothing works!

> pi@raspberrypi:~ $ cd /opt/zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/lib/util/yaml.js:21
        throw error;
        ^
YAMLException: 

	====================================================================
	Your YAML file '/opt/zigbee2mqtt/data/configuration.yaml' is invalid
	Use e.g. https://jsonformatter.org/yaml-validator to find and fix the issue.
	====================================================================

duplicated mapping key at line 12, column 1:
    \n\nadvanced:\n    network_key:  ... 
    ^
    at generateError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:167:10)
    at throwError (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:173:9)
    at storeMappingPair (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:335:7)
    at readBlockMapping (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1098:9)
    at composeNode (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1359:12)
    at readDocument (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1525:3)
    at loadDocuments (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1588:5)
    at load (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1614:19)
    at Object.safeLoad (/opt/zigbee2mqtt/node_modules/js-yaml/lib/js-yaml/loader.js:1637:10)
    at Object.read (/opt/zigbee2mqtt/lib/util/yaml.js:7:21) {
  reason: 'duplicated mapping key',
  mark: Mark {
    name: null,
    buffer: 'homeassistant: false\n' +
      'permit_join: true\n' +
      'mqtt:\n' +
      '  base_topic: zigbee2mqtt\n' +
      "  server: 'mqtt://localhost'\n" +
      'serial:\n' +
      '  port: /dev/ttyACM0\n' +
      "'\\n\\nadvanced:\\n    network_key': GENERATE\n" +
      'devices:\n' +
      "  '0x00158d0002c9d15e':\n" +
      "    friendly_name: '0x00158d0002c9d15e'\n" +
      '\\n\\nadvanced:\\n    network_key: GENERATE\n' +
      '\u0000',
    position: 245,
    line: 11,
    column: 0
  }
}
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-19T10_37_26_744Z-debug.log

/opt/zigbee2mqtt/data/configuration.yaml
`  GNU nano 3.2                  /opt/zigbee2mqtt/data/configuration.yaml                              

homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial:
  port: /dev/ttyACM0
'\n\nadvanced:\n    network_key': GENERATE
devices:
  '0x00158d0002c9d15e':
    friendly_name: '0x00158d0002c9d15e'
\n\nadvanced:\n    network_key: GENERATE
`

thanks for any help and a nice Sunday

Tam-Lin, 2020-07-19:
You have duplicated keys, it appears.  You also have what should be new lines (represented by "\n") in the wrong format.  Those need to be actual new lines.  Did you do what the error message said, and try pasting your YAML into https://jsonformatter.org/yaml-validator?

stefano099, 2020-07-20:
thanks i made a fresh install and now i have another problem i made a new issue thanks for your help!

closed 2020-07-20 by stefano099