Koenkk/zigbee2mqtt #3940: Conbee Issue with LivingWise LVS-SC7

joe540, 2020-07-20:
# Bug Report
## What happened

I switched from a CC2531 to a Conbee II USB stick (to help with range issues. My Scenes controllers (LivingWise LVS-SC7, for which I helped to add support) stopped working. They are still recognized and seems to work as routers, but the button press are no longer received

I tried to reset the whole network multiple time. If I simply switch back to my CC2531 (with bad signal, ahah) it works again

## What did you expect to happen

With the CC2531, I see this in the log:
```
zigbee2mqtt:info  2020-07-20 00:25:13: MQTT publish: topic 'zigbee2mqtt/Basement Scene Controller', payload '{"linkquality":5,"action":"button_1_click"}'
```

With the Conbee II, there's nothing that appears in the log when a button is pressed. I enabled debug for both zigbee2mqtt and zigbee-shepherd, but there's nothing in the log when a button is pressed

## How to reproduce it (minimal and precise)

Use a Conbee II stick instead of a CC2531 (and enable conbee in config...)

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531 and Conbee II

Koenkk, 2020-07-20:
> but there's nothing in the log when a button is pressed

In that case the conbee may not receive the messages (and therefore not passing them to zigbee2mqtt). Have you tried repairing this device? (remove it first: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove)

joe540, 2020-07-20:
When I switched to the Conbee II, I started with a fresh zigbee2mqtt instal, including a different network key. I repaired the device 2 or 3 times but I did reset it locally, not via the remove command of zigbee2mqtt. I will try to remove it using the command tonight. 

So you think it might be due to the Conbee II firmware?

closed 2020-08-27 by github-actions[bot]

spurgelaurels, 2020-11-25:
Ah bummer this was closed. 

Having the same issue with an LVS-SC7. It worked dandy with my CC2531, but I moved to a ZZH CC2652. Initially, it paired, but it wasn't until a while later that I realized that my scene controllers weren't working. I checked it out and they were connected and reporting LQI, but they weren't reporting the button presses. 

Tonight I tried re-pairing one, along with a LVS-ZB500D, but neither of them will connect or pair and I don't see anything in my logs. 

**Not entirely sure if this is related but the only thing I see in my debug logs is this:** 
```Zigbee2MQTT:debug 2020-11-24 20:12:35: Error: Command 0x7cb03eaa00a494f8/3 genOnOff.off({}, {"timeout":10000,"disableResponse":false,"disableRecovery":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 665 - 3 - 220 - 6 - 11 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.16.1/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7)```

Assuming that 0x7cb is a zigbee device, it's not in my active device list.

spurgelaurels, 2020-11-25:
I should add that I'm running the latest 3.x.0 stack firmware as well