Koenkk/zigbee2mqtt #3941: Xiaomi aqara window / door contact

ochstobi, 2020-07-20:
# Bug Report
## What happened
Currently I have 9 devices (Iluminize 511.201 & 511.202) added to my network.
Now I got some Xiaomi Aqara door & window contact sensor (MCCGQ11LM). I can easily add them to the network and they will connect to the nearest router (Iluminize) and work for some minutes. But after some time the sensors are not sending messages to the network. I can see in the network map, that the devices are not longer connected to their parent devices and are "floating" around. 
[image]
I have no image of the connected devices but this shouldn't be a problem with the link quality, because the devices are in the same room und it showed a link quality >100

## How to reproduce it (minimal and precise)
Add Xiaomi Aqara door & window contact sensor (MCCGQ11LM) to an router (Iluminize 511.201 or 511.202) in the network.
After some time the Xiaomi stop sending messages to the network.

## Debug Info
Zigbee2mqtt version: 1.14.1 
Adapter hardware: CC2531
Adapter firmware version: {'type': 'zStack12', 'meta': {'transportrev': 2, 'product': 0, 'majorrel': 2, 'minorrel': 6, 'maintrel': 3, 'revision': 20190608}}

## Log file
Here is the log while joining the device.
https://hastebin.com/agenebakiz.apache

Koenkk, 2020-07-20:
Can you provide the zigbee herdsman debug logging when you expect messages to be send but nothing shows up in zigbee2qmtt?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

ochstobi, 2020-07-21:
Hello, I have started the container with the herdsman debug. The log is from startup and I have triggered the Windowsensor once. [https://hastebin.com/amuhifatuc.makefile](log).
Thanks for your help.

Koenkk, 2020-07-21:
I indeed see no messages being received by herdsman. To dive further into this we have to sniff the traffic, are you able to do this? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

ochstobi, 2020-07-22:
Do I need a second CC2531 Stick to do this? Or can I use one as coordinator and sniffer at the same time?
If I need a second one I have to order it first and maybe would wait some days to get a zzh! Stick (sold out at the moment) as herdsman.

Koenkk, 2020-07-22:
You need a second CC2531, no problem!

fluffymadness, 2020-07-28:
Maybe you ran into this specific issue. 
...
Xiaomi sensors - Xiaomi uses a very long sleep time. Many routers (smart plugs, bulbs, etc) on the market are not compatible with the Xiaomi as they will send a reassociation request when the Xiaomi wakes up. This in turn resets the Xiaomi sesnor and knocks it off the network. If you plan on using Xiaomi devices you need to carefully vet and test each router you add to the network to make sure they are compatible. Currently the best list of such devices is [here](https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623).
...
https://www.reddit.com/r/homeassistant/comments/eym1h8/state_of_zigbee_in_2020_zha_z2m_deconz/

Right now I only have some Xiaomi and Heiman plugs, and this is why I'm a bit hesistant to buy certain non-xiaomi plugs and router devices atm.

Koenkk, 2020-07-28:
@fluffymadness thanks, added this to the docs

ochstobi, 2020-07-29:
@fluffymadness this seems to be the problem in my network. I assume there is noting I can do about this, other than buying a router that can handle xiaomi sensors or other window sensors?
If this is a hardware problem we can't get a workaround or something this issue can be closed.

Thanks for your help

Koenkk, 2020-07-29:
@ochstobi it indeed seems to be a firmware issue (not so zigbee compliant) of the Xiaomi sensors. Since Xiaomi doesn't provide updates we cannot fix this. Only workaround would be to use a different router.

closed 2020-07-29 by Koenkk