Koenkk/zigbee2mqtt #3942: Battery devices being reported as offline by Availability

derekoharrow, 2020-07-20:
# Bug Report
## What happened

Zigbee2mqtt is reporting Battery devices as "offline" 

e.g. 

```
zigbee2mqtt/Kitchen_PIR = {"battery":100,"voltage":3025,"illuminance":1515,"illuminance_lux":15,"linkquality":0,"occupancy":false,"last_seen":"2020-07-16T20:33:17+01:00"}
zigbee2mqtt/Kitchen_PIR/availability = "offline"
```

## What did you expect to happen

Availability shouldn't be reported on Battery devices, as per the documentation. 

## How to reproduce it (minimal and precise)

configuration.yaml defined as follows:

```
  availability_timeout: 3600
  availability_blacklist:
    - Lux_Sensor
```

Battery devices not included in the availability_blacklist are being reported as Offline.

## Debug Info
Zigbee2mqtt version: 1.14.1-dev
Adapter hardware: CC2531
Adapter firmware version: 20190619 (source routing FW)
Platform: Raspberry Pi

Koenkk, 2020-07-20:
Non-pingable (battery powered) devices are marked as unavailable when they haven't send a single message in 25 hours (https://www.zigbee2mqtt.io/information/availability.html#non-pingable-devices).

derekoharrow, 2020-07-20:
They’re not showing as unavailable, they’re showing as offline. Does that issue a different mqtt message?

Koenkk, 2020-07-20:
With unavailable I meant offline, if you trigger the sensor it should mark itself `online` again.

closed 2020-08-27 by github-actions[bot]