Koenkk/zigbee2mqtt #3943: Error while starting zigbee-herdsman

stefano099, 2020-07-20:
hello yesterday had a problem that I was able to do by freshly installing the pis using RPI3 and Rasbian light. Now I have another problem, everything was fine, but
after rebooting I have this problem

`pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-07-20 12:35:35: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-07-20.12-35-35' filename: log.txt
zigbee2mqtt:info  2020-07-20 12:35:35: Starting zigbee2mqtt version 1.14.1 (commit #f26791a)
zigbee2mqtt:info  2020-07-20 12:35:36: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-07-20 12:35:36: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-07-20 12:35:36: Failed to start zigbee
zigbee2mqtt:error 2020-07-20 12:35:36: Exiting...
zigbee2mqtt:error 2020-07-20 12:35:36: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:123:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:121:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-20T11_35_36_540Z-debug.log
`

have now found out when I 

`pi@raspberrypi:/opt/zigbee2mqtt $ sudo systemctl stop zigbee2mqtt
pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-07-20 12:27:52: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-07-20.12-27-52' filename: log.txt
zigbee2mqtt:info  2020-07-20 12:27:53: Starting zigbee2mqtt version 1.14.1 (commit #f26791a)
zigbee2mqtt:info  2020-07-20 12:27:53: Starting zigbee-herdsman...
zigbee2mqtt:info  2020-07-20 12:27:55: zigbee-herdsman started
zigbee2mqtt:info  2020-07-20 12:27:55: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'
zigbee2mqtt:info  2020-07-20 12:27:55: Currently 2 devices are joined:
zigbee2mqtt:info  2020-07-20 12:27:55: 0x00158d0002ca0960 (0x00158d0002ca0960): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:info  2020-07-20 12:27:55: 0x00158d0002c9d15e (0x00158d0002c9d15e): WSDCGQ11LM - Xiaomi Aqara temperature, humidity and pressure sensor (EndDevice)
zigbee2mqtt:warn  2020-07-20 12:27:55: `permit_join` set to  `true` in configuration.yaml.
zigbee2mqtt:warn  2020-07-20 12:27:55: Allowing new devices to join.
zigbee2mqtt:warn  2020-07-20 12:27:55: Set `permit_join` to `false` once you joined all devices.
zigbee2mqtt:info  2020-07-20 12:27:55: Zigbee: allowing new devices to join.
zigbee2mqtt:info  2020-07-20 12:27:55: Connecting to MQTT server at mqtt://localhost
zigbee2mqtt:info  2020-07-20 12:27:55: Connected to MQTT server
zigbee2mqtt:info  2020-07-20 12:27:55: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'online'
zigbee2mqtt:info  2020-07-20 12:27:55: MQTT publish: topic 'zigbee2mqtt/0x00158d0002ca0960', payload '{"battery":100,"voltage":3035,"temperature":27.58,"humidity":36.52,"pressure":992.6,"linkquality":120}'
zigbee2mqtt:info  2020-07-20 12:27:55: MQTT publish: topic 'zigbee2mqtt/0x00158d0002c9d15e', payload '{"battery":100,"voltage":3055,"temperature":27.04,"humidity":42.96,"pressure":992.4,"linkquality":97}'
zigbee2mqtt:info  2020-07-20 12:27:55: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.1","commit":"f26791a","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"info","permit_join":true}'

zigbee2mqtt:info  2020-07-20 12:31:16: MQTT publish: topic 'zigbee2mqtt/0x00158d0002ca0960', payload '{"battery":100,"voltage":3035,"temperature":27.08,"humidity":36.52,"pressure":992.6,"linkquality":120}'
zigbee2mqtt:info  2020-07-20 12:31:16: MQTT publish: topic 'zigbee2mqtt/0x00158d0002ca0960', payload '{"battery":100,"voltage":3035,"temperature":27.08,"humidity":36.49,"pressure":992.6,"linkquality":120}'
zigbee2mqtt:info  2020-07-20 12:31:16: MQTT publish: topic 'zigbee2mqtt/0x00158d0002ca0960', payload '{"battery":100,"voltage":3035,"temperature":27.08,"humidity":36.49,"pressure":992.5,"linkquality":120}'
^Czigbee2mqtt:info  2020-07-20 12:34:54: MQTT publish: topic 'zigbee2mqtt/bridge/state', payload 'offline'
zigbee2mqtt:info  2020-07-20 12:34:54: Disconnecting from MQTT server
zigbee2mqtt:info  2020-07-20 12:34:54: zigbee-herdsman stopped

` it runs? but this is not correct? thanks for any help and my bad english...

Koenkk, 2020-07-20:
I've added an additional section to the FAQ about this error: https://www.zigbee2mqtt.io/information/FAQ.html#error-resource-temporarily-unavailable-cannot-lock-port

Hope it helps.

stefano099, 2020-07-21:
hi thanks for you help first i do this 

pi@raspberrypi:~ $ ls -l /dev/ttyACM0
crw-rw---- 1 root dialout 166, 0 Jul 20 20:17 /dev/ttyACM0

than i do this

ls -l /proc/[0-9]/fd/ |grep /dev/ttyACM0

reboot it but nothing changes

pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-07-21 10:52:32: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-07-21.10-52-31' filename: log.txt
zigbee2mqtt:info  2020-07-21 10:52:32: Starting zigbee2mqtt version 1.14.1 (commit #f26791a)
zigbee2mqtt:info  2020-07-21 10:52:32: Starting zigbee-herdsman...
zigbee2mqtt:error 2020-07-21 10:52:33: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-07-21 10:52:33: Failed to start zigbee
zigbee2mqtt:error 2020-07-21 10:52:33: Exiting...
zigbee2mqtt:error 2020-07-21 10:52:33: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:123:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:121:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-21T08_52_33_227Z-debug.log

the i see in your link obove 

pi@raspberrypi:~ $ test -w [PORT] && echo success || echo failure
failure
pi@raspberrypi:~ $ sudo chown pi /dev/ttyACM0
pi@raspberrypi:~ $ udevadm info -a -n /dev/ttyACM0 | grep 'serial'
    ATTRS{serial}=="__0X00124B0014D9AF63"
    ATTRS{serial}=="3f980000.usb"

 changes looks like  sudo nano /etc/udev/rules.d/99-usb-serial.rule
`SUBSYSTEM=="tty", ATTRS{idVendor}=="0451", ATTRS{idProduct}=="16a8", ATTRS{serial}=="__0X00124B0014D9AF63", SYMLINK="ttyUSB.CC2531-01", OWNER="pi"
`
changes my nano /opt/zigbee2mqtt/data/configuration.yaml to

`homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial: port: /dev/ttyUSB.CC2531-01
'\n\nadvanced:\n    network_key': GENERATE
devices:
  '0x00158d0002ca0960':
    friendly_name: '0x00158d0002ca0960'
  '0x00158d0002c9d15e':
    friendly_name: '0x00158d0002c9d15e'

than i made this 
[... 74 lines cut ...]
9 verbose lifecycle zigbee2mqtt@1.14.1~start: CWD: /opt/zigbee2mqtt
10 silly lifecycle zigbee2mqtt@1.14.1~start: Args: [ '-c', 'node index.js' ]
11 silly lifecycle zigbee2mqtt@1.14.1~start: Returned: code: 1  signal: null
12 info lifecycle zigbee2mqtt@1.14.1~start: Failed to exec start script
13 verbose stack Error: zigbee2mqtt@1.14.1 start: `node index.js`
13 verbose stack Exit status 1
13 verbose stack     at EventEmitter.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/index.js:332:16)
13 verbose stack     at EventEmitter.emit (events.js:315:20)
13 verbose stack     at ChildProcess.<anonymous> (/usr/lib/node_modules/npm/node_modules/npm-lifecycle/lib/spawn.js:55:14)
13 verbose stack     at ChildProcess.emit (events.js:315:20)
13 verbose stack     at maybeClose (internal/child_process.js:1021:16)
13 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:286:5)
14 verbose pkgid zigbee2mqtt@1.14.1
15 verbose cwd /opt/zigbee2mqtt
16 verbose Linux 5.4.51-v7+
17 verbose argv "/usr/bin/node" "/usr/bin/npm" "start"
18 verbose node v12.18.2
19 verbose npm  v6.14.5
20 error code ELIFECYCLE
21 error errno 1
22 error zigbee2mqtt@1.14.1 start: `node index.js`
22 error Exit status 1
23 error Failed at the zigbee2mqtt@1.14.1 start script.
23 error This is probably not a problem with npm. There is likely additional logging output above.
24 verbose exit [ 1, true ]

Koenkk, 2020-07-21:
Your `configuration.yaml` is not formatted correctly, use http://www.yamllint.com/ to find the error.

stefano099, 2020-07-21:
ok thanks thats works but now i have another problem
npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-07-21 20:06:50: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-07-21.20-06-49' filename: log.txt
zigbee2mqtt:error 2020-07-21 20:06:50: Refusing to start, configuration.yaml is not valid, found the following errors:
zigbee2mqtt:error 2020-07-21 20:06:50: 	 - serial should be object
zigbee2mqtt:error 2020-07-21 20:06:50: If you don't know how to solve this, read https://www.zigbee2mqtt.io/configuration/configuration.html
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-21T18_06_50_651Z-debug.log

stefano099, 2020-07-21:
but this link did not work [https://www.zigbee2mqtt.io/configuration/configuration.html](url)

stefano099, 2020-07-21:
so looks lik my configuration.yamal 
`
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial: port:/dev/ttyUSB.CC2531-01
'\n\nadvanced:\n    network_key': GENERATE
devices:
  '0x00158d0002ca0960':
    friendly_name: '0x00158d0002ca0960'
  '0x00158d0002c9d15e':
    friendly_name: '0x00158d0002c9d15e'

`
is this correct so? and thanks for your quick awnser

Koenkk, 2020-07-21:
No, the indentation matter, use:

```yaml
homeassistant: false
permit_join: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://localhost'
serial: 
  port: /dev/ttyUSB.CC2531-01
advanced:
  network_key: GENERATE
devices:
  '0x00158d0002ca0960':
    friendly_name: '0x00158d0002ca0960'
  '0x00158d0002c9d15e':
    friendly_name: '0x00158d0002c9d15e'
```

stefano099, 2020-07-22:
thanks for your help but now i have the next problem

npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

/opt/zigbee2mqtt/lib/util/logger.js:12
const output = settings.get().advanced.log_output;
                                      ^

TypeError: Cannot read property 'log_output' of null
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/util/logger.js:12:39)
    at Module._compile (internal/modules/cjs/loader.js:1138:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1158:10)
    at Module.load (internal/modules/cjs/loader.js:986:32)
    at Function.Module._load (internal/modules/cjs/loader.js:879:14)
    at Module.require (internal/modules/cjs/loader.js:1026:19)
    at require (internal/modules/cjs/helpers.js:72:18)
    at Object.<anonymous> (/opt/zigbee2mqtt/lib/mqtt.js:2:16)
    at Module._compile (internal/modules/cjs/loader.js:1138:30)
    at Object.Module._extensions..js (internal/modules/cjs/loader.js:1158:10)
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-22T06_09_59_057Z-debug.log

Koenkk, 2020-07-23:
Your indentation is still not correct, make sure to put 2 spaces before `network_key` after `advanced:` (like shown in https://github.com/Koenkk/zigbee2mqtt/issues/3943#issuecomment-662032849).

I will improve the error message.

stefano099, 2020-07-23:
ok thanks so i have the same problem as befor, no i made this to the configuration.yaml `advanced:
  log_level: debug`
`pi@raspberrypi:/opt/zigbee2mqtt $ DEBUG=zigbee-herdsman* npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-07-23 17:37:10: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-07-23.17-37-10' filename: log.txt
zigbee2mqtt:debug 2020-07-23 17:37:10: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-07-23.15-56-13'
zigbee2mqtt:debug 2020-07-23 17:37:11: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-07-23 17:37:11: Starting zigbee2mqtt version 1.14.1 (commit #f26791a)
zigbee2mqtt:info  2020-07-23 17:37:11: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-07-23 17:37:11: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB.CC2531-01"},"adapter":{"concurrent":null}}'
  zigbee-herdsman:adapter Path '/dev/ttyUSB.CC2531-01' is valid for 'ZStackAdapter' +0ms
  zigbee-herdsman:controller:log Starting with options '{"network":{"networkKeyDistribute":false,"networkKey":[1,3,5,7,9,11,13,15,0,2,4,6,8,10,12,13],"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11]},"serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB.CC2531-01"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","adapter":{"concurrent":null}}' +0ms
  zigbee-herdsman:adapter:zStack:znp:log Opening SerialPort with /dev/ttyUSB.CC2531-01 and {"baudRate":115200,"rtscts":true,"autoOpen":false} +0ms
zigbee2mqtt:error 2020-07-23 17:37:12: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-07-23 17:37:12: Failed to start zigbee
zigbee2mqtt:error 2020-07-23 17:37:12: Exiting...
zigbee2mqtt:error 2020-07-23 17:37:12: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:123:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:121:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.1 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.1 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.

npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-23T15_37_12_148Z-debug.log

pi@raspberrypi:/opt/zigbee2mqtt $ npm start

> zigbee2mqtt@1.14.1 start /opt/zigbee2mqtt
> node index.js

zigbee2mqtt:info  2020-07-23 17:38:19: Logging to console and directory: '/opt/zigbee2mqtt/data/log/2020-07-23.17-38-19' filename: log.txt
zigbee2mqtt:debug 2020-07-23 17:38:19: Removing old log directory '/opt/zigbee2mqtt/data/log/2020-07-23.15-57-27'
zigbee2mqtt:debug 2020-07-23 17:38:19: Loaded state from file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:info  2020-07-23 17:38:20: Starting zigbee2mqtt version 1.14.1 (commit #f26791a)
zigbee2mqtt:info  2020-07-23 17:38:20: Starting zigbee-herdsman...
zigbee2mqtt:debug 2020-07-23 17:38:20: Using zigbee-herdsman with settings: '{"network":{"panID":6754,"extendedPanID":[221,221,221,221,221,221,221,221],"channelList":[11],"networkKey":"HIDDEN"},"databasePath":"/opt/zigbee2mqtt/data/database.db","databaseBackupPath":"/opt/zigbee2mqtt/data/database.db.backup","backupPath":"/opt/zigbee2mqtt/data/coordinator_backup.json","serialPort":{"baudRate":115200,"rtscts":true,"path":"/dev/ttyUSB.CC2531-01"},"adapter":{"concurrent":null}}'
zigbee2mqtt:error 2020-07-23 17:38:20: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-07-23 17:38:20: Failed to start zigbee
zigbee2mqtt:error 2020-07-23 17:38:20: Exiting...
zigbee2mqtt:error 2020-07-23 17:38:20: Error: Error while opening serialport 'Error: Error Resource temporarily unavailable Cannot lock port'
    at Znp.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:123:32)
    at Generator.next (<anonymous>)
    at /opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:8:71
    at new Promise (<anonymous>)
    at __awaiter (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:4:12)
    at SerialPort.<anonymous> (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:121:49)
    at SerialPort._error (/opt/zigbee2mqtt/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
[... cut ...]
npm ERR! A complete log of this run can be found in:
npm ERR!     /home/pi/.npm/_logs/2020-07-23T15_38_20_605Z-debug.log
`

thats my configuration.yaml

> ```
> homeassistant: false
> permit_join: true
> mqtt:
>   base_topic: zigbee2mqtt
>   server: 'mqtt://localhost'
> serial: 
>   port: /dev/ttyUSB.CC2531-01
>   advanced:
>   log_level: debug
>   network_key: GENERATE
> devices:
>   '0x00158d0002ca0960':
>     friendly_name: '0x00158d0002ca0960'
>   '0x00158d0002c9d15e':
>     friendly_name: '0x00158d0002c9d15e'
> ```
 is this so ok?

stefano099, 2020-07-23:
Excuse me for I don't have that much experience

Koenkk, 2020-07-23:
Yes looks good, now you have the same error as https://github.com/Koenkk/zigbee2mqtt/issues/3943#issuecomment-661146329 again (so please read that carefully).

For future support questions please use one of the support channels ([Forum](https://zigbee2mqtt.discourse.group/) or [Discord](https://discord.com/invite/NyseBeK)). Note that the issue tracker is meant to track bugs, not user setup errors.

closed 2020-07-23 by Koenkk