Koenkk/zigbee2mqtt #3944: Xiaomi Aqara QBKG11LM switch drops connection

derekoharrow, 2020-07-20:
I have numerous Zigbee devices connected to my Pi running Z2M using a CC2531

Most work fine, but I have a Xiaomi Aqara QBKG11LM light switch which works for a while when paired, but after a while stops responding and I have to re-pair again. This is problematic as this device seems to be the main router node for many of my devices (despite having other routers nearby - an Ikea Tradfri repeater and an Innr Flex LED controller).

Is there any way to tell why this keeps happening, or to prevent it?

Thanks

Zigbee2mqtt version:  1.14.1-dev
Adapter hardware: CC2531
Adapter firmware version: 20190619 Source Routing

Koenkk, 2020-07-20:
Xiaomi devices are known to drop when linkquality is bad, what is the linkquality? The linkquality is included in the published payload when e.g. turning the switch on/off.

derekoharrow, 2020-07-21:
Thanks - just tried that and it says:

```
debug 2020-07-21 09:55:35: Received Zigbee message from 'Cabin_Lights', type 'attributeReport', cluster 'genOnOff', data '{"61440":63810560,"onOff":0}' from endpoint 1 with groupID 0
info  2020-07-21 09:55:35: MQTT publish: topic 'zigbee2mqtt/Cabin_Lights', payload '{"power":98.79,"consumption":41.53,"temperature":25,"linkquality":0,"state":"OFF","last_seen":"2020-07-21T09:55:35+01:00"}'
debug 2020-07-21 09:55:35: Received Zigbee message from 'Cabin_Lights', type 'attributeReport', cluster 'genAnalogInput', data '{"presentValue":0}' from endpoint 2 with groupID 0
info  2020-07-21 09:55:35: MQTT publish: topic 'zigbee2mqtt/Cabin_Lights', payload '{"power":0,"consumption":41.53,"temperature":25,"linkquality":0,"state":"OFF","last_seen":"2020-07-21T09:55:35+01:00"}'
debug 2020-07-21 09:55:35: Received Zigbee message from 'Cabin_Lights', type 'attributeReport', cluster 'genMultistateInput', data '{"presentValue":1}' from endpoint 5 with groupID 0
info  2020-07-21 09:55:35: MQTT publish: topic 'zigbee2mqtt/Cabin_Lights', payload '{"power":0,"consumption":41.53,"temperature":25,"linkquality":0,"state":"OFF","click":"single","last_seen":"2020-07-21T09:55:35+01:00"}'
```

So, Linkquality=0? (but it is working currently).

Koenkk, 2020-07-21:
What does your networkmap say? (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgenetworkmap)

derekoharrow, 2020-07-22:
It says:

(Problem device is "ieeeAddr":"0x00158d0002ec1d9e" / "friendlyName":"Cabin_Lights")

`{"nodes":[{"ieeeAddr":"0x00124b000e0eaa6f","friendlyName":"Coordinator","type":"Coordinator","networkAddress":0,"failed":[],"lastSeen":1594901351283},{"ieeeAddr":"0x04cf8cdf3c771bea","friendlyName":"Lux_Sensor","type":"EndDevice","networkAddress":29858,"manufacturerName":"LUMI","modelID":"lumi.sen_ill.mgl01","lastSeen":1594921008344},{"ieeeAddr":"0x00158d0004484828","friendlyName":"Kitchen_PIR","type":"EndDevice","networkAddress":46431,"manufacturerName":"LUMI","modelID":"lumi.sensor_motion.aq2","lastSeen":1594927997405},{"ieeeAddr":"0x00124b001e721f58","friendlyName":"Ryans_Blanket","type":"Router","networkAddress":7616,"manufacturerName":"SONOFF","modelID":"BASICZBR3","failed":[],"lastSeen":1595405238189},{"ieeeAddr":"0x00124b001e7286ac","friendlyName":"Laurens_Blanket","type":"Router","networkAddress":50941,"manufacturerName":"SONOFF","modelID":"BASICZBR3","failed":[],"lastSeen":1595405192759},{"ieeeAddr":"0x00124b001e73367e","friendlyName":"Dereks_Blanket","type":"Router","networkAddress":49658,"manufacturerName":"SONOFF","modelID":"BASICZBR3","failed":[],"lastSeen":1595405250472},{"ieeeAddr":"0x00124b001e728d41","friendlyName":"Lindseys_Blanket","type":"Router","networkAddress":23617,"manufacturerName":"SONOFF","modelID":"BASICZBR3","failed":[],"lastSeen":1595405336027},{"ieeeAddr":"0x000d6ffffe2ee6a9","friendlyName":"IkeaRepeater01","type":"Router","networkAddress":26760,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI signal repeater","failed":[],"lastSeen":1595404922730},{"ieeeAddr":"0x00158d00044ad1ba","friendlyName":"Bedroom_Button1","type":"EndDevice","networkAddress":11751,"manufacturerName":"LUMI","modelID":"lumi.sensor_switch.aq3","lastSeen":1595403163828},{"ieeeAddr":"0x00158d0003440f7c","friendlyName":"Cabin_LEDs","type":"Router","networkAddress":57293,"manufacturerName":"innr","modelID":"FL 130 C","failed":[],"lastSeen":1595405316936},{"ieeeAddr":"0x00158d0004601611","friendlyName":"Cabin_TempHumid","type":"EndDevice","networkAddress":38558,"manufacturerName":"LUMI","modelID":"lumi.weather","lastSeen":1595404443373},{"ieeeAddr":"0xccccccfffe3cc886","friendlyName":"Cabin_Heater","type":"Router","networkAddress":16285,"manufacturerName":"LDS","modelID":"ZB-ONOFFPlug-D0005","failed":["lqi"],"lastSeen":1595404950818},{"ieeeAddr":"0x00158d0002ec1d9e","friendlyName":"Cabin_Lights","type":"Router","networkAddress":52652,"manufacturerName":"LUMI","modelID":"lumi.ctrl_ln1.aq1","failed":[],"lastSeen":1595405194919},{"ieeeAddr":"0x00124b001e732aed","friendlyName":"TV_LEDs","type":"Router","networkAddress":34658,"manufacturerName":"SONOFF","modelID":"BASICZBR3","failed":[],"lastSeen":1595405096812},{"ieeeAddr":"0x00124b001e722204","friendlyName":"Porch_Lights","type":"Router","networkAddress":32224,"manufacturerName":"SONOFF","modelID":"BASICZBR3","failed":[],"lastSeen":1595405274003},{"ieeeAddr":"0x001e5e0902145d6d","friendlyName":"Tumble_Dryer","type":"Router","networkAddress":8758,"manufacturerName":"Computime","modelID":"SP600","failed":[],"lastSeen":1595404770450},{"ieeeAddr":"0x001e5e0902145f59","friendlyName":"Washing_Machine","type":"Router","networkAddress":14843,"manufacturerName":"Computime","modelID":"SP600","failed":["lqi","routingTable"],"lastSeen":1595405165084},{"ieeeAddr":"0xccccccfffe381d7c","friendlyName":"Dishwasher","type":"Router","networkAddress":21156,"manufacturerName":"LDS","modelID":"ZB-ONOFFPlug-D0005","failed":[],"lastSeen":1595405001429},{"ieeeAddr":"0x90fd9ffffe2b589d","friendlyName":"IkeaRepeater02","type":"Router","networkAddress":41588,"manufacturerName":"IKEA of Sweden","modelID":"TRADFRI signal repeater","failed":[],"lastSeen":1595405069090},{"ieeeAddr":"0x00158d000429e66a","friendlyName":"Cabin_Button","type":"EndDevice","networkAddress":51996,"manufacturerName":"LUMI","modelID":"lumi.sensor_switch.aq3","lastSeen":1595403654633},{"ieeeAddr":"0x00158d0002582ad6","friendlyName":"Bedroom_Button2","type":"EndDevice","networkAddress":32303,"manufacturerName":"LUMI","modelID":"lumi.sensor_switch.aq3","lastSeen":1595403245633},{"ieeeAddr":"0x00158d00045cff29","friendlyName":"Unused_Door_Window_Sensor","type":"EndDevice","networkAddress":11453,"manufacturerName":"LUMI","modelID":"lumi.sensor_magnet.aq2","lastSeen":1595403138752},{"ieeeAddr":"0x00158d00045ed50a","friendlyName":"Cabin_Door","type":"EndDevice","networkAddress":28563,"manufacturerName":"LUMI","modelID":"lumi.sensor_magnet.aq2","lastSeen":1595403565251},{"ieeeAddr":"0x00158d0003f3c797","friendlyName":"Cabin_PIR","type":"EndDevice","networkAddress":28386,"manufacturerName":"LUMI","modelID":"lumi.sensor_motion.aq2","lastSeen":1595405328444},{"ieeeAddr":"0x00158d00045c5c55","friendlyName":"Front Door","type":"EndDevice","networkAddress":23612,"manufacturerName":"LUMI","modelID":"lumi.sensor_magnet.aq2","lastSeen":1595403444989},{"ieeeAddr":"0x00158d00045ed4f0","friendlyName":"Letterbox","type":"EndDevice","networkAddress":25663,"manufacturerName":"LUMI","modelID":"lumi.sensor_magnet.aq2","lastSeen":1595403834085}],"links":[{"source":{"ieeeAddr":"0x00124b001e721f58","networkAddress":7616},"target":{"ieeeAddr":"0x00124b000e0eaa6f","networkAddress":0},"linkquality":1,"depth":1,"routes":[{"destinationAddress":7616,"status":"ACTIVE","nextHop":7616},{"destinationAddress":34658,"status":"ACTIVE","nextHop":7616}],"sourceIeeeAddr":"0x00124b001e721f58","targetIeeeAddr":"0x00124b000e0eaa6f","sourceNwkAddr":7616,"lqi [... cut ...] 367e","networkAddress":49658},"linkquality":20,"depth":1,"routes":[{"destinationAddress":7616,"status":"ACTIVE","nextHop":7616},{"destinationAddress":23617,"status":"ACTIVE","nextHop":7616},{"destinationAddress":21156,"status":"ACTIVE","nextHop":7616},{"destinationAddress":41588,"status":"ACTIVE","nextHop":7616},{"destinationAddress":57293,"status":"ACTIVE","nextHop":7616},{"destinationAddress":26760,"status":"ACTIVE","nextHop":7616},{"destinationAddress":52652,"status":"ACTIVE","nextHop":7616}`

Koenkk, 2020-07-23:
Sorry for not mentioning, but could you provide the `graphviz` version? (https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#graphviz)

derekoharrow, 2020-07-24:
Here you go... As you can see in this one, the Cabin_Lights device says "Failed", which probably means I need to re-pair it yet again...

```
digraph G {
node[shape=record];
  "0x00124b000e0eaa6f" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b000e0eaa6f (0)|0 seconds ago}"];
  "0x00124b000e0eaa6f" -> "0x00124b001e721f58" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b000e0eaa6f" -> "0x00124b001e7286ac" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b000e0eaa6f" -> "0x00124b001e73367e" [penwidth=0.5, weight=0, color="#994444", label="3"]
  "0x00124b000e0eaa6f" -> "0x00124b001e728d41" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b000e0eaa6f" -> "0x000d6ffffe2ee6a9" [penwidth=0.5, weight=0, color="#994444", label="47"]
  "0x00124b000e0eaa6f" -> "0x00124b001e732aed" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x00124b000e0eaa6f" -> "0x00124b001e722204" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x04cf8cdf3c771bea" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Lux_Sensor|0x04cf8cdf3c771bea (29858)|Xiaomi MiJia light intensity sensor (GZCGQ01LM)|1 week, 1 day ago}"];
  "0x00158d0004484828" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Kitchen_PIR|0x00158d0004484828 (46431)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|1 week, 1 day ago}"];
  "0x00124b001e721f58" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Ryans_Blanket|0x00124b001e721f58 (7616)|SONOFF Zigbee smart switch (BASICZBR3)|2 minutes ago}"];
  "0x00124b001e721f58" -> "0x00124b000e0eaa6f" [penwidth=2, weight=1, color="#009900", label="1 (routes: 16285)"]
  "0x00124b001e721f58" -> "0x00124b001e7286ac" [penwidth=2, weight=1, color="#009900", label="1 (routes: 49658)"]
  "0x00124b001e721f58" -> "0x00124b001e73367e" [penwidth=2, weight=1, color="#009900", label="32 (routes: 7616,21156,41588,57293,26760,52652,14843,32224)"]
  "0x00124b001e721f58" -> "0x00124b001e728d41" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e721f58" -> "0x000d6ffffe2ee6a9" [penwidth=2, weight=1, color="#009900", label="168 (routes: 7616)"]
  "0x00124b001e721f58" -> "0x00124b001e732aed" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e721f58" -> "0x00124b001e722204" [penwidth=2, weight=1, color="#009900", label="1 (routes: 49658)"]
  "0x00124b001e721f58" -> "0x001e5e0902145d6d" [penwidth=2, weight=1, color="#009900", label="254 (routes: 23617,7616)"]
  "0x00124b001e721f58" -> "0x001e5e0902145f59" [penwidth=0.5, weight=0, color="#994444", label="252"]
  "0x00124b001e721f58" -> "0xccccccfffe381d7c" [penwidth=2, weight=1, color="#009900", label="116 (routes: 7616,49658)"]
  "0x00124b001e7286ac" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Laurens_Blanket|0x00124b001e7286ac (50941)|SONOFF Zigbee smart switch (BASICZBR3)|2 minutes, 48 seconds ago}"];
  "0x00124b001e7286ac" -> "0x00124b000e0eaa6f" [penwidth=2, weight=1, color="#009900", label="1 (routes: 50941)"]
  "0x00124b001e7286ac" -> "0x00124b001e721f58" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e7286ac" -> "0x00124b001e73367e" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x00124b001e7286ac" -> "0x00124b001e728d41" [penwidth=2, weight=1, color="#009900", label="7 (routes: 50941)"]
  "0x00124b001e7286ac" -> "0x000d6ffffe2ee6a9" [penwidth=0.5, weight=0, color="#994444", label="48"]
  "0x00124b001e7286ac" -> "0x00124b001e732aed" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e7286ac" -> "0x00124b001e722204" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e7286ac" -> "0x001e5e0902145d6d" [penwidth=0.5, weight=0, color="#994444", label="253"]
  "0x00124b001e7286ac" -> "0x001e5e0902145f59" [penwidth=2, weight=1, color="#009900", label="254 (routes: 50941)"]
  "0x00124b001e73367e" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Dereks_Blanket|0x00124b001e73367e (49658)|SONOFF Zigbee smart switch (BASICZBR3)|1 minute, 50 seconds ago}"];
  "0x00124b001e73367e" -> "0x00124b000e0eaa6f" [penwidth=2, weight=1, color="#009900", label="12 (routes: 34658,32224,49658,7616,14843,26760,57293)"]
  "0x00124b001e73367e" -> "0x00124b001e721f58" [penwidth=2, weight=1, color="#009900", label="30 (routes: 49658)"]
  "0x00124b001e73367e" -> "0x00124b001e7286ac" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x00124b001e73367e" -> "0x00124b001e728d41" [penwidth=2, weight=1, color="#009900", label="2 (routes: 49658,52652)"]
  "0x00124b001e73367e" -> "0x000d6ffffe2ee6a9" [penwidth=0.5, weight=0, color="#994444", label="71"]
  "0x00124b001e73367e" -> "0xccccccfffe3cc886" [penwidth=2, weight=1, color="#009900", label="51 (routes: 49658)"]
  "0x00124b001e73367e" -> "0x00124b001e732aed" [penwidth=2, weight=1, color="#009900", label="1 (routes: 49658)"]
  "0x00124b001e73367e" -> "0x00124b001e722204" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x00124b001e73367e" -> "0x001e5e0902145d6d" [penwidth=0.5, weight=0, color="#994444", label="253"]
  "0x00124b001e73367e" -> "0x001e5e0902145f59" [penwidth=0.5, weight=0, color="#994444", label="163"]
  "0x00124b001e73367e" -> "0xccccccfffe381d7c" [penwidth=0.5, weight=0, color="#994444", label="44"]
[... cut ...]
  "0x00158d00045ed4f0" -> "0x00124b001e722204" [penwidth=1, weight=0, color="#994444", label="49"]
  "0x90fd9ffffe2b589d" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{IkeaRepeater02|0x90fd9ffffe2b589d (19952)failed: lqi,routingTable|undefined undefined|20 hours, 54 minutes ago}"];
  "0x90fd9ffffe2b589d" -> "0x00124b001e73367e" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x90fd9ffffe2b589d" -> "0x00124b001e722204" [penwidth=0.5, weight=0, color="#994444", label="0"]
}
```

Koenkk, 2020-07-24:
Can you repair it and provide the same again? The lqi request failed so there is no linkquality information.

derekoharrow, 2020-07-24:
No problem - see below.

Also, just noticed after I repaired that I am getting messages for that device again - got these two one right after the other and I notice one has Link Qual of 5 and the other 0.

```
info  2020-07-24 16:38:00: MQTT publish: topic 'zigbee2mqtt/Cabin_Lights', payload '{"power":0,"consumption":43.43,"temperature":34,"linkquality":5,"state":"OFF","last_seen":"2020-07-24T16:38:00+01:00"}'
info  2020-07-24 16:38:00: MQTT publish: topic 'zigbee2mqtt/Cabin_Lights', payload '{"power":0,"consumption":43.43,"temperature":34,"linkquality":0,"state":"OFF","last_seen":"2020-07-24T16:38:00+01:00"}'
```

```
digraph G {
node[shape=record];
  "0x00124b000e0eaa6f" [style="bold, filled", fillcolor="#e04e5d", fontcolor="#ffffff", label="{Coordinator|0x00124b000e0eaa6f (0)|0 seconds ago}"];
  "0x00124b000e0eaa6f" -> "0x00124b001e721f58" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b000e0eaa6f" -> "0x00124b001e7286ac" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b000e0eaa6f" -> "0x00124b001e73367e" [penwidth=0.5, weight=0, color="#994444", label="2"]
  "0x00124b000e0eaa6f" -> "0x00124b001e728d41" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b000e0eaa6f" -> "0x000d6ffffe2ee6a9" [penwidth=0.5, weight=0, color="#994444", label="34"]
  "0x00124b000e0eaa6f" -> "0x00124b001e732aed" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x00124b000e0eaa6f" -> "0x00124b001e722204" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x04cf8cdf3c771bea" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Lux_Sensor|0x04cf8cdf3c771bea (29858)|Xiaomi MiJia light intensity sensor (GZCGQ01LM)|1 week, 1 day ago}"];
  "0x00158d0004484828" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{Kitchen_PIR|0x00158d0004484828 (46431)|Xiaomi Aqara human body movement and illuminance sensor (RTCGQ11LM)|1 week, 1 day ago}"];
  "0x00124b001e721f58" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Ryans_Blanket|0x00124b001e721f58 (7616)|SONOFF Zigbee smart switch (BASICZBR3)|1 minute, 2 seconds ago}"];
  "0x00124b001e721f58" -> "0x00124b000e0eaa6f" [penwidth=2, weight=1, color="#009900", label="4 (routes: 7616,8758,21156)"]
  "0x00124b001e721f58" -> "0x00124b001e7286ac" [penwidth=2, weight=1, color="#009900", label="4 (routes: 23617,7616)"]
  "0x00124b001e721f58" -> "0x00124b001e73367e" [penwidth=2, weight=1, color="#009900", label="32 (routes: 7616,16285,23617,57293,50941,32224)"]
  "0x00124b001e721f58" -> "0x00124b001e728d41" [penwidth=2, weight=1, color="#009900", label="1 (routes: 49658,34658,50941)"]
  "0x00124b001e721f58" -> "0x000d6ffffe2ee6a9" [penwidth=0.5, weight=0, color="#994444", label="167"]
  "0x00124b001e721f58" -> "0xccccccfffe3cc886" [penwidth=0.5, weight=0, color="#994444", label="73"]
  "0x00124b001e721f58" -> "0x00124b001e732aed" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e721f58" -> "0x00124b001e722204" [penwidth=2, weight=1, color="#009900", label="1 (routes: 7616)"]
  "0x00124b001e721f58" -> "0x001e5e0902145f59" [penwidth=0.5, weight=0, color="#994444", label="244"]
  "0x00124b001e721f58" -> "0xccccccfffe381d7c" [penwidth=0.5, weight=0, color="#994444", label="127"]
  "0x00124b001e7286ac" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Laurens_Blanket|0x00124b001e7286ac (50941)|SONOFF Zigbee smart switch (BASICZBR3)|1 minute, 50 seconds ago}"];
  "0x00124b001e7286ac" -> "0x00124b000e0eaa6f" [penwidth=2, weight=1, color="#009900", label="1 (routes: 50941,14843)"]
  "0x00124b001e7286ac" -> "0x00124b001e721f58" [penwidth=2, weight=1, color="#009900", label="6 (routes: 50941)"]
  "0x00124b001e7286ac" -> "0x00124b001e73367e" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e7286ac" -> "0x00124b001e728d41" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e7286ac" -> "0x000d6ffffe2ee6a9" [penwidth=0.5, weight=0, color="#994444", label="58"]
  "0x00124b001e7286ac" -> "0x00124b001e732aed" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e7286ac" -> "0x00124b001e722204" [penwidth=0.5, weight=0, color="#994444", label="1"]
  "0x00124b001e7286ac" -> "0x001e5e0902145f59" [penwidth=2, weight=1, color="#009900", label="250 (routes: 50941)"]
  "0x00124b001e7286ac" -> "0xccccccfffe381d7c" [penwidth=0.5, weight=0, color="#994444", label="39"]
  "0x00124b001e73367e" [style="rounded, filled", fillcolor="#4ea3e0", fontcolor="#ffffff", label="{Dereks_Blanket|0x00124b001e73367e (49658)|SONOFF Zigbee smart switch (BASICZBR3)|51 seconds ago}"];
  "0x00124b001e73367e" -> "0x00124b000e0eaa6f" [penwidth=2, weight=1, color="#009900", label="11 (routes: 34658,32224,49658,57293,16285)"]
  "0x00124b001e73367e" -> "0x00124b001e721f58" [penwidth=2, weight=1, color="#009900", label="34 (routes: 34658,49658)"]
  "0x00124b001e73367e" -> "0x00124b001e7286ac" [penwidth=2, weight=1, color="#009900", label="1 (routes: 34658)"]
  "0x00124b001e73367e" -> "0x00124b001e728d41" [penwidth=2, weight=1, color="#009900", label="9 (routes: 7616,52652)"]
  "0x00124b001e73367e" -> "0x000d6ffffe2ee6a9" [penwidth=0.5, weight=0, color="#994444", label="75"]
  "0x00124b001e73367e" -> "0xccccccfffe3cc886" [penwidth=2, weight=1, color="#009900", label="46 (routes: 49658)"]
  "0x00124b001e73367e" -> "0x00124b001e732aed" [penwidth=2, weight=1, color="#009900", label="1 (routes: 49658)"]
[... cut ...]
  "0x00158d00045ed4f0" -> "0x00124b001e722204" [penwidth=1, weight=0, color="#994444", label="39"]
  "0x90fd9ffffe2b589d" [style="rounded, dashed, filled", fillcolor="#fff8ce", fontcolor="#000000", label="{IkeaRepeater02|0x90fd9ffffe2b589d (19952)failed: lqi,routingTable|undefined undefined|1 day, 2 hours ago}"];
  "0x90fd9ffffe2b589d" -> "0x00124b001e73367e" [penwidth=0.5, weight=0, color="#994444", label="0"]
  "0x90fd9ffffe2b589d" -> "0x00124b001e722204" [penwidth=0.5, weight=0, color="#994444", label="0"]
}
```

Koenkk, 2020-07-24:
The networkmap doesn't seem to contain the LQI for this device because it probably doesn't support linkquality scans.

Given that it shows linkquality 0 and 5 in the message, the reason that it drops is probably the weak connection. I would aim for a linkquality of at least +- 20 (rule of thumb) by adding additional routers to your network.

derekoharrow, 2020-07-25:
It sits about 3 feet from 3 other routers - IkeaRepeater02, Cabin_LEDs and Cabin_Heater...

Koenkk, 2020-07-25:
I'm not sure why the linkquality is so low then, either a lot of interference or the switch didn't connect through those. Does linkquality increase when having it close (and repairing it) to the coordinator?

zhapwey, 2020-07-26:
excuse me meddling.
My experience with xiaomi devices is that they only link correctly against the coordinator or a rooter with z2m firmware, nor with the xiaomi rooters themselves I get to link correctly.

closed 2020-09-02 by github-actions[bot]

Rupunx, 2020-09-17:
**I have the same problem**

I have performed a series of tests and am confident that the case is related to the Xiaomi QBKG11LM switches

**My config:**

1. Latest dev zigbee2mqtt
2. сc2538 located on the USB extension cable
3. channel 25 (wifi on 1 channel)

**Devices:**

- 8 switches Xiaomi QBKG11LM - they are routers,
- more than 10 end devices,
- 3 Xiaomi ZNCZ02LM power plug routers,
- 3 Blitzwolf shp-13 power plug routers

### Testing
**Test 1. Initial work.**
Blitzwolf shp-13 was not available online (not bought yet at that time)

From time to time, some Xiaomi QBKG11LM switches stopped responding to requests, they switched to re-pair them.
Through analysis, I found out that these were devices that are connected according to the following **_scheme_**:
Coordinator -> router switch Xiaomi QBKG11LM -> router power plug Xiaomi ZNCZ02LM -> end device (sensor / button)
**_Key point:_** the route from power plug Xiaomi ZNCZ02LM goes through the Xiaomi switch QBKG11LM
**Conclusion**: switch Xiaomi QBKG11LM breaks the signal from routers power plug Xiaomi ZNCZ02LM

**Test 2. Removing power plug Xiaomi ZNCZ02LM**
I removed all power plug Xiaomi ZNCZ02LM from the network
**Scheme**:
Coordinator -> router switch Xiaomi QBKG11LM -> end device (sensor / button)
**Conclusion**: The network worked steadily for more than a month and I continued the series of tests

**Test 3. Now (09/17/2020). Adding a Blitzwolf shp-13 power plug router**
I restored the network as in test 1, replacing the Xiaomi power plugs with Blitzwolf power plugs
**Scheme**:
Coordinator -> router switch Xiaomi QBKG11LM -> router power plug Blitzwolf shp-13 -> end device (sensor / button)
**Now**: everything returned as it was in test 1. The first signal from the end devices stopped reaching, the switches were no longer controlled (1 day after the start of the test)

**What errors are in the log now:**
1) There are such errors (this is the error of the Xiaomi QBKG11LM switch, as in the Scheme above)
`error 2020-09-17 12:00:15: Failed to execute LQI for 'child_swtich_with_n'`

2) An error that occurs when the device does not work (does not accept the command). Those. I turn on / off Xiaomi QBKG11LM via zigbee2mqtt (via Home Assistant). Xiaomi QBKG11LM does not react to the command in any way.

```
error 2020-09-17 15:55:58: Publish 'set' 'state' to 'bedr_swtich_with_n' failed: 'Error: Command 0x00158d0002335e7d / 1 genOnOff.on ({}, {"timeout": 10000, "disableResponse": false, "disableDefaultResponse": false, "direction": 0, "srcEndpoint": null, "reservedBits": 0, "manufacturerCode": null, "transactionSequenceNumber": null}) failed (Timeout - 31502 - 1 - 225 - 6 - 11 after 10000ms) '
info 2020-09-17 15:55:58: MQTT publish: topic 'zigbee2mqtt / bridge / log', payload '{"message": "Publish' set '' state 'to' bedr_swtich_with_n 'failed:' Error: Command 0x00158d0002335e7d / 1 genOnOff.on ({}, {\ "timeout \": 10000, \ "disableResponse \": false, \ "disableDefaultResponse \": false, \ "direction \": 0, \ "srcEndpoint \": null, \ "reservedBits \": 0, \ "manufacturerCode \": null, \ "transactionSequenceNumber \": null}) failed (Timeout - 31502 - 1 - 225 - 6 - 11 after 10000ms) '"," meta ": {" friendly_name ":" bedr_swtich_with_n "}," type ":" zigbee_publish_error "} '

```