Koenkk/zigbee2mqtt #3947: MiJia motion sensor being detected as XBee router

mikeage, 2020-07-21:
# Bug Report
## What happened

I paired a new MiJia motion sensor (I have several already, pair earlier), but it's being detected as an XBee router in the message sent to home assistant. From what I can tell from the logs, zigbee2mqtt is detecting it properly and does report occupancy correctly via mqtt, but the homeassistant auto configuration is incorrect

Autoconfig:
```
$ mosquitto_sub -t 'homeassistant/#' -h localhost -v  | grep 0x00158d000353ec72
homeassistant/sensor/0x00158d000353ec72/linkquality/config {"icon":"mdi:signal","unit_of_measurement":"lqi","value_template":"{{ value_json.linkquality }}","state_topic":"zigbee2mqtt/0x00158d000353ec72","json_attributes_topic":"zigbee2mqtt/0x00158d000353ec72","name":"0x00158d000353ec72_linkquality","unique_id":"0x00158d000353ec72_linkquality_zigbee2mqtt","device":{"identifiers":["zigbee2mqtt_0x00158d000353ec72"],"name":"0x00158d000353ec72","sw_version":"Zigbee2mqtt 1.14.1-dev","model":"Router (XBee)","manufacturer":"Digi"},"availability_topic":"zigbee2mqtt/bridge/state"}
```

Actual data sent:
```
$ mosquitto_sub -t 'zigbee2mqtt/0x00158d000353ec72/#' -h localhost -v
zigbee2mqtt/0x00158d000353ec72 {"battery":100,"voltage":3055,"linkquality":84,"occupancy":true}
```

## What did you expect to happen

Device detected and reports properly

## How to reproduce it (minimal and precise)

Log at https://pastebin.com/1K0BRg2c shows a force_remove, permit_join, then a reset of the device via the reset button.

## Debug Info
Zigbee2mqtt version: dev branch, 57b45f662489fab7ca2c20b88d76a2a0076d2395
Adapter hardware: CC2531
Adapter firmware version: Coordinator firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190425}}'

Koenkk, 2020-07-21:
Can you try to `force_remove` the device and then repair it? https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove

mikeage, 2020-07-21:
In the log above, I began with a force remove and then a join

mikeage, 2020-07-21:
(I've since upgraded to ad0bbbec97a5ccef1db24539c23ae0e55dcf886d, but there's no change in the same scenario (force_remove, permit_join, reset) )

Koenkk, 2020-07-21:
Can you check if this has been fixed in the latest dev branch? https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html

mikeage, 2020-07-21:
So with the latest commit (1d3a577fc49e34374c55acc63521ead49092dd65) it actually looks a bit worse during the interview stage, I think (and still sends the wrong HomeAssistant auto discovery), but after restarting zigbee2mqtt, it sends the proper autodiscovery messages without any extra pairing!

Pastebin of the discovery: https://pastebin.com/2eBf2PRg

Pastebin of the restart: https://pastebin.com/xQ62bsyB (has some extra stuff, sorry)

mikeage, 2020-07-21:
BTW, I just went back to the commit before, and it sends proper autodiscovery messages upon restart. So both are identical in this regard.

Koenkk, 2020-07-21:
Did some more fixes, can you try again?

mikeage, 2020-07-21:
Afraid not. Log: https://pastebin.com/7fcsmeeh

I see that the homeassistant autodiscovery messages are sent almost immediately, rather than at the end when (I think) the interview is complete.

I have to stop now, but I will be back to try more experiments in ~8 hours or so. Thanks

Koenkk, 2020-07-21:
Can you edit `/app/zigbee2mqtt/lib/extension/homeassistant.js` and change the `onZigbeeEvent` function to 

```js
onZigbeeEvent(type, data, resolvedEntity) {
    // Don't discover on deviceJoined as definition might not be stable yet.
    // https://github.com/Koenkk/zigbee2mqtt/issues/3947
    console.log('ONZIGBEEEVENT', type, data);
    if (resolvedEntity && type !== 'deviceLeave' && type !== 'deviceJoined' && this.mqtt.isConnected()) {
        console.log('ONZIGBEEEVENT - DISCOVER!', type, data);
        this.discover(resolvedEntity);
    }
}
```

and provide the logging again?

mikeage, 2020-07-22:
Here you go: https://pastebin.com/kMaqgwdT

mikeage, 2020-07-22:
BTW, I don't know if it's directly related, but even after the restart, when occupancy is correctly reported, battery and voltage are not being reported correctly (or at all) in the mqtt message. I wasn't sure if this was related to a new version of the device, so I went ahead and repaired an older device that does report (originally paired with 1.14.1-dev, commit df5f740), and it too no longer reports battery (and has the same issue with the wrong autodiscovery message). I have a full log since July 1st (the original pairing of the older device was on July 14th) if you'd like

mikeage, 2020-07-22:
I'm using `git bisect` now to find the offending commit

mikeage, 2020-07-22:
Ok, the problem was introduced in e7a799d (which makes sense; that's when the use of the fingerprint started, from what I understart)

Koenkk, 2020-07-22:
> Ok, the problem was introduced in [e7a799d](https://github.com/Koenkk/zigbee2mqtt/commit/e7a799d8cc7181d99217e82e313b2a8527cc7cf3) (which makes sense; that's when the use of the fingerprint started, from what I understart)

The problem was indeed introduced with that commit. Issue should be fixed in latest dev branch now.

mikeage, 2020-07-22:
working with 28d48f3d9663b3bda87f90847b0b7c3e39422ef8. Thank you!

closed 2020-07-22 by mikeage