Koenkk/zigbee2mqtt #3948: Using state: toggle to turn on doesn't set brightness

elfhack, 2020-07-21:
# Bug Report
## What happened
When using 
```json
{ "state" : "toggle" }
```
on a bulb (an Ikea TRADFRI bulb E27 WS opal 980lm on fw 2.3.050 in this case), z2mqtt sets the brightness to 0 regardless if the new state is off or on. Also, after turning it off with toggle, setting the state via "state": "on" does nothing, because zigbee2mqtt" remembers the brightness is set to 0, which effectively means the bulb should be off. This now applies to "toggle" as well. Because of this, one needs to first set the brightness > 0 to effectively turn the bulb back on, and restore normal operation
## What did you expect to happen
Using toggle to turn back on the bulb should restore brightness

## How to reproduce it (minimal and precise)
Turn the bulb off using "state": "toggle", then back on.
## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version: 20190608

Koenkk, 2020-07-21:
- Do you have a default `transition` specified? 
- Can you share your `configuration.yaml`?
- Can you provide the herdsman debug logging of this issue?

To enable herdsman debug logging, see https://www.zigbee2mqtt.io/information/debug.html#zigbee-herdsman-debug-logging

elfhack, 2020-07-21:
@Koenkk 
transition - indeed I do, 0.5 for lights
configuration.yaml:
```
advanced:
  report: true
  log_output:
    - console
homeassistant: true
mqtt:
  base_topic: zigbee2mqtt
  server: 'mqtt://mosquitto.homeassistant'
permit_join: false
serial:
  port: /dev/zigbee
  disable_led: true
devices:
  '0x000b57fffe396706':
    friendly_name: Kitchen_Remote
  '0x000b57fffe39eb73':
    friendly_name: Couch_Remote
  '0x000b57fffe368edd':
    friendly_name: Bedroom_Remote
  '0x00124b001e728db4':
    friendly_name: Kitchentop
    retain: true
  '0x000b57fffedbd90c':
    friendly_name: Kitchen
    retain: true
    transition: 0.5
groups: {}
```
herdsman logs attached, as there's too much of it. The moment you're looking for is when there's a toggle, then an off, then an on.
[z2m.txt](https://github.com/Koenkk/zigbee2mqtt/files/4955548/z2m.txt)

Koenkk, 2020-07-21:
Does it work correctly without `transition`?

elfhack, 2020-07-21:
No, result is exactly the same 
[image]

elfhack, 2020-07-21:
Oh, I don't know if I mentioned this. The bulb actually retains it's brightness on toggle, even if z2mqtt reports brightness 0!

Koenkk, 2020-07-21:
Can you clear the `data/state.json` entry first?

elfhack, 2020-07-21:
For just that one bulb, or can I nuke the whole file safely?

Koenkk, 2020-07-21:
For just that one bulb (removing it entirely is also possible, but then device will have no states until they send it again)

elfhack, 2020-07-21:
I removed just the bulb while z2mqtt was off, the entry didn't get recreated yet. The result is still the same - brightness 0 on toggle and not being reset when toggled to on

Koenkk, 2020-07-21:
Can you provide me the complete debug logging to reproduce this issue after a fresh start from zigbee2mqtt, without the device being in `data/state.json` and without a `transition` specified for this device?

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

elfhack, 2020-07-21:
Btw i think these are the lines you're looking for
```
[34mzigbee2mqtt:debug[39m 2020-07-21 20:57:10: Received MQTT message on 'zigbee2mqtt/Kitchen/set' with data '{"state": "toggle"}'
[34mzigbee2mqtt:debug[39m 2020-07-21 20:57:10: Publishing 'set' 'state' to 'Kitchen'
2020-07-21T18:57:10.073Z zigbee-herdsman:controller:endpoint Command 0x000b57fffedbd90c/1 genOnOff.toggle({}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":false,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-07-21T18:57:10.074Z zigbee-herdsman:adapter:zStack:znp:SREQ --> AF - dataRequest - {"dstaddr":4937,"destendpoint":1,"srcendpoint":1,"clusterid":6,"transid":11,"options":0,"radius":30,"len":3,"data":{"type":"Buffer","data":[1,6,2]}}
2020-07-21T18:57:10.074Z zigbee-herdsman:adapter:zStack:unpi:writer --> frame [254,13,36,1,73,19,1,1,6,0,11,0,30,3,1,6,2,103]
2020-07-21T18:57:10.084Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,1,100,1,0,100]
2020-07-21T18:57:10.084Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,1,100,1,0,100]
2020-07-21T18:57:10.085Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 1 - 3 - 4 - 1 - [0] - 100
2020-07-21T18:57:10.085Z zigbee-herdsman:adapter:zStack:znp:SRSP <-- AF - dataRequest - {"status":0}
2020-07-21T18:57:10.085Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-21T18:57:10.091Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,68,128,0,1,11,205]
2020-07-21T18:57:10.091Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,68,128,0,1,11,205]
2020-07-21T18:57:10.091Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 4 - 128 - [0,1,11] - 205
2020-07-21T18:57:10.091Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - dataConfirm - {"status":0,"endpoint":1,"transid":11}
2020-07-21T18:57:10.091Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-21T18:57:10.119Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,73,19,0,216]
2020-07-21T18:57:10.119Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,73,19,0,216]
2020-07-21T18:57:10.120Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [73,19,0] - 216
2020-07-21T18:57:10.120Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":4937,"relaycount":0,"relaylist":[]}
2020-07-21T18:57:10.120Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-21T18:57:10.124Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,3,69,196,73,19,0,216]
2020-07-21T18:57:10.124Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,3,69,196,73,19,0,216]
2020-07-21T18:57:10.124Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 3 - 2 - 5 - 196 - [73,19,0] - 216
2020-07-21T18:57:10.124Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- ZDO - srcRtgInd - {"dstaddr":4937,"relaycount":0,"relaylist":[]}
2020-07-21T18:57:10.124Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
2020-07-21T18:57:10.178Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,25,68,129,0,0,6,0,73,19,1,1,0,94,0,207,27,245,0,0,5,8,6,11,2,0,73,19,29,186]
2020-07-21T18:57:10.178Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,25,68,129,0,0,6,0,73,19,1,1,0,94,0,207,27,245,0,0,5,8,6,11,2,0,73,19,29,186]
2020-07-21T18:57:10.178Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 25 - 2 - 4 - 129 - [0,0,6,0,73,19,1,1,0,94,0,207,27,245,0,0,5,8,6,11,2,0,73,19,29] - 186
2020-07-21T18:57:10.178Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":4937,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":94,"securityuse":0,"timestamp":16063439,"transseqnumber":0,"len":5,"data":{"type":"Buffer","data":[8,6,11,2,0]}}
2020-07-21T18:57:10.179Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":false,"reservedBits":0},"transactionSequenceNumber":6,"manufacturerCode":null,"commandIdentifier":11},"Payload":{"cmdId":2,"statusCode":0}},"address":4937,"endpoint":1,"linkquality":94,"groupID":0}'
2020-07-21T18:57:10.179Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext []
[32mzigbee2mqtt:info [39m 2020-07-21 20:57:10: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"brightness":0,"linkquality":94,"update_available":false,"color_temp":302,"state":"ON"}'
2020-07-21T18:57:10.181Z zigbee-herdsman:adapter:zStack:unpi:parser <-- [254,27,68,129,0,0,6,0,73,19,1,1,0,92,0,225,27,245,0,0,7,8,24,10,0,0,16,1,73,19,29,154]
2020-07-21T18:57:10.181Z zigbee-herdsman:adapter:zStack:unpi:parser --- parseNext [254,27,68,129,0,0,6,0,73,19,1,1,0,92,0,225,27,245,0,0,7,8,24,10,0,0,16,1,73,19,29,154]
2020-07-21T18:57:10.182Z zigbee-herdsman:adapter:zStack:unpi:parser --> parsed 27 - 2 - 4 - 129 - [0,0,6,0,73,19,1,1,0,92,0,225,27,245,0,0,7,8,24,10,0,0,16,1,73,19,29] - 154
2020-07-21T18:57:10.182Z zigbee-herdsman:adapter:zStack:znp:AREQ <-- AF - incomingMsg - {"groupid":0,"clusterid":6,"srcaddr":4937,"srcendpoint":1,"dstendpoint":1,"wasbroadcast":0,"linkquality":92,"securityuse":0,"timestamp":16063457,"transseqnumber":0,"len":7,"data":{"type":"Buffer","data":[8,24,10,0,0,16,1]}}
[... cut ...] ifier":10},"Payload":[{"attrId":0,"dataType":16,"attrData":1}]},"address":4937,"endpoint":1,"linkquality":92,"groupID":0}'
[34mzigbee2mqtt:debug[39m 2020-07-21 20:57:10: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genOnOff', data '{"onOff":1}' from endpoint 1 with groupID 0
[32mzigbee2mqtt:info [39m 2020-07-21 20:57:10: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"brightness":0,"linkquality":92,"update_available":false,"color_temp":302,"state":"ON"}'
```

Koenkk, 2020-07-21:
Yes, but it also depends on various states, so please follow https://github.com/Koenkk/zigbee2mqtt/issues/3948#issuecomment-662048081

elfhack, 2020-07-21:
Sure, attached
[z2m.log](https://github.com/Koenkk/zigbee2mqtt/files/4955625/z2m.log)

Koenkk, 2020-07-22:
Can you check if this has been fixed?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

elfhack, 2020-07-22:
@Koenkk I'll report back tommorow then

elfhack, 2020-07-25:
@Koenkk I am happy to report this is now fixed, I'm on 1.14.2-dev (commit #00f836f). Is this is 1.14.2, or do I need to wait for 1.14.3? Either way, thanks for the fix :)

Koenkk, 2020-07-26:
The fix is not in 1.14.2 so you have to wait for the next release (2.5 weeks from now) or stay on dev. Thanks for confirming.

closed 2020-07-26 by Koenkk

barrymossel, 2020-09-16:
I still have the same issue 1.14.4 (HA add-on, also tried edge) with (at least) IKEA Tradfri E27 1000lm.

```
{"brightness":190,"color_temp":100,"linkquality":84,"state":"ON","update_available":false}
qos : 0, retain : false, cmd : publish, dup : false, topic : zigbee2mqtt/Hoeklamp, messageId : , length : 112
{"state": "TOGGLE", "brightness": 254, "color_temp": 100}
qos : 0, retain : false, cmd : publish, dup : false, topic : zigbee2mqtt/Hoeklamp/set, messageId : , length : 83
```
configuration.yaml:

```
{
  "external_converters": [],
  "devices": "devices.yaml",
  "groups": "groups.yaml",
  "homeassistant": true,
  "permit_join": false,
  "mqtt": {
    "base_topic": "zigbee2mqtt",
    "server": "mqtt://core-mosquitto",
    "user": XXXXX,
    "password": XXXXX
  },
  "serial": {
    "port": "/dev/serial/by-id/usb-Texas_Instruments_TI_CC2531_USB_CDC___0X00124B0018E2EE14-if00"
  },
  "advanced": {
    "log_level": "info",
    "pan_id": 6754,
    "channel": 11,
    "network_key": [
      1,
      3,
      5,
      7,
      9,
      11,
      13,
      15,
      0,
      2,
      4,
      6,
      8,
      10,
      12,
      13
    ],
    "availability_blocklist": [],
    "availability_passlist": []
  },
  "device_options": {},
  "blocklist": [],
  "passlist": [],
  "queue": {},
  "experimental": {}
}
```

Transition is set to 0 for this light.

DL099, 2021-05-25:
Was this ever resolved? as I'm currently also running into this. Whenever i'm using toggle instead of on or off my HUE light group does not process the requested brightness. It just uses the last brightness setting it was on when turned off.

Rafaille, 2021-08-15:
Hi, not fixed for me either.
Exact same issue as DL099 above.
Cheers

ahocquet, 2021-12-06:
Not fixed for me as well. Device used: `TRADFRI bulb E27 WS opal 1000lm`