Koenkk/zigbee2mqtt #3949: transition should have a debounce/not report immiediately

elfhack, 2020-07-21:
# Bug Report
## What happened
When a lightbulb has a transition enabled, even a small one (0.5 seconds), and a parameter is set (like brightness), zigbee2mqtt publishes the requested value immediately, then publishes an interim value from the transition, then after the transition is done it lands on the requested value. This confuses things like homeassistant where settings the brightness makes the slider go haywire, since it tries to set values even before the user lets go of the slider, which sometime results in the end value not being the one set by HA, since zigbee2mqtt was still processing the value set while sliding the slider.

This has similar results in thing like node-red, where one has to take into account the transition value and delay any commands by it

## What did you expect to happen
zigbee2mqtt should either ignore any value set before the transition finishes? Im pretty sure it shouldn't publish the set value as the current state immediately.

## How to reproduce it (minimal and precise)
set transition: 1 on a bulb

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC2531, CC2530, CC26X2R1 or CC1352P-2
Adapter firmware version: 20190608

Koenkk, 2020-07-21:
Wether zigbee2mqtt should publish the new state after the command has been executed or after the transition has been completed is debatable. But the issue in Home Assistant that the command is send before the user releases the slider sounds like a Home Assistant bug?

elfhack, 2020-07-21:
@Koenkk I'm sorry, I actually went back and re-checked, and this isn't the case. I must have confused it when trying out node-red. However, it is still the case that when setting things via home-assistant like brightness, the slider will bounce around and sometimes settle not on the value that was set (ditto with the color wheel)

Koenkk, 2020-07-22:
@elfhack can you post the zigbee2mqtt debug logging while this happens?

elfhack, 2020-07-25:
I have set transition: 5 to artificially exacerbate the problem. The bulb was at a very low brightness state, and was requested a brightness of 250. As you can see it initially reports a 250 brightness, the immediately goes down to 6 and ramps up to 250.

```
zigbee2mqtt:info 2020-07-25 20:42:37: Switching log level to 'debug'
zigbee2mqtt:info 2020-07-25 20:42:37: MQTT publish: topic 'zigbee2mqtt/bridge/config', payload '{"version":"1.14.2-dev","commit":"00f836f","coordinator":{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}},"log_level":"debug","permit_join":false}'
zigbee2mqtt:debug 2020-07-25 20:42:52: Received MQTT message on 'zigbee2mqtt/Kitchen/set' with data '{"state": "ON", "brightness": 250}'
zigbee2mqtt:debug 2020-07-25 20:42:52: Publishing 'set' 'brightness' to 'Kitchen'
zigbee2mqtt:debug 2020-07-25 20:42:52: Received MQTT message on 'zigbee2mqtt/Kitchen/set' with data '{"state": "ON", "brightness": 250}'
zigbee2mqtt:debug 2020-07-25 20:42:52: Publishing 'set' 'brightness' to 'Kitchen'
zigbee2mqtt:info 2020-07-25 20:42:52: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"update_available":false,"state":"ON","brightness":250,"linkquality":84,"color_temp":313}'
zigbee2mqtt:info 2020-07-25 20:42:52: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"update_available":false,"state":"ON","brightness":250,"linkquality":84,"color_temp":313}'
zigbee2mqtt:debug 2020-07-25 20:42:52: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":6}' from endpoint 1 with groupID 0
zigbee2mqtt:info 2020-07-25 20:42:52: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"update_available":false,"state":"ON","brightness":6,"linkquality":81,"color_temp":313}'
zigbee2mqtt:debug 2020-07-25 20:42:55: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":153}' from endpoint 1 with groupID 0
zigbee2mqtt:info 2020-07-25 20:42:55: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"update_available":false,"state":"ON","brightness":153,"linkquality":78,"color_temp":313}'
zigbee2mqtt:debug 2020-07-25 20:42:58: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":250}' from endpoint 1 with groupID 0
zigbee2mqtt:info 2020-07-25 20:42:58: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"update_available":false,"state":"ON","brightness":250,"linkquality":84,"color_temp":313}'
```

elfhack, 2020-07-25:
This [confuses the hell out of home assistant](https://imgur.com/zLcsVcT). Again, this is 5 seconds, but sometimes 0.5 transition time has the same effect if the setting delta is big enough for the bulb

Koenkk, 2020-07-26:
Ah I understand now, the device reports the brightness (probably because your are using https://www.zigbee2mqtt.io/information/report.html). As the brightness values are already reported by the bulb, you don't need the optimistic/assumed values after command succeeded. 

A solution would be to let the user ignore these, e.g.

```yaml
devices:
  `0x12345678':
     friendly_name: 'my_device'
     ignore_optimistic_attributes: ['brightness']
```

elfhack, 2020-07-26:
Hmm, I haven't considered the impact of reporting in this. Would this nullify reporting of those attributes altogether, or just for the time of the transition?

Koenkk, 2020-07-26:
@elfhack  all together, note that when the brightness is changed via the remote this is reported by the device (which will not be ignored)

elfhack, 2020-07-26:
I don't think I understood that one :) The bulb will report the brightness change via the remote, but not when zigbee2mqtt changes it?

Koenkk, 2020-07-27:
There are 2 kind of state changes:
1. Actual state (messages) reports from the device (this will be send when `report: true` is used).
2. Optimistic state messages from zigbee2mqtt whenever a command succeeds (e.g. if a command setting the brightness to 150 succeeds, we assume that the brightness of the bulb is now 150).

Number 2 is causing the problem here that's why I propose to have an option to disable this. Try replacing `/app/zigbee2mqtt/lib/extension/publish.js` with https://github.com/Koenkk/zigbee2mqtt/blob/a9a5d5dabc5fbd7c0af24cceb87f0b34537643a4/lib/extension/publish.js and in `configuration.yaml` put `ignore_optimistic_attributes` for this device (like the example below).

```yaml
devices:
  `0x12345678':
     friendly_name: 'my_device'
     ignore_optimistic_attributes: ['brightness']
```

elfhack, 2020-07-28:
Ah, i understand now. I just tried out the patch, and it doesn't really fix the situation at all, given my test case.
```
  '0x000b57fffedbd90c':
    friendly_name: Kitchen
    retain: true
    transition: 2
    ignore_optimistic_attributes: ['brightness']
```
Unless i made an error in this config, it behaves exactly the same, because I think the exact inverse of what you propose should be done - it should report the optimistic state for $transition seconds of the device, ignoring the reports from the device.

This is how this looks in home assistant https://imgur.com/a/lIqlLGD
And here are the debug logs:

```
zigbee2mqtt:debug 2020-07-28 18:30:29: Received MQTT message on 'zigbee2mqtt/Kitchen/set' with data '{"state": "ON", "brightness": 5}'
zigbee2mqtt:debug 2020-07-28 18:30:29: Publishing 'set' 'brightness' to 'Kitchen'
zigbee2mqtt:debug 2020-07-28 18:30:29: Received MQTT message on 'zigbee2mqtt/Kitchen/set' with data '{"state": "ON", "brightness": 5}'
zigbee2mqtt:debug 2020-07-28 18:30:29: Publishing 'set' 'brightness' to 'Kitchen'
zigbee2mqtt:info  2020-07-28 18:30:29: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"state":"ON","update_available":false,"brightness":248,"linkquality":89,"color_temp":334}'
zigbee2mqtt:info  2020-07-28 18:30:29: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"state":"ON","update_available":false,"brightness":248,"linkquality":89,"color_temp":334}'
zigbee2mqtt:debug 2020-07-28 18:30:29: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":238}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-28 18:30:29: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"state":"ON","update_available":false,"brightness":238,"linkquality":86,"color_temp":334}'
zigbee2mqtt:debug 2020-07-28 18:30:32: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":5}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-28 18:30:32: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"state":"ON","update_available":false,"brightness":5,"linkquality":86,"color_temp":334}'
zigbee2mqtt:debug 2020-07-28 18:30:33: Received MQTT message on 'zigbee2mqtt/Kitchen/set' with data '{"state": "ON", "brightness": 248}'
zigbee2mqtt:debug 2020-07-28 18:30:33: Publishing 'set' 'brightness' to 'Kitchen'
zigbee2mqtt:info  2020-07-28 18:30:33: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"state":"ON","update_available":false,"brightness":5,"linkquality":86,"color_temp":334}'
zigbee2mqtt:debug 2020-07-28 18:30:35: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":216}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-28 18:30:35: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"state":"ON","update_available":false,"brightness":216,"linkquality":89,"color_temp":334}'
zigbee2mqtt:debug 2020-07-28 18:30:37: Saving state to file /data/state.json
zigbee2mqtt:debug 2020-07-28 18:30:38: Received Zigbee message from 'Kitchen', type 'attributeReport', cluster 'genLevelCtrl', data '{"currentLevel":248}' from endpoint 1 with groupID 0
zigbee2mqtt:info  2020-07-28 18:30:38: MQTT publish: topic 'zigbee2mqtt/Kitchen', payload '{"state":"ON","update_available":false,"brightness":248,"linkquality":86,"color_temp":334}'
```

Koenkk, 2020-07-28:
Ah I see, my fix indeed doesn't work.

Ignoring messages from the device is tricky, since you will also get non when a bound remote changes the state.

elfhack, 2020-07-28:
During the transition? I guess that's a problem in that edge case, but except for that it would be fine. However, let me approach this from a different angle, since I believe turning off reporting would "fix" this issue for me. I want to bring up polling - i know zigbee2mqtt doesn't do it, but is it possible at all in the zigbee world? I think Hue does it, since when I used tradfri remotes with bulbs on the hue zigbee network, changes caused by remotes would pop up on Hue, but with a delay. Are you just principally opposed to the concept, or is it not possible with the hardware at all? I wanted to ask about it anyway since I have Innr bulbs which don't support reporting anyway.

Koenkk, 2020-07-29:
Note that reporting is currently not uninitialized correctly, you need to set `report: false` and after that factory reset the bulb (will be fixed in https://github.com/Koenkk/zigbee2mqtt/issues/3826)

Polling is possible (see https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttfriendly_nameget), I try to stay away from this as much as possible because it add a lot of stress on the network.

elfhack, 2020-07-29:
Ok, so i just disabled reporting, and it works perfectly. I was concerned about availability (if someone were to turn off the bulb), but it looks like you alredy have that covered with availability_timeout, and a modest 5 minut ping works perfectly, along with rejecting any failed state changes. Since I'm not using any bindings, I'm happy with this state of affairs.

However, I still think that implementing some kind of careful per-device polling mechanism, with a shared FIFO queue so as not to bog down the network would work wonders, especially for people whose devices don't support reporting

Koenkk, 2020-07-29:
@elfhack this is already done for specific cases (https://github.com/Koenkk/zigbee2mqtt/blob/b91b456f981012d60afd880097af2aa933a77e09/lib/extension/report.js#L209) so extending that is possible

closed 2020-09-05 by github-actions[bot]