Koenkk/zigbee2mqtt #3953: Incorrect device type

to4ko, 2020-07-22:
# Bug Report
## What happened
Device type incorrectly identified - QBKG03LM not a router. It's correct identified in description, but incorrect in mqtt data 

[image]

## What did you expect to happen
Correct identification
## How to reproduce it (minimal and precise)
Pair one of the QBKG03LM
## Debug Info
Zigbee2mqtt version: 0.14.1
Adapter hardware: CC2538 (same on CC1352P)
Adapter firmware version: 20020211

Koenkk, 2020-07-23:
Fixed, can you verify?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

to4ko, 2020-07-27:
> Fixed, can you verify?
> 
> Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

running edge addon - same story
[image]

Koenkk, 2020-07-28:
How does it show in startup log?

Mine:

```
zigbee2mqtt:info  2020-07-28 19:17:50: Currently 1 devices are joined:
zigbee2mqtt:info  2020-07-28 19:17:50: 0x00158d00022baf28 (0x00158d00022baf28): QBKG04LM - Xiaomi Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter (EndDevice)
zigbee2mqtt:warn  2020-07-28 19:17:50: `permit_join` set to  `true` in configuration.yaml.
```

And `zigbee2mqtt/bridge/config/devices/get`

```
zigbee2mqtt:info  2020-07-28 19:18:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"devices","message":[{"ieeeAddr":"0x00124b0018e1af91","type":"Coordinator","networkAddress":0,"friendly_name":"Coordinator","softwareBuildID":"zStack12","dateCode":"20190608","lastSeen":1595956696970},{"ieeeAddr":"0x00158d00022baf28","type":"EndDevice","networkAddress":14197,"model":"QBKG04LM","vendor":"Xiaomi","description":"Aqara single key wired wall switch without neutral wire. Doesn't work as a router and doesn't support power meter","friendly_name":"0x00158d00022baf28","manufacturerID":4151,"manufacturerName":"LUMI","powerSource":"DC Source","modelID":"lumi.ctrl_neutral1","hardwareVersion":38,"dateCode":"11-11-2016","lastSeen":1595956599240}]}'
```

If it shows as router there try to reconfigure the device via https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure

closed 2020-09-05 by github-actions[bot]