Koenkk/zigbee2mqtt #3954: 1.14.2 Docker image seems broken!

PedroLamas, 2020-07-22:
# Bug Report

## What happened

Pulling the latest docker image (1.14.2) is currently failing, as shown here:

[image]

## What did you expect to happen

Pulling image work as expected!

## How to reproduce it (minimal and precise)

```sh
docker pull koenkk/zigbee2mqtt:1.14.2
```

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC26X2R1
Adapter firmware version: (latest)

Koenkk, 2020-07-22:
What architecture? Seems to work fine for amd64 here:

[image]

PedroLamas, 2020-07-22:
This is a NanoPi M4V2, so arm64 architecture.

PedroLamas, 2020-07-22:
Hum... might be something wrong in this system as I managed to pull that image on Docker under WSL2 with this:

```
docker pull koenkk/zigbee2mqtt:1.14.2@sha256:7c068f9302968b216edc30a8c1a95c21b66b86d3de298d6fdf9eabc68842492f
```

PedroLamas, 2020-07-22:
After a lot of googling around for a solution, I ended up following [this suggestion](https://stackoverflow.com/questions/60209172/docker-cannot-delete-intermediate-images-from-a-broken-pull#comment106498678_60209172) and the problem is now gone!

closed 2020-07-22 by PedroLamas