Koenkk/zigbee2mqtt #3956: Unable to pair Tuya Zigbee Switch X712A 'Interview failed because can not get node descriptor'

yaroslavbon, 2020-07-22:
# Bug Report
## What happened
I'm having problems while pairing [Tuya Zigbee Switch X712A](https://www.aliexpress.com/item/4000298926256.html) (2 buttons)
## What did you expect to happen
Switch paired successfully
## How to reproduce it
Press and hold switch buttons (10 sec) until it will enter pairing mode
## Debug Info
Logs: https://pastebin.com/8XHZgzqF

Note: my switch does not want to enter pairing mode every time I hold the buttons for 10 sec. I mean, It starts blinking like it is in pairing mode, but there are no corresponding logs in zigbee2mqtt. So, it takes somewhere from 3 to 10 attempts for it to start the pairing process at all. 
Also, I've tried to use extension USB cable and put CC2531 right near the switch - no effect.
So, I assume that the switch itself might be faulty since there is a lot of comments in Ali where people saying that they have little to no problems connecting this device to zigbee2mqtt.

Zigbee2mqtt version: 1.14.2 (commit #b91b456) | also tried on latest dev
Adapter hardware: CC2531
Adapter firmware version: 20190608

yaroslavbon, 2020-07-28:
Hello @Koenkk. Can I ask you to take a look at the attached logs and comment on them anyhow? I mean, even if you do not have a solution it would be great if you can help me understand whether something wrong with the software or with the switch itself.

Koenkk, 2020-07-28:
The device does not respond to any of the requests, you can try pairing the device closer to the coordinator, also make sure there are not other zigbee network running.

closed 2020-09-05 by github-actions[bot]