Koenkk/zigbee2mqtt #3957: Can't 1.14.2 install in HomeAssistant

alexspires83, 2020-07-23:
# Bug Report
## What happened
20-07-23 00:15:34 ERROR (SyncWorker_2) [supervisor.docker.interface] Can't install dwelch2101/zigbee2mqtt-armhf:1.14.2 -> 404 Client Error: Not Found ("manifest for dwelch2101/zigbee2mqtt-armhf:1.14.2 not found: manifest unknown: manifest unknown").
## What did you expect to happen
Addon Installed
## How to reproduce it (minimal and precise)
Remove Addon and try to reinstall
## Debug Info
Zigbee2mqtt version:
Adapter hardware: CC2531
Adapter firmware version: Zstack12

Stupco, 2020-07-23:
Same issue, after upgrading from 1.14.1 to 1.14.2 the addon failed to start on restart (nothing in logs).

So I uninstalled and re-installed, but it won't install get this error: "Failed to install addon, Unknown Error, see logs"
Nothing in logs.

alexspires83, 2020-07-23:
I install the EDGE (Dev) version to continue my work on zigbee devices, my lucky is that I didn't have everything setted up...waiting to repair the STABLE 1.14.2 version.

Gooman-rus, 2020-07-23:
I have the same problem when I'm trying to update from 1.14.1 to 1.14.2.

Koenkk, 2020-07-23:
Fixed

closed 2020-07-23 by Koenkk