Koenkk/zigbee2mqtt #3961: Feature Request: Dresden Elektronic phoscon to zigbee2mqtt configuration converter.

renne, 2020-07-23:
# Feature Request
## What happened
Cannot import Dresden Elektronik Phoscon configuration (JSON formatted .dat-file) into zigbee2mqtt.
## What did you expect to happen
Import Dresden Elektronik Phoscon configuration (JSON formatted.dat-file) into zigbee2mqtt successfully.

At least the coordinator- and device configuration, groups and bindings should be possible to import from the Phoscon App.
This would save hours to to re-register 50+ end devices.

## How to reproduce it (minimal and precise)
Try to find e.g. a command-line  tool for this task.

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: Dresden Conbee II
Adapter firmware version: 26580700

renne, 2020-09-07:
Stale ping

closed 2020-10-17 by github-actions[bot]

samsam-rolon, 2021-11-29:
Very intrested in this feature ! My whole house is in Legrand and re-paring every device is just : PAINFUL