Koenkk/zigbee2mqtt #3964: Unable to connect to

Tam-Lin, 2020-07-24:
# Bug Report

## What happened

After upgrading to 1.14 from 1.12, zigbee2mqtt stopped working.

## What did you expect to happen

zigbee2mqtt to start correctly.

## How to reproduce it (minimal and precise)

Try starting zigbee2mqtt.  

## Debug Info
Zigbee2mqtt version: Currently 1.14.2.  Have tried 1.14.0, 1.14.1, 1.12.
Adapter hardware: CC26X2R1 
Adapter firmware version:  Have tried both CC26X2R1_20200417 and CC26X2R1_20191106

I suspect at this point something's wrong with the board, but I'm not sure how to fix it, assuming a fix is possible.  I've tried multiple Raspberry Pis, multiple USB cables, multiple power supplies, and installed zigbee2mqtt from scratch, as opposed to running it under home assistant as I was before, but all to no avail.  I've validated that the user has write access to the /dev/ttyACM0 port.  I updated XDS110 to the latest version.

Trying to start on a fresh install of zigbee2mqtt, with the basic config.yaml from https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html  yields the following:

https://pastebin.com/kLjZ5eqK

Koenkk, 2020-07-24:
Can you check if it works with autodetect? (leave out the complete `serial` section in `configuration.yaml`)

Tam-Lin, 2020-07-25:
Still failing in the same way.  See https://pastebin.com/5sJd37gH.  It says it detected the adapter at /dev/ttyACM0, and then fails.

Koenkk, 2020-07-25:
That would indicate something is wrong with the firmware. What adapter do you have, the CC26X2R1 launchpad? Can you try reflashing by following the following guide: (pay attention to step 5): https://www.zigbee2mqtt.io/information/flashing_via_uniflash.html

Tam-Lin, 2020-07-27:
Reflashed again, and this time, it worked.  I was following the steps before, so I'm not sure what happened differently this time.  Nor why it broken in the first place.  Closing.  Thank you for your help.

closed 2020-07-27 by Tam-Lin