Koenkk/zigbee2mqtt #3966: Devices not changing state with the group

Skeletorjus, 2020-07-24:
# Bug Report
## What happened
When a group is toggled on/off the state of the devices are not updated by zigbee2mqtt. The devices themselves are working as they should. This goes the other way as well, devices are not changing state of the group.
optimistic is set to true in groups.yaml.
This worked in 1.14.1.

## What did you expect to happen
The groups to follow the devices and vice versa.

## How to reproduce it (minimal and precise)
Send command to group or device (from Home Assistant in my instance) and observe the results.

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC26X2R1
Adapter firmware version: 20191106

Koenkk, 2020-07-25:
I cannot reproduce this, example where I control `bulb` which is in `group_1`:

```
zigbee2mqtt:debug 2020-07-25 12:57:45: Received MQTT message on 'zigbee2mqtt/bulb/set' with data 'off'
zigbee2mqtt:debug 2020-07-25 12:57:45: Publishing 'set' 'state' to 'bulb'
zigbee2mqtt:info  2020-07-25 12:57:45: MQTT publish: topic 'zigbee2mqtt/group_1', payload '{"state":"OFF","brightness":0}'
zigbee2mqtt:info  2020-07-25 12:57:45: MQTT publish: topic 'zigbee2mqtt/bulb', payload '{"state":"OFF","brightness":0,"linkquality":131,"update_available":true}'
```

Can you provide your
- `configuration.yaml`
- `data/database.db`
- debug log when controlling the bulb 

To enable debug logging set in `configuration.yaml`:
```yaml
advanced:
  log_level: debug
```

Skeletorjus, 2020-07-25:
I think I found out the root issue on my end and how to reproduce it.

I have a group defined that consists of two bulbs:
```
'12':
  friendly_name: group_sengelamper_jt
  retain: false
  optimistic: true
  devices:
    - '0x000b57fffe9041db'
    - '0x000b57fffe8ce957'
```
The thing is that one of the bulbs is switched off (unpowered) at the time of creation and therefore unable to be added to a group:
```
zigbee2mqtt:error 2020-07-25 16:58:12: Failed to call 'Groups' 'onZigbeeStarted' (Error: Command 0x000b57fffe9041db/1 genGroups.add({"groupid":12,"groupname":""}, {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null}) failed (Timeout - 55446 - 1 - 6 - 4 - 0 after 10000ms)
    at Timeout._onTimeout (/zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/utils/waitress.js:46:35)
    at listOnTimeout (internal/timers.js:549:17)
    at processTimers (internal/timers.js:492:7))
```
This makes all my other groups behave like described in the bug report.
I have commented out group 12 from my groups.yaml and restarted zigbee2mqtt, and everything is fine again.

Logs when controlling two groups (rom_kjokken, rom_stue) while group 12 is in groups.yaml:
https://hastebin.com/erabuvuyux.nginx
Logs of the same situation with the offendig group removed:
https://hastebin.com/jalucojuca.sql

Would you still need database.db?

Koenkk, 2020-07-26:
Thanks, should be fixed in latest dev branch.

Skeletorjus, 2020-07-26:
Works great! Thank you for the swift response and fix. Closing this 👍

closed 2020-07-26 by Skeletorjus