Koenkk/zigbee2mqtt #3967: Apply device (specific) options to types or groups, instead of single or all devices

SaturnusDJ, 2020-07-24:
Following the advise in the manual, I was planning to add a debounce (of value 2) for WSDCGQ11LM devices (https://www.zigbee2mqtt.io/devices/WSDCGQ11LM.html). 12 of those here.

Normally you have to either add this per device or add it as `device_option` in config to apply it to all devices. 

How about a config option to apply it to all devices of a certain type, and second, nice for people who use groups, to apply it to groups.

Those options don't exist at this moment, right?

zhapwey, 2020-07-25:
if in the general device option you set debounce: 1 set this property for all devices? because I haven't gotten it to work.

Koenkk, 2020-07-25:
> How about a config option to apply it to all devices of a certain type

How would you define a "certain type"? (like router/end devices?)

SaturnusDJ, 2020-07-25:
@zhapwey 
Like in https://github.com/Koenkk/zigbee2mqtt/issues/3922.
That part is standalone in the config and applies to all device where it can be applied to.

@Koenkk 
I should have used the word model, that's how it is in the docs. E.g. `WSDCGQ11LM`.

closed 2020-09-01 by github-actions[bot]

joshuaspence, 2022-02-22:
I am also looking for a way to do this.