Koenkk/zigbee2mqtt #3968: Yale Assure YRD226/246 TSDB lock provides no status

cerebrate, 2020-07-25:
# Bug Report
## What happened

I am attempting to control a Yale Assure YRD226/246 TSDB lock from Home Assistant via Zigbee2mqtt. While I am able to pair the lock successfully, and can send lock and unlock commands to the lock, no data appears to ever be received from the lock; i.e., the battery and link quality sensors remain "unknown", and the state of the lock entity in HA remains at "unlocked" regardless of the actual state of the lock. Confirming this by watching the MQTT traffic, no messages are ever sent to the lock's _state_topic_ .

I have tried the solution to a similar issue given in #2635 to no effect.

## What did you expect to happen

Once paired, I expected lock battery, link quality, and lock status information to be received as expected.

## How to reproduce it (minimal and precise)

1. Install Zigbee2mqtt on a system running Home Assistant via Hass.IO, equipped with a ConBee Zigbee adapter.
2. Set permit_join on with an MQTT message.
3. Pair a Yale Assure YRD266/246 lock with the network.
4. Unset permit_join with an MQTT message.
5. Observe that the sensor.[addr]_battery and sensor.[addr]_linkquality entities remain "unknown". Also that the lock.[addr] entity remains at "unlocked" regardless of the state of the lock.
6. Using an MQTT client, observe that no state messages are received from the lock.

## Debug Info
Zigbee2mqtt version: 1.14.2-dev (also tested against 1.14.1)
Adapter hardware: ConBee
Adapter firmware version: {'type': 'RaspBee', 'meta': {'transportrev': 0, 'product': 0, 'majorrel': 38, 'minorrel': 31, 'maintrel': 0, 'revision': '0x261f0500'}}

Configuration and debug/herdsman-debug logs are here: https://gist.github.com/cerebrate/0fd2751bae84fa7dc3a87f86de747b44

cerebrate, 2020-07-25:
Additional: as can be seen in the supplied logs, I am confident of the functioning of Zigbee2mqtt in general, since other Zigbee devices, such as the Aqara Cube seen, operate correctly.

Koenkk, 2020-07-25:
The easiest way to debug is to sniff the traffic while the device is being paired and after that locked/unlocked manually, do you have a CC2531 to sniff with? https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

cerebrate, 2020-07-25:
I haven't yet, but I should have one tomorrow. 😃

closed 2020-09-01 by github-actions[bot]

cerebrate, 2020-09-12:
Sorry for the delay in getting back to you on this issue; I finally had the time to get the sniffer to work.

As far as I can tell, there are _no_ specific messages being sent from the locks when the door is manually locked or unlocked (per the door_use_sniff.pcapng file in captures.zip, below). The lock does seem to join the network okay, as far as I can tell with my limited knowledge of the Zigbee protocol (door_join_sniff.pcapng, also in captures.zip), but I'm not entirely sure what I ought to be looking for?

[captures.zip](https://github.com/Koenkk/zigbee2mqtt/files/5213094/captures.zip)

Koenkk, 2020-09-13:
@cerebrate can you send me your network key on telegram? (@koenkk) Otherwise I cannot decrypt the sniff.

cerebrate, 2020-09-13:
Oops, sorry. I don't have Telegram and it's embedded in the sniff anyway (so I figured I'd have to change it after the fact), so I'll just put it here:

`0c01060805090b0b000c000009090704`

Thanks!

Koenkk, 2020-09-14:
Where did you sniff the trafifc? I see that in e.g. the door_use_sniff no lock/unlock command is present. Make sure to sniff the traffic close to the lock to make sure all commands are in this sniff.

cerebrate, 2020-09-14:
Ah, there shouldn't be a command in that one, since in door_use_sniff I'm manually locking and unlocking them (i.e., with the knob on the lock body) and looking for the lock to report its status?

If I send a lock/unlock command to the lock, the command works - and I see both it and the command acknowledgement in the traffic - but the status doesn't update then, either.

Koenkk, 2020-09-15:
The command acknowledgement  means the command has been received which != door locked (of course this will be most of the time but not always). Can you provide a sniff when pairing the device while sniffing close to the lock?

cerebrate, 2020-09-16:
[near_sniff.zip](https://github.com/Koenkk/zigbee2mqtt/files/5233489/near_sniff.zip)

This sniff was taken about a foot away from the lock; no change, alas. (The previous ones were only about 15' away.) After the pairing, the file includes me locking and unlocking the lock multiple times (locally, that is, rather than remotely), and no status updates are apparent.

Koenkk, 2020-09-16:
@cerebrate sorry I forgot to ask to remove the device before doing so, can you provide a sniff: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigremove ?

cerebrate, 2020-09-16:
I assumed you'd want me to do that, so I removed the lock from the network before I made the sniff so that I could join it again cleanly. Or, to clarify, do you want a sniff that includes the removal as well as the rejoining? I can make one of those when I get back to the house later today.

Koenkk, 2020-09-17:
Provide one sniff for: remove, factory reset lock, pairing, control via zigbee2mqtt, manual control

cerebrate, 2020-09-19:
Sniffs for each enclosed.

[sniffs.zip](https://github.com/Koenkk/zigbee2mqtt/files/5248956/sniffs.zip)

Koenkk, 2020-09-19:
Can you try reconfiguring the device via: https://www.zigbee2mqtt.io/information/mqtt_topics_and_message_structure.html#zigbee2mqttbridgeconfigure , check in the log if this goes OK, provide the sniff of this and  manual control it afterwards?

cerebrate, 2020-09-19:
Can do. Per the log (excerpt included below) it seems to reconfigure correctly, but on manually controlling it afterwards (at approximately 130 in the attached sniff) there is still no update of the lock status.

[reconfigure_capture.zip](https://github.com/Koenkk/zigbee2mqtt/files/5250488/reconfigure_capture.zip)

Log is as follows:

```
zigbee2mqtt:debug 2020-09-19 14:02:29: Received MQTT message on 'zigbee2mqtt/bridge/configure' with data 'front_door_lock'
zigbee2mqtt:info  2020-09-19 14:02:29: Configuring 'front_door_lock'
2020-09-19T19:02:29.331Z zigbee-herdsman:controller:endpoint Bind 0x000d6f000e878f0b/1 closuresDoorLock from '0x00212effff0245f6/1'
2020-09-19T19:02:29.340Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x3f5e EP:0 SeqNr. 217 request id: 184
2020-09-19T19:02:29.367Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 184 status: 0
2020-09-19T19:02:30.794Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 218
2020-09-19T19:02:30.808Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-09-19T19:02:33.561Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-09-19T19:02:33.601Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 219
2020-09-19T19:02:33.625Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x3f5e request id: 184 confirm status: 0
2020-09-19T19:02:34.057Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-19T19:02:34.106Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 220
2020-09-19T19:02:34.121Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 220 srcAddr: 0x3f5e destAddr: 0x0 profile id: 0x0 cluster id: 0x8021
2020-09-19T19:02:34.121Z zigbee-herdsman:deconz:frameParser response payload: 184,0
2020-09-19T19:02:34.121Z zigbee-herdsman:deconz:adapter BIND RESPONSE - addr: 0x3f5e status: 0
2020-09-19T19:02:34.122Z zigbee-herdsman:controller:endpoint Bind 0x000d6f000e878f0b/1 genPowerCfg from '0x00212effff0245f6/1'
2020-09-19T19:02:34.156Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x3f5e EP:0 SeqNr. 221 request id: 185
2020-09-19T19:02:34.185Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 185 status: 0
2020-09-19T19:02:39.579Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-09-19T19:02:39.628Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 222
2020-09-19T19:02:39.643Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x3f5e request id: 185 confirm status: 0
2020-09-19T19:02:40.075Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-19T19:02:40.082Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 223
2020-09-19T19:02:40.107Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 223 srcAddr: 0x3f5e destAddr: 0x0 profile id: 0x0 cluster id: 0x8021
2020-09-19T19:02:40.107Z zigbee-herdsman:deconz:frameParser response payload: 185,0
2020-09-19T19:02:40.107Z zigbee-herdsman:deconz:adapter BIND RESPONSE - addr: 0x3f5e status: 0
2020-09-19T19:02:40.111Z zigbee-herdsman:controller:endpoint ConfigureReporting 0x000d6f000e878f0b/1 closuresDoorLock([{"attribute":"lockState","minimumReportInterval":0,"maximumReportInterval":3600,"reportableChange":0}], {"timeout":10000,"disableResponse":false,"disableDefaultResponse":true,"direction":0,"srcEndpoint":null,"reservedBits":0,"manufacturerCode":null,"transactionSequenceNumber":null})
2020-09-19T19:02:40.112Z zigbee-herdsman:deconz:adapter zclFrame.payload:
2020-09-19T19:02:40.113Z zigbee-herdsman:deconz:adapter [
  {
    direction: 0,
    attrId: 0,
    dataType: 48,
    minRepIntval: 0,
    maxRepIntval: 3600,
    repChange: 0
  }
]
2020-09-19T19:02:40.132Z zigbee-herdsman:deconz:driver DATA_REQUEST - destAddr: 0x3f5e EP:1 SeqNr. 224 request id: 186
2020-09-19T19:02:40.155Z zigbee-herdsman:deconz:frameParser DATA_REQUEST RESPONSE - request id: 186 status: 0
2020-09-19T19:02:40.571Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10100110
2020-09-19T19:02:40.584Z zigbee-herdsman:deconz:driver DATA_CONFIRM - sending data state request - SeqNr. 225
2020-09-19T19:02:40.603Z zigbee-herdsman:deconz:frameParser DATA_CONFIRM RESPONSE - destAddr: 0x3f5e request id: 186 confirm status: 0
2020-09-19T19:02:40.603Z zigbee-herdsman:deconz:adapter sendZclFrameToEndpoint - message send
2020-09-19T19:02:40.786Z zigbee-herdsman:deconz:driver send read device state from queue. seqNr: 226
2020-09-19T19:02:40.795Z zigbee-herdsman:deconz:frameParser device state: 10100010
2020-09-19T19:02:41.067Z zigbee-herdsman:deconz:frameParser DEVICE_STATE changed: 10101010
2020-09-19T19:02:41.086Z zigbee-herdsman:deconz:driver DATA_INDICATION - sending read data request - SeqNr. 227
2020-09-19T19:02:41.115Z zigbee-herdsman:deconz:frameParser DATA_INDICATION RESPONSE - seqNr. 227 srcAddr: 0x3f5e destAddr: 0x0 profile id: 0x104 cluster id: 0x101
2020-09-19T19:02:41.115Z zigbee-herdsman:deconz:frameParser response payload: 24,40,7,0
2020-09-19T19:02:41.118Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":40,"manufacturerCode":null,"commandIdentifier":7},"Payload":[{"status":0}]},"address":16222,"endpoint" [... cut ...]
2020-09-19T19:02:42.108Z zigbee-herdsman:controller:log Received 'zcl' data '{"frame":{"Header":{"frameControl":{"frameType":0,"manufacturerSpecific":false,"direction":1,"disableDefaultResponse":true,"reservedBits":0},"transactionSequenceNumber":41,"manufacturerCode":null,"commandIdentifier":7},"Payload":[{"status":0}]},"address":16222,"endpoint":1,"linkquality":0,"groupID":null}'
2020-09-19T19:02:42.109Z zigbee-herdsman:deconz:adapter response received
zigbee2mqtt:info  2020-09-19 14:02:42: Successfully configured 'front_door_lock'
```

Koenkk, 2020-09-19:
Everything goes well indeed, I honestly have no clue why the lock won't send its status. To further debug this we would need to sniff the traffic when connected to a bridge where this works. https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

cerebrate, 2020-09-19:
I have access to an Echo Plus, so I'll try the lock with that and see if I can get a sniff.

Tronadorau, 2020-10-28:
How did you go resolving this? I have the exact same lock and I am having the same issue.

cerebrate, 2020-10-28:
@Tronadorau Are you also using a ConBee / deconz?

I accidentally found a sort-of solution insofar as while it never worked with the ConBee, it did work when I replaced it with the CC2531 I was previously using as a sniffer (reflashed with the regular firmware).