Koenkk/zigbee2mqtt #3969: Link quality not transmitted with Hue Philips Hue White Single bulb B22 (8718696449691) light

jyavenard, 2020-07-25:
# Bug Report
## What happened:
The Philips Hue White Single bulb B22 (8718696449691) doesn't transmit in the payload the linkquality. This causes HA to consider it to be unavailable on restart.
Same with a Philips Hue white ambiance E26/E27 (8718696548738)

## What did you expect to happen
The linkquality property should be sent in the payload; so that HA considers the device as online.

## How to reproduce it (minimal and precise)
Pair a "Philips Hue White Single bulb B22" device (similar to https://www.philips-hue.com/en-au/p/hue-white-1-pack-b22/8718699631185, though the end serial appear different)

Look at the payload sent.
info  2020-07-25 20:36:40: MQTT publish: topic 'zigbee2mqtt/sebastien_light_2', payload '{"state":"OFF","brightness":0,"update_available":false}'

vs another hue:
info  2020-07-25 20:45:26: MQTT publish: topic 'zigbee2mqtt/dressing_light_1', payload '{"state":"OFF","linkquality":48,"update_available":false,"brightness":0,"color_temp":366}'
which is recognised when HA restarts.

I'm only assuming this is the reason why HA believes the light to not be available, as comparing various other zigbee devices ; that may not be it. If so, and it is expected for the payload of a device to not always send the link-quality; feel free to close this report.

## Debug Info
Zigbee2mqtt version: 1.14.1
Adapter hardware: CC26X2R1
Adapter firmware version: 4.10.00.78 CC26X2R1_20200417.zip

Koenkk, 2020-07-25:
I don't expect the missing `linkquality` causing Home Assistant marking devices as unavailable (it is pretty common for bulbs to have no linkquality as they normally don't send messages to the coordinator). Are you using https://www.zigbee2mqtt.io/information/availability.html ?

jyavenard, 2020-07-25:
> I don't expect the missing `linkquality` causing Home Assistant marking devices as unavailable (it is pretty common for bulbs to have no linkquality as they normally don't send messages to the coordinator). Are you using https://www.zigbee2mqtt.io/information/availability.html ?

I am yes. Would this have an impact?

Koenkk, 2020-07-25:
Yes, if the device is not reachable zigbee2mqtt will mark them as unavailable when using this feature. (You can check the logs to see if ping fails or succeeds)

jyavenard, 2020-07-25:
> 
> 
> Yes, if the device is not reachable zigbee2mqtt will mark them as unavailable when using this feature. (You can check the logs to see if ping fails or succeeds)

I never see any messages with those lights about a failure to ping them so long as they are powered (even if off).
There will be a message about the light not being reachable once the light is turned off at the wall.

However, they will always show up as offline as soon as HA is restarted.

So there's something missing in how z2m is communicating the state once it gets the hass/status via mqtt. don't know what that is though

jyavenard, 2020-07-25:
I restarted z2m to test the behaviour according to availability_timeout;

And now the lights are all reported the linkquality, and all are marked as online!
Seems that restarting zigbee2mqtt fixed the issue.

Koenkk, 2020-07-25:
Good! Are you able to reproduce the issue? If not please close.

jyavenard, 2020-07-27:
I can't reproduce now.

I think this is related to me adding those devices and then restarting HA.
A recently added device seems to not have its status properly reported when HA restart.

closed 2020-09-03 by github-actions[bot]