Koenkk/zigbee2mqtt #3970: Add support for ZTS-EUB1 moeshouse zigbee wall switch

Tontze, 2020-07-25:
I tried myself, but miserably failed :/ Also it looks like i dont know how to add code to github anymore, it looks awfull ... Im sorry.

`zigbee2mqtt:warn  2020-07-25 14:37:40: Device '0xec1bbdfffe79ddee' with Zigbee model 'TS0601' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-07-25 14:37:40: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe79ddee","supported":false}}'
`
Added this to devices.js 

`    {
        zigbeeModel: ['TS0601'], // The model ID from: Device with modelID 'lumi.sens' is not supported.
        model: 'ZTS-EUB1', // Vendor model number, look on the device for a model number
        vendor: 'MoesHouse', // Vendor of the device (only used for documentation and startup logging)
        description: 'Tuya ZigBee Push Light Switch', // Description of the device, copy from vendor site. (only used for documentation and startup logging)
        supports: 'Light on/off', // Actions this device supports (only used for documentation)
        fromZigbee: [], // We will add this later
        toZigbee: [], // Should be empty, unless device can be controlled (e.g. lights, switches).
    },
`

This was as far i managed to get, dont know what and how to parse .. : 

`zigbee2mqtt:debug 2020-07-25 15:04:01: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-25 15:04:01: No converter available for 'ZTS-EUB1' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
zigbee2mqtt:debug 2020-07-25 15:04:36: Saving state to file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:debug 2020-07-25 15:06:09: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-25 15:06:09: No converter available for 'ZTS-EUB1' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
zigbee2mqtt:debug 2020-07-25 15:08:19: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-25 15:08:19: No converter available for 'ZTS-EUB1' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
zigbee2mqtt:debug 2020-07-25 15:09:36: Saving state to file /opt/zigbee2mqtt/data/state.json
zigbee2mqtt:debug 2020-07-25 15:10:29: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
zigbee2mqtt:debug 2020-07-25 15:10:29: No converter available for 'ZTS-EUB1' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
`

Koenkk, 2020-07-25:
First can you update to zigbee2mqtt 1.14.2 and see if already works there? If not share your data/databse.db

Tontze, 2020-07-25:
Tried to update via instructions on [https://www.zigbee2mqtt.io/getting_started/running_zigbee2mqtt.html](url)

But what is "npm ci" supposed to do ?

Tontze, 2020-07-25:
After updating to zigbee2mqtt 1.14.2, it is reported as curtain motor TS0601 ? 

`zigbee2mqtt/bridge/log {"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe79ddee","model":"TS0601","vendor":"TuYa","description":"Curtain motor","supported":true}}
`

Koenkk, 2020-07-25:
Can you share your data/database.db?

Tontze, 2020-07-25:
{"id":1,"type":"Coordinator","ieeeAddr":"0x00124b0019387355","nwkAddr":0,"manufId":0,"epList":[1,2,3,4,5,6,8,11,12,13,110],"endpoints":{"1":{"profId":260,"epId":1,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"2":{"profId":257,"epId":2,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"3":{"profId":261,"epId":3,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"4":{"profId":263,"epId":4,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"5":{"profId":264,"epId":5,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"6":{"profId":265,"epId":6,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"8":{"profId":260,"epId":8,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"11":{"profId":260,"epId":11,"devId":1024,"inClusterList":[],"outClusterList":[1280],"clusters":{},"binds":[]},"12":{"profId":49246,"epId":12,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]},"13":{"profId":260,"epId":13,"devId":5,"inClusterList":[25],"outClusterList":[],"clusters":{},"binds":[]},"110":{"profId":260,"epId":110,"devId":5,"inClusterList":[],"outClusterList":[],"clusters":{},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":null}
{"id":2,"ieeeAddr":"0x00158d0002bf9b42","nwkAddr":90,"modelId":"lumi.weather","epList":[1],"endpoints":{"1":{"epId":1,"inClusterList":[],"outClusterList":[],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2483}},"msRelativeHumidity":{"attributes":{"measuredValue":3971}},"msPressureMeasurement":{"attributes":{"16":9846,"20":-1,"measuredValue":984}},"genBasic":{"attributes":{"65281":{"1":3015,"4":17320,"5":38,"6":[0,1],"10":0,"100":2473,"101":3692,"102":98476},"modelId":"lumi.weather"}}},"binds":[]}},"interviewCompleted":false,"meta":{},"lastSeen":1589749735262}
{"id":3,"type":"EndDevice","ieeeAddr":"0x00158d0002bf9b42","nwkAddr":12782,"manufId":4151,"manufName":"LUMI","powerSource":"Battery","modelId":"lumi.weather","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":24321,"inClusterList":[0,3,65535,1026,1027,1029],"outClusterList":[0,4,65535],"clusters":{"msTemperatureMeasurement":{"attributes":{"measuredValue":2273}},"msRelativeHumidity":{"attributes":{"measuredValue":4743}},"msPressureMeasurement":{"attributes":{"16":9898,"20":-1,"measuredValue":989}},"genBasic":{"attributes":{"65281":{"1":3015,"4":5032,"5":38,"6":[0,1],"10":0,"100":2274,"101":4796,"102":98899},"modelId":"lumi.weather","manufacturerName":"LUMI","powerSource":3,"zclVersion":1,"appVersion":3,"stackVersion":2,"hwVersion":30,"dateCode":"20161129","swBuildId":"3000-0001"}}},"binds":[]}},"appVersion":3,"stackVersion":2,"hwVersion":30,"dateCode":"20161129","swBuildId":"3000-0001","zclVersion":1,"interviewCompleted":true,"meta":{},"lastSeen":1595685022173}
{"id":4,"type":"Router","ieeeAddr":"0xec1bbdfffe79ddee","nwkAddr":35514,"manufId":4098,"manufName":"_TZE200_7tdtqgwv","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,10,4,5,61184],"outClusterList":[25],"clusters":{"genBasic":{"attributes":{"65503":"&0�&\u0012(0�&\u0013","modelId":"TS0601","manufacturerName":"_TZE200_7tdtqgwv","powerSource":1,"zclVersion":3,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":82,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1595689352766}

Tontze, 2020-07-26:
This is the [switch ](https://www.moeshouse.com/collections/tuya-zigbee-switch/products/tuya-zigbee-push-light-switch-zigbee-hub-required-smart-home-automation-wireless-remote-control-work-with-tuya-zigbee-hub-only)

Koenkk, 2020-07-26:
Please try with:

```js
{
    fingerprint: [{modelID: 'TS0601', manufacturerName: '_TZE200_7tdtqgwv'}],
    model: 'ZTS-EUB1 ',
    vendor: 'Moes house',
    description: '1 gang switch',
    supports: 'on/off',
    fromZigbee: [fz.tuya_switch2, fz.ignore_time_read],
    toZigbee: [tz.tuya_switch_state],
},
```

Tontze, 2020-07-26:
Better, but there is no mqtt state changes when switching ? 

zigbee2mqtt/bridge/log {"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}
zigbee2mqtt/bridge/log {"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}
zigbee2mqtt/bridge/log {"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}
zigbee2mqtt/bridge/log {"type":"device_connected","message":{"friendly_name":"0xec1bbdfffe79ddee"}}
zigbee2mqtt/bridge/log {"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}
zigbee2mqtt/bridge/log {"type":"device_announced","message":"announce","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}
zigbee2mqtt/bridge/log {"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe79ddee","model":"ZTS-EUB1 ","vendor":"Moes house","description":"1 gang switch","supported":true}}

Tontze, 2020-07-27:
Some more log after re-adding : 

`info  2020-07-27 11:51:45: Device '0xec1bbdfffe79ddee' joined
info  2020-07-27 11:51:45: Starting interview of '0xec1bbdfffe79ddee'
info  2020-07-27 11:51:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xec1bbdfffe79ddee"}}'
info  2020-07-27 11:51:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
debug 2020-07-27 11:51:45: Device '0xec1bbdfffe79ddee' announced itself
info  2020-07-27 11:51:45: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0601"}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_7tdtqgwv"}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"_TZE200_7tdtqgwv"}'
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1}'
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"appVersion":82}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":82}'
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":0}'
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
debug 2020-07-27 11:51:46: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:46: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":""}'
debug 2020-07-27 11:51:47: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:47: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{}'
info  2020-07-27 11:51:47: Successfully interviewed '0xec1bbdfffe79ddee', device has successfully been paired
info  2020-07-27 11:51:47: Device '0xec1bbdfffe79ddee' is supported, identified as: Moes house 1 gang switch (ZTS-EUB1 )
info  2020-07-27 11:51:47: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe79ddee","model":"ZTS-EUB1 ","vendor":"Moes house","description":"1 gang switch","supported":true}}'
debug 2020-07-27 11:51:52: Received Zigbee message from '0xec1bbdfffe79ddee', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:59: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:59: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
debug 2020-07-27 11:51:59: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65503":"M|�&\u0012O|�&\u0013"}' from endpoint 1 with groupID 0
debug 2020-07-27 11:51:59: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'attributeReport' and data '{"65503":"M|�&\u0012O|�&\u0013"}'
`

Koenkk, 2020-07-27:
Did some fix, please update to the latest dev branch and try again.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Tontze, 2020-07-28:
Still says no converter, but announces mqtt now, but it does not respond to mqtt state changes :) 

warn  2020-07-28 19:45:34: Device '0xec1bbdfffe79ddee' left the network
info  2020-07-28 19:45:34: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
info  2020-07-28 19:45:37: Device '0xec1bbdfffe79ddee' joined
info  2020-07-28 19:45:37: Starting interview of '0xec1bbdfffe79ddee'
info  2020-07-28 19:45:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xec1bbdfffe79ddee"}}'
info  2020-07-28 19:45:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
debug 2020-07-28 19:45:37: Device '0xec1bbdfffe79ddee' announced itself
info  2020-07-28 19:45:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0601"}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_7tdtqgwv"}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"_TZE200_7tdtqgwv"}'
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1}'
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"appVersion":82}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":82}'
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":0}'
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
debug 2020-07-28 19:45:38: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:38: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":""}'
debug 2020-07-28 19:45:39: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:39: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'readResponse' and data '{}'
info  2020-07-28 19:45:39: Successfully interviewed '0xec1bbdfffe79ddee', device has successfully been paired
info  2020-07-28 19:45:39: Device '0xec1bbdfffe79ddee' is supported, identified as: Moes house 1 gang switch (ZTS-EUB1 )
info  2020-07-28 19:45:39: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe79ddee","model":"ZTS-EUB1 ","vendor":"Moes house","description":"1 gang switch","supported":true}}'
debug 2020-07-28 19:45:44: Received Zigbee message from '0xec1bbdfffe79ddee', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:50: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:50: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
debug 2020-07-28 19:45:51: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65503":"�<�&\u0012�<�&\u0013"}' from endpoint 1 with groupID 0
debug 2020-07-28 19:45:51: No converter available for 'ZTS-EUB1 ' with cluster 'genBasic' and type 'attributeReport' and data '{"65503":"�<�&\u0012�<�&\u0013"}'
debug 2020-07-28 19:46:03: Received Zigbee message from '0xec1bbdfffe79ddee', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":63,"dp":257,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0
[... cut ...]
info  2020-07-28 19:46:03: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe79ddee/linkquality', payload '28'
debug 2020-07-28 19:46:14: Received Zigbee message from '0xec1bbdfffe79ddee', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":64,"dp":257,"fn":0,"data":{"type":"Buffer","data":[1]}}' from endpoint 1 with groupID 0
info  2020-07-28 19:46:14: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe79ddee/state', payload 'ON'
info  2020-07-28 19:46:14: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe79ddee/linkquality', payload '26'

Koenkk, 2020-07-28:
Can you provide the debug log when trying to change the state?

Tontze, 2020-07-28:
debug 2020-07-28 22:36:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":6,"dp":257,"fn":0,"data":{"type":"Buffer","data":[1]}}' from endpoint 1 with groupID 0
info  2020-07-28 22:36:16: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe79ddee/state', payload 'ON'
info  2020-07-28 22:36:16: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe79ddee/linkquality', payload '47'
debug 2020-07-28 22:36:27: Received Zigbee message from '0xec1bbdfffe79ddee', type 'commandSetDataResponse', cluster 'manuSpecificTuyaDimmer', data '{"status":0,"transid":7,"dp":257,"fn":0,"data":{"type":"Buffer","data":[0]}}' from endpoint 1 with groupID 0
info  2020-07-28 22:36:27: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe79ddee/state', payload 'OFF'
info  2020-07-28 22:36:27: MQTT publish: topic 'zigbee2mqtt/0xec1bbdfffe79ddee/linkquality', payload '44'

Koenkk, 2020-07-29:
@galzi noticed this is the same device as https://github.com/Koenkk/zigbee2mqtt/issues/3989 , so let's continue there.

closed 2020-07-29 by Koenkk