Koenkk/zigbee2mqtt #3971: Pairing Ikea switch

CHBPro, 2020-07-25:
Hi,
I am new on zigbee.
I followed everything on https://www.zigbee2mqtt.io/ to install it.
I work with a CC2531.

I paired my first Ikea E1603 switch with zigbee2mqtt and Domoticz.
I have got this log
https://hastebin.com/orayitelol.apache
( hope you can see the log this way)

It seems that the pairing went ok.
[image]

But the switch does not appear as a switch
[image]

What am I doing wrong?
Do I have to fill some extra config files?

Hope someone can help me and give me advice.

Koenkk, 2020-07-25:
Are you using https://github.com/stas-demydiuk/domoticz-zigbee2mqtt-plugin , if so it might be better to ask the question there.

Wuppi68, 2020-07-30:
same issue detected.....

Solution:
Remove the default group (901) from the bulb. After that you can set your own group ...

`bind remote to a group 
mosquitto_pub -h mqtt.heibox.intern -t zigbee2mqtt-dev/bridge/bind/Remote_OnOff_4 -m gr_Remote_OnOff_4 

Remove default group from device: 
mosquitto_pub -h mqtt.heibox.intern -t zigbee2mqtt-dev/bridge/unbind/Remote_OnOff_4 -m default_bind_group`

is a snipped from my own remarks

CHBPro, 2020-08-30:
@Wuppi68 

It has been a while, sorry.
It is my first Ikea Tradfri outlet E1603.
I cannot see in Domoticz that the outlet is connect with any groups.
Are that special groups in zigbee2mqtt or in mosquitto?
Can you give me more explanation where I can find those groups and what to do?

closed 2020-09-07 by github-actions[bot]