Koenkk/zigbee2mqtt #3972: DJT11LM vibration sensor new version?

kitopopo, 2020-07-25:
Dear @Koenkk ,

I have an old vibration sensor. Recently i bought two more sensors.
I think that this two new sensor have a difference with my old sensor. They works different.

When a vibration appears, the old sensor always detect the vibration ( action: vibration) but the new sensors not always detect the action,  almost never is detected. 

I paste the log from the three sensors:

jul 25 11:30:40 raspberrypi npm[481]: zigbee2mqtt:info 2020-07-25 11:30:40: MQTT publish: topic 'zigbee2mqtt/Sensor vibracion', payload '{"battery":100,"voltage":3045,"linkquality":118,"angle":7,"angle_x":6,"angle_y":-78,"angle_z":11,"angle_x_absolute":84,"angle_y_absolute":168,"strength":19,"action":"vibration"}'

jul 25 12:14:12 raspberrypi npm[484]: zigbee2mqtt:info 2020-07-25 12:14:12: MQTT publish: topic 'zigbee2mqtt/sensor vibracion 2', payload '{"battery":100,"voltage":3015,"linkquality":118,"angle":5,"angle_x":10,"angle_y":-79,"angle_z":5,"angle_x_absolute":80,"angle_y_absolute":169,"sensitivity":"low","strength":51}'

jul 25 12:15:41 raspberrypi npm[484]: zigbee2mqtt:info 2020-07-25 12:15:41: MQTT publish: topic 'zigbee2mqtt/sensor vibracion balcon', payload '{"battery":68,"voltage":2945,"linkquality":63,"angle":6,"angle_x":12,"angle_y":-75,"angle_z":8,"angle_x_absolute":78,"angle_y_absolute":165,"strength":225}'

The first one log is the old version, and the other ones are the news.
What is the difference? i check uptdates with the xiaomi gateway but i don't get firmware updates in any sensor. Looks like that they are the same model in state.json file

Thanks in advanced

Best regards

Koenkk, 2020-07-25:
Im not aware of a new version, do they work correctly with the original gateway?

nanobra1n, 2020-07-27:
Did you change the sensitivity setting on your old sensor to `medium` or `high`? Just saying because one of the new ones mentions `"sensitivity":"low"` and I found that it takes a herd of stampeding elephants to make the detector register anything at that settings.

kitopopo, 2020-07-27:
> Did you change the sensitivity setting on your old sensor to `medium` or `high`? Just saying because one of the new ones mentions `"sensitivity":"low"` and I found that it takes a herd of stampeding elephants to make the detector register anything at that settings.

Hi, I did some tests to adjust my awnings, but the problem occurs exactly the same in the other new sensor in which the sensitivity is not configured.

kitopopo, 2020-07-27:
> Im not aware of a new version, do they work correctly with the original gateway?

Dear @Koenkk ,

If the news sensor are paired directly to the gateway usingin the xiaomi home app, i can see in the xiaomi app all logs correctly.
 
I have the gateway paired with domoticz using internal LAN way (without zigbe2mqtt) . If i link the new sensor with the gateway, domoticz detects only 1 device with four states. If i check this log device, I can see some empty detections (without value of vibration, tilt or drop) , also i can see correct detections. The problem is the same without zigbee2mqtt.

septeven, 2020-08-30:
Hi, any update on this ? I have also noticed the same than OP

closed 2020-10-09 by github-actions[bot]