Koenkk/zigbee2mqtt #3973: Philips Hue lightstrip Plus Received message from unsupported device with Zigbee model 'LCL001'

Skopea, 2020-07-25:
I recently bought the newest model lightstrip from Philips Hue, the lightstrip plus. 

Link to the device:
https://www.bol.com/nl/p/philips-hue-lightstrip-plus-wit-gekleurd-licht-2m-basis/9300000004671570/?Referrer=ADVNLGOO002065-G-58522980718-S-477314803395-9300000004671570&gclid=CjwKCAjwsO_4BRBBEiwAyagRTRy1fl8KmcVibfSiIttg_daegyE8GccKevvTLjtI1VhFIvNgw5IGdxoC3-oQAvD_BwE

I am trying to connect to my Hassio through the zigbee2mqtt addon. Zigbee sees the device but i get an error:
The device is unsupported, so i cannot connect it to my Home assistant UI.

`zigbee2mqtt:warn  2020-07-25 19:02:57: Received message from unsupported device with Zigbee model 'LCL001'
zigbee2mqtt:warn  2020-07-25 19:02:57: Please see: https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html.`

`zigbee2mqtt:info  2020-07-25 18:56:26: MQTT publish: topic 'homeassistant/device_automation/0x001788010805b50a/action_off-press/config', payload '{"automation_type":"trigger","type":"action","subtype":"off-press","payload":"off-press","topic":"zigbee2mqtt/0x001788010805b50a/action","device":{"identifiers":["zigbee2mqtt_0x001788010805b50a"],"name":"0x001788010805b50a","sw_version":"Zigbee2mqtt 1.14.2","model":"Hue dimmer switch (324131092621)","manufacturer":"Philips"}}'
zigbee2mqtt:info  2020-07-25 18:56:26: MQTT publish: topic 'zigbee2mqtt/0x001788010805b50a/action', payload 'off-press'
zigbee2mqtt:warn  2020-07-25 18:56:37: Device '0x0017880108d533e0' left the network
zigbee2mqtt:info  2020-07-25 18:56:37: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0017880108d533e0"}}'
zigbee2mqtt:warn  2020-07-25 18:56:38: Device '0x0017880108d533e0' left the network
zigbee2mqtt:info  2020-07-25 18:56:38: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0x0017880108d533e0"}}'
zigbee2mqtt:info  2020-07-25 18:56:41: Device '0x0017880108d533e0' joined
zigbee2mqtt:info  2020-07-25 18:56:41: Starting interview of '0x0017880108d533e0'
zigbee2mqtt:info  2020-07-25 18:56:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0x0017880108d533e0"}}'
zigbee2mqtt:info  2020-07-25 18:56:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0x0017880108d533e0"}}'
zigbee2mqtt:info  2020-07-25 18:56:41: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0x0017880108d533e0"}}'
zigbee2mqtt:info  2020-07-25 18:56:44: Successfully interviewed '0x0017880108d533e0', device has successfully been paired
zigbee2mqtt:warn  2020-07-25 18:56:44: Device '0x0017880108d533e0' with Zigbee model 'LCL001' is NOT supported, please follow https://www.zigbee2mqtt.io/how_tos/how_to_support_new_devices.html
zigbee2mqtt:info  2020-07-25 18:56:44: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0x0017880108d533e0","supported":false}}'`

Can someone help me with adding this new device from the hue series to my Home assistant? Thank you in advance!

Koenkk, 2020-07-26:
Added support for this device.

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

closed 2020-07-26 by Koenkk

Skopea, 2020-07-26:
Thanks alot! @Koenkk