Koenkk/zigbee2mqtt #3978: Failing after restore a snapshot in a different SDCard

jd1900, 2020-07-26:
# Bug Report

After restoring a snapshot in a different SDCard, I can't start zigbee2mqtt.

## What happened

I get this error:
```
zigbee2mqtt:error 2020-07-26 12:48:49: Error while starting zigbee-herdsman
zigbee2mqtt:error 2020-07-26 12:48:49: Failed to start zigbee
zigbee2mqtt:error 2020-07-26 12:48:49: Exiting...
zigbee2mqtt:error 2020-07-26 12:48:49: Error: Coordinator failed to start, probably the panID is already in use, try a different panID or channel
    at /zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:171:23
    at Generator.throw (<anonymous>)
    at rejected (/zigbee2mqtt-1.14.2/node_modules/zigbee-herdsman/dist/adapter/z-stack/adapter/startZnp.js:25:65)
npm
 ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.2 start: `node index.js`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the zigbee2mqtt@1.14.2 start script.
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-07-26T10_48_49_378Z-debug.log
2020-07-26T12:48:52: PM2 log: App [npm:0] exited with code [1] via signal [SIGINT]
2020-07-26T12:48:52: PM2 log: App [npm:0] starting in -fork mode-
2020-07-26T12:48:52: PM2 log: App [npm:0] online
> zigbee2mqtt@1.14.2 start /zigbee2mqtt-1.14.2
> node index.js
```

## What did you expect to happen
Properly start.

## Debug Info
Zigbee2mqtt version: 1.14.2 
Adapter hardware: CC26X2R1

Koenkk, 2020-07-26:
Make sure to also copy `data/coordinator_backup.json`, otherwise you have to change the `pan_id` (as the existing one is already in use by your current network) which means you have to repair all your devices.

jd1900, 2020-07-26:
> Make sure to also copy `data/coordinator_backup.json`, otherwise you have to change the `pan_id` (as the existing one is already in use by your current network) which means you have to repair all your devices.

It was there.

I don't exactly know how. 
But I restored a few times, rebooted a few times, and tried unplugging and plugging again, and finally it worked.

closed 2020-07-26 by jd1900