Koenkk/zigbee2mqtt #3982: [Docker] Docker images no longer starts using Synology Docker in latest version

unholyhimura, 2020-07-26:
### Bug Report
**What happened:**
I pulled the latest images and tried to start the container. The container doesn't start and I receive the following error:

/usr/local/bin/docker-entrypoint.sh: exec: line 17: illegal option --

### What did you expect to happen
The container starts up successfully as with previous images.

### How to reproduce it (minimal and precise)
Running Docker on a Synology DSM, using the official docker images. Pull docker image, create a container with the relevant environment variables, and try and start the container.

1.14.0 version is working fine. After 1.14.1 - I'm unable to start the latest version. 
it's happening because I'm executing: "--device /dev/ttyACM0"  after the entry point command, with "run.sh" this is fine.

Looks like the entrypoint was renamed from run.sh to docker-entrypoint.sh, which is not available in Synology Docker?

Debug Info
Zigbee2mqtt version: latest and latest-dev tags

Koenkk, 2020-07-27:
Can you post your `docker run` command?

unholyhimura, 2020-07-28:
Hi @Koenkk, unfortunately, docker in Synology does require the `docker run` command, everything is done directly in the GUI.

[image]

But I've managed to export the container settings, let me know if this can give you the information you are looking for.

```
`{
   "cap_add" : null,
   "cap_drop" : null,
   "cmd" : "--device /dev/ttyACM0",
   "cpu_priority" : 50,
   "devices" : null,
   "enable_publish_all_ports" : false,
   "enable_restart_policy" : false,
   "enabled" : true,
   "entrypoint_default" : "./run.sh",
   "env_variables" : [
      {
         "key" : "PATH",
         "value" : "/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin"
      },
      {
         "key" : "NODE_VERSION",
         "value" : "12.18.0"
      },
      {
         "key" : "YARN_VERSION",
         "value" : "1.22.4"
      },
      {
         "key" : "TZ",
         "value" : "America/New_York"
      }
   ],
   "exporting" : false,
   "id" : "8913ff32e7dc572a373a0fbea6ff5b218c2a1b85321079c5f583f20017b86725",
   "image" : "koenkk/zigbee2mqtt:1.14.0",
   "is_ddsm" : false,
   "is_package" : false,
   "links" : [],
   "memory_limit" : 0,
   "name" : "koenkk-zigbee2mqtt1",
   "network" : [
      {
         "driver" : "host",
         "name" : "host"
      }
   ],
   "network_mode" : "host",
   "port_bindings" : [],
   "privileged" : true,
   "shortcut" : {
      "enable_shortcut" : false
   },
   "use_host_network" : true,
   "volume_bindings" : [
      {
         "host_volume_file" : "/docker/Zigbee2MQTT",
         "mount_point" : "/app/data",
         "type" : "rw"
      }
   ]
}
`
```

Koenkk, 2020-07-28:
Can you try changing the command to `npm start`?

unholyhimura, 2020-08-01:
Hi  @Koenkk , here is the result using `npm start`

```

date | stream | content
-- | -- | --
2020-08-01 02:25:28 | stdout | [0m
2020-08-01 02:25:28 | stdout | [0m[37;40mnpm[0m [0m[31;40mERR![0m[35m[0m     /root/.npm/_logs/2020-08-01T02_25_28_211Z-debug.log
2020-08-01 02:25:28 | stdout | [37;40mnpm[0m [0m[31;40mERR![0m[35m[0m A complete log of this run can be found in:
2020-08-01 02:25:28 | stdout | [0m
2020-08-01 02:25:28 | stdout | [0m[37;40mnpm[0m [0m[31;40mERR![0m[35m[0m This is probably not a problem with npm. There is likely additional logging output above.
2020-08-01 02:25:28 | stdout | [0m[37;40mnpm[0m [0m[31;40mERR![0m[35m[0m Failed at the zigbee2mqtt@1.14.2 start script.
2020-08-01 02:25:28 | stdout | [0m[37;40mnpm[0m [0m[31;40mERR![0m[35m[0m
2020-08-01 02:25:28 | stdout | [0m[37;40mnpm[0m [0m[31;40mERR![0m[35m[0m Exit status 1
2020-08-01 02:25:28 | stdout | [0m[37;40mnpm[0m [0m[31;40mERR![0m[35m[0m zigbee2mqtt@1.14.2 start: `node index.js`
2020-08-01 02:25:28 | stdout | [0m[37;40mnpm[0m [0m[31;40mERR![0m [0m[35merrno[0m 1
2020-08-01 02:25:28 | stdout | [37;40mnpm[0m [0m[31;40mERR![0m [0m[35mcode[0m ELIFECYCLE
2020-08-01 02:25:28 | stdout | at /app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:242:12
2020-08-01 02:25:28 | stdout | at SerialPort._error (/app/node_modules/zigbee-herdsman/node_modules/@serialport/stream/lib/index.js:198:14)
2020-08-01 02:25:28 | stdout | at SerialPort.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:133:49)
2020-08-01 02:25:28 | stdout | at __awaiter (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:23:12)
2020-08-01 02:25:28 | stdout | at new Promise (<anonymous>)
2020-08-01 02:25:28 | stdout | at /app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:27:71
2020-08-01 02:25:28 | stdout | at Generator.next (<anonymous>)
2020-08-01 02:25:28 | stdout | at Znp.<anonymous> (/app/node_modules/zigbee-herdsman/dist/adapter/z-stack/znp/znp.js:135:32)
2020-08-01 02:25:28 | stdout | [31mzigbee2mqtt:error[39m 2020-07-31 22:25:28: Error: Error while opening serialport 'Error: Error: No such file or directory, cannot open /dev/ttyACM0'
2020-08-01 02:25:28 | stdout | [31mzigbee2mqtt:error[39m 2020-07-31 22:25:28: Exiting...
2020-08-01 02:25:28 | stdout | [31mzigbee2mqtt:error[39m 2020-07-31 22:25:28: Failed to start zigbee
2020-08-01 02:25:28 | stdout | [31mzigbee2mqtt:error[39m 2020-07-31 22:25:28: Error while starting zigbee-herdsman
2020-08-01 02:25:27 | stdout | [32mzigbee2mqtt:info [39m 2020-07-31 22:25:27: Starting zigbee-herdsman...
2020-08-01 02:25:27 | stdout | [32mzigbee2mqtt:info [39m 2020-07-31 22:25:27: Starting zigbee2mqtt version 1.14.2 (commit #faaf3e4)
2020-08-01 02:25:26 | stdout | [32mzigbee2mqtt:info [39m 2020-07-31 22:25:26: Logging to console and directory: '/app/data/log/2020-07-31.22-25-26' filename: log.txt
2020-08-01 02:25:20 | stdout |  
2020-08-01 02:25:20 | stdout | > node index.js
2020-08-01 02:25:20 | stdout | > zigbee2mqtt@1.14.2 start /app
2020-08-01 02:25:20 | stdout |  
2020-08-01 02:25:17 | stdout | Using '/app/data' as data directory

```

Koenkk, 2020-08-01:
Note that the `--device` is not something from Zigbee2MQTT but from Docker itself. Zigbee2MQTT uses the `configuration.yaml` to determine the path

- Can you try with `--device /dev/ttyACM0 npm start`?
- Reading this tutorial, it seems that people are using SSH to pass a device to a container on synology: https://philhawthorne.com/installing-home-assistant-io-on-a-synology-diskstation-nas/

closed 2020-09-09 by github-actions[bot]

joshkay, 2020-09-09:
I am also having the above issue, but I am using portainer to setup the docker image.

```
> zigbee2mqtt@1.14.4 start /app
> node index.js
Zigbee2MQTT:info  2020-09-08 19:12:43: Logging to console and directory: '/app/data/log/2020-09-08.19-12-43' filename: log.txt
Zigbee2MQTT:info  2020-09-08 19:12:44: Starting Zigbee2MQTT version 1.14.4 (commit #654817a)
Zigbee2MQTT:info  2020-09-08 19:12:44: Starting zigbee-herdsman...
CREATED DECONZ ADAPTER
Zigbee2MQTT:error 2020-09-08 19:12:54: Error while starting zigbee-herdsman
Zigbee2MQTT:error 2020-09-08 19:12:54: Failed to start zigbee
Zigbee2MQTT:error 2020-09-08 19:12:54: Exiting...
Zigbee2MQTT:error 2020-09-08 19:12:54: undefined
npm ERR! code ELIFECYCLE
npm ERR! errno 1
npm ERR! zigbee2mqtt@1.14.4 start: `node index.js`
npm ERR! Exit status 1
npm ERR! This is probably not a problem with npm. There is likely additional logging output above.
npm ERR! A complete log of this run can be found in:
npm ERR!     /root/.npm/_logs/2020-09-08T23_14_21_167Z-debug.log
```

This is the other log file https://pastebin.com/raw/FkSK1cnX

CamHaug, 2020-09-20:
I too am having this issue with the latest docker file.

unholyhimura, 2020-10-17:
Hello guys, a quick update on my situation. 

Looking at your comments I decided to try portainer as well inside Synology and to my surprise it did work.

I don't need any command to identify the proper USB port. The configuration can be done directly on the configuration file.

I'm running the latest version now without any issue! 👍

Boldfor, 2023-12-31:
> I pulled the latest images and tried to start the container. The container doesn't start and I receive the following error:
> 
> /usr/local/bin/docker-entrypoint.sh: exec: line 17: illegal option --

I'm having the same problem. Does anyone know of a fix for this problem? I've been searching a lot on the web, but this post and [this one here](https://github.com/Koenkk/zigbee2mqtt/issues/3834) are among the only hits.