Koenkk/zigbee2mqtt #3983: QBKG11/12LM as light via Home Assistant discovery

pixelseventy2, 2020-07-26:
Apologies if I’ve missed this, but I’ve searched and couldn’t find anything. I have 2 wired Xiaomi QBKG12LM light switches which show up in Home Assistant as switches. Is it possible to set them to be registered as lights instead while using discovery?

Thanks.

Koenkk, 2020-07-27:
It is not possible to discover them as lights. However you can use https://www.home-assistant.io/integrations/light.switch/ to map this to a light in Home Assistant (for use in e.g. light groups).

pixelseventy2, 2020-07-28:
Can I make this a feature request then please? Tasmota, for example, handles this on a per-device basis with an option (30):

https://tasmota.github.io/docs/Commands/#setoptions

AFAIK it's just a case of sending the same discovery payload to the `homeassistant/light/` topic instead of `homeassistant/switch/`, so maybe it could be an option like `friendly_name` in the yaml file? This saves having to manually set up another entity, which slightly defeats the purpose of using discovery.

Thanks :)

closed 2020-09-05 by github-actions[bot]

b4dpxl, 2020-12-08:
This would also be useful for contact sensors, which default to `door`. It would be good to be able to change them to `window`.