Koenkk/zigbee2mqtt #3986: WSDCGQ11LM not updating when permit_join disabled

ellavas, 2020-07-27:
# Bug Report
## What happened
When the permit join parameter is set to false, the sensor is not updating. The messages are not received by the adapter. I need to reset it to get it back on the zigbee network. as soon as the parameter changes, it doesn't work anymore. 

It worked perfectly in the past.

I have tried to reset the sensor and also to change the battery, but it does not help.

The sensor works perfectly with the Aqara Gateway v3.

## Debug Info
Zigbee2mqtt version: 1.14.2
Adapter hardware: CC2531
Adapter firmware version: '{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20190608}}'

Koenkk, 2020-07-27:
To debug this I think we need to sniff the traffic of the working vs non-working situation when the sensor is triggered. You can follow: https://www.zigbee2mqtt.io/how_tos/how_to_sniff_zigbee_traffic.html

ellavas, 2020-07-28:
Ok I'll make some capture. Do you recommend me to use a second coordinator to sniff the traffic ? (

Koenkk, 2020-07-28:
Yes you need a second stick.

But first check if your network contains any of the following non-compatible Xiaomi routers: https://community.hubitat.com/t/xiaomi-aqara-devices-pairing-keeping-them-connected/623 (Keeping Xiaomi / Aqara Devices Connected)

ellavas, 2020-07-29:
Thanks, I have Legrand routers (https://www.zigbee2mqtt.io/devices/067776.html) in my network. There are not present in this list but it could be a clue.

closed 2020-09-05 by github-actions[bot]