Koenkk/zigbee2mqtt #3987: Zigbee2mqtt stops working after power on of a GLEDOPTO led strip controller

Claude2666, 2020-07-27:
# Bug Report
## What happened
Zigbee2mqtt stops working after power on of GLEDOPTO led controller
## What did you expect to happena
Zigbee2mqtt continues working
## How to reproduce it (minimal and precise)
With a GLEDOPTO led strip controller such as a GL-C-007-1ID,  GL-C-008-1ID:

- unplug the led string for a few seconds
- plug it back in

Zigbee2mqtt throws an error:

Jul 14 01:36:57 Zigbee2MQTT npm[26842]: (node:26854) UnhandledPromiseRejectionWarning: RangeError: Maximum call stack size exceeded
Jul 14 01:36:57 Zigbee2MQTT npm[26842]:     at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1527:20)
Jul 14 01:36:57 Zigbee2MQTT npm[26842]:     at Object.convertGet (/opt/zigbee2mqtt/node_modules/zigbee-herdsman-converters/converters/toZigbee.js:1527:68)
Jul 14 01:36:57 Zigbee2MQTT npm[26842]: (node:26854) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of
Jul 14 01:36:57 Zigbee2MQTT npm[26842]: (node:26854) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that

From that moment on, no more info is written in the log.txt  but zigbee2mqtt continues working for several hours or even days until it goes offline completely

I think this started happening after I updated to dev version 0.14.x  3-4 weeks ago. On 0.13 it did not happen.
I have the availability timeout feature enabled because the GLEDOPTO LED strip controllers are always on after a power return. The idea is when they become available again, to switch them off with an automation in Home Assistant.

This problem happens also if due to poor network or WiFi interference, the GLEDOPTO  looses connection to the zigbee network and after it re-connects, the same error occurs.

--
[zigbee2mqtt-log.txt](https://github.com/Koenkk/zigbee2mqtt/files/4980734/zigbee2mqtt-log.txt)

--

## Debug Info
Zigbee2mqtt version:  1.14.2-dev (commit #00f836f)
Adapter hardware: CC2530 + CC2591
Adapter firmware version: 20190619

Settings in configuration.yaml

....
advanced:
  availability_timeout: 30
  baudrate: 115200
  rtscts: false
  cache_state: true
  log_level: debug
queue:
  delay: 250
  simultaneously: 5

Attached a logfile with debug and herdsman debug enabled.

Koenkk, 2020-07-27:
Should be fixed, can you confirm?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

Claude2666, 2020-07-28:
I can confirm that with the latest dev version, the problem is solved. 

Thank you very much for this amazing project.

closed 2020-07-28 by Claude2666