Koenkk/zigbee2mqtt #3989: Wrong identification of device

galzil, 2020-07-27:
[TuYa GDKES-01TZXD](https://www.zigbee2mqtt.io/devices/GDKES-01TZXD.html) light switch is identified as: TuYa Curtain motor (TS0601)
I'll appreciate help with this

galzil, 2020-07-27:
I've removed both TS0601 items from devices.js
Curtain motor
TS0601_thermostat

there's a Lonsonho switch there, but it's not recognized
` {
        zigbeeModel: ['TS0011'],
        model: 'GDKES-01TZXD',
        vendor: 'TuYa',
        description: 'Smart light switch - 1 gang without neutral wire',
        supports: 'on/off',
        fromZigbee: [fz.on_off],
        toZigbee: [tz.on_off],
    },`

I've added
`{
		zigbeeModel: ['TS0601'],
		model: 'ZSQB-SFB-11ZG',
        vendor: 'Tuya',
        description: '1 gang one way switch no neutral',
        supports: 'on/off',
        fromZigbee: [fz.tuya_switch2, fz.ignore_time_read],
        toZigbee: [tz.tuya_switch_state],
    },`

which solves the problem locally

galzil, 2020-07-27:
Maybe it's a newer version,
Had to add to the convertors as well

`tuya_switch3: {
        cluster: 'manuSpecificTuyaDimmer',
        type: ['commandSetDataResponse', 'commandGetData'],
        convert: (model, msg, publish, options, meta) => {
            const multiEndpoint = model.meta && model.meta.multiEndpoint;
            const dp = msg.data.dp;
            const state = msg.data.data[0] ? 'ON' : 'OFF';
            if (multiEndpoint) {
                const lookup = {257: 'l1', 258: 'l2', 259: 'l3'};
                const endpoint = lookup[dp];
                if (endpoint in model.endpoint(msg.device)) {
                    return {[`state_${endpoint}`]: state};
                }
            } else if (dp === 257) {
                return {state: state};
            }
            return null;
        },
    },`

Koenkk, 2020-07-27:
Can you share your `data/database.db` entry of this device?

galzil, 2020-07-28:
@Koenkk this is the line:
`{"id":3,"type":"Router","ieeeAddr":"0x842e14fffe13ceae","nwkAddr":5929,"manufId":4098,"manufName":"_TZE200_7tdtqgwv","powerSource":"Mains (single phase)","modelId":"TS0601","epList":[1],"endpoints":{"1":{"profId":260,"epId":1,"devId":81,"inClusterList":[0,4,5,61184],"outClusterList":[25,10],"clusters":{"genBasic":{"attributes":{"modelId":"TS0601","manufacturerName":"_TZE200_7tdtqgwv","powerSource":1,"zclVersion":3,"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":""}}},"binds":[]}},"appVersion":64,"stackVersion":0,"hwVersion":1,"dateCode":"","zclVersion":3,"interviewCompleted":true,"meta":{},"lastSeen":1595912320360}`

Koenkk, 2020-07-28:
Can you check if it works now?

Changes will be available in the latest dev branch in a few hours (https://www.zigbee2mqtt.io/how_tos/how-to-switch-to-dev-branch.html)

galzil, 2020-07-28:
I will check it soon. 
Please note, I'm not sure it's the same as the other switch. It looks the same but there's also a small difference in the converter

galzil, 2020-07-28:
This issue seems like it's the same device [ZTS-EUB1 moeshouse zigbee wall switch](https://github.com/Koenkk/zigbee2mqtt/issues/3970),
I was wrong referncing Lonsonho X711A,
I updates: tuya_switch2 to
`tuya_switch3: { cluster: 'manuSpecificTuyaDimmer', type: ['commandSetDataResponse', 'commandGetData'], convert: (model, msg, publish, options, meta) => { const multiEndpoint = model.meta && model.meta.multiEndpoint; const dp = msg.data.dp; const state = msg.data.data[0] ? 'ON' : 'OFF'; if (multiEndpoint) { const lookup = {257: 'l1', 258: 'l2', 259: 'l3'}; const endpoint = lookup[dp]; if (endpoint in model.endpoint(msg.device)) { return {[state_${endpoint}]: state}; } } else if (dp === 257) { return {state: state}; } return null; }, },`

Koenkk, 2020-07-29:
What is the difference between tuya_switch3 ad tuya_switch2? I don't see it.

galzil, 2020-07-29:
@Koenkk it's a very small difference, I added "commandSetDataResponse" which was missing for the status responce
`No converter available for 'ZSQB-SFB-11ZG' with cluster 'manuSpecificTuyaDimmer' and type 'commandSetDataResponse' and data '{"status":0,"transid":107,"dp":257,"fn":0,"data":{"type":"Buffer","data":[1]}}'`
`type: ['commandSetDataResponse', 'commandGetData'],`

Koenkk, 2020-07-29:
I've already added that a few days ago (latest dev branch): https://github.com/Koenkk/zigbee-herdsman-converters/blob/259acf98cb8f0f0254c4a7aadc0d5b2347cd9667/converters/fromZigbee.js#L5114 , so that looks good!

Please let me know if it works out-of-the-box in the latest dev branch.

Tontze, 2020-07-29:
@galzil Does your switch start pairing procedure atomatically (fast blinking) when you power it up ? Does it also have beep sound when operated ? Just want to make sure we have same switch :)

galzil, 2020-07-29:
@Tontze Yes, it's the same one. 
It's blinking on powering and starts pairing. 
I got it to work with the changes I made. 
I will test the dev branch later when I get back home. It should work. 
I didn't know about the beep sound when I bought it, the seller told me it can't be muted, very annoying.

Tontze, 2020-07-29:
@galzil yeah, beeping is very annoying. I have lonsonho also, its so much better. No beeping and much softer click too.

galzil, 2020-07-29:
@Koenkk I've just tested the dev branch, the switch was paired out-of-the-box.
`Device '0x842e14fffe13ceae' is supported, identified as: Lonsonho 1 gang switch (X711A)`
Thank you so much for your work and quick responce

Tontze, 2020-07-29:
Im having hard time to get it ti work, i see it pairing, and supported (X711A), but see also no converter found messages, do they matter ? It announces to mqtt, but doesnt respond to mqtt state on/off ?

galzil, 2020-07-29:
@Tontze Did you update to the newest dev version?
You need to unpair and pair again (5 short presses and than 1 long press until it blinks)

Tontze, 2020-07-30:
> 
> 
> @Tontze Did you update to the newest dev version?
> You need to unpair and pair again (5 short presses and than 1 long press until it blinks).
> I'm new to this too, learning as I go

i did yes

galzil, 2020-07-30:
Very weired.
I just had to re-flash my cc2531 stick as it stopped working for some reason.
I re-installed the dev branch (as I went back to the master) and re-paired my devices, it was paired succesfully and works well.
Maybe try editing the FromZigbee.js yourself?
you need to change only one line, look for tuya_switch2
edit
`type:  'commandGetData',`
to
`type: ['commandSetDataResponse', 'commandGetData'],`

Tontze, 2020-07-30:
As i am in dev branch, i have that edit allready.. This is what i see after resetting the device : 

warn  2020-07-30 17:17:12: Device '0xec1bbdfffe79ddee' left the network
info  2020-07-30 17:17:12: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_removed","message":"left_network","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
info  2020-07-30 17:17:15: Device '0xec1bbdfffe79ddee' joined
info  2020-07-30 17:17:15: Starting interview of '0xec1bbdfffe79ddee'
info  2020-07-30 17:17:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_connected","message":{"friendly_name":"0xec1bbdfffe79ddee"}}'
info  2020-07-30 17:17:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_started","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
debug 2020-07-30 17:17:15: Device '0xec1bbdfffe79ddee' announced itself
info  2020-07-30 17:17:15: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"device_announced","message":"announce","meta":{"friendly_name":"0xec1bbdfffe79ddee"}}'
debug 2020-07-30 17:17:15: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"modelId":"TS0601"}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"manufacturerName":"_TZE200_7tdtqgwv"}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{"manufacturerName":"_TZE200_7tdtqgwv"}'
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"powerSource":1}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{"powerSource":1}'
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"zclVersion":3}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{"zclVersion":3}'
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"appVersion":82}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{"appVersion":82}'
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"stackVersion":0}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{"stackVersion":0}'
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"hwVersion":1}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{"hwVersion":1}'
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{"dateCode":""}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{"dateCode":""}'
debug 2020-07-30 17:17:16: Received Zigbee message from '0xec1bbdfffe79ddee', type 'readResponse', cluster 'genBasic', data '{}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:16: No converter available for 'X711A' with cluster 'genBasic' and type 'readResponse' and data '{}'
info  2020-07-30 17:17:16: Successfully interviewed '0xec1bbdfffe79ddee', device has successfully been paired
info  2020-07-30 17:17:16: Device '0xec1bbdfffe79ddee' is supported, identified as: Lonsonho 1 gang switch (X711A)
info  2020-07-30 17:17:16: MQTT publish: topic 'zigbee2mqtt/bridge/log', payload '{"type":"pairing","message":"interview_successful","meta":{"friendly_name":"0xec1bbdfffe79ddee","model":"X711A","vendor":"Lonsonho","description":"1 gang switch","supported":true}}'
debug 2020-07-30 17:17:21: Received Zigbee message from '0xec1bbdfffe79ddee', type 'read', cluster 'genTime', data '["localTime"]' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:26: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65506":21,"appVersion":82}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:26: No converter available for 'X711A' with cluster 'genBasic' and type 'attributeReport' and data '{"65506":21,"appVersion":82}'
debug 2020-07-30 17:17:26: Received Zigbee message from '0xec1bbdfffe79ddee', type 'attributeReport', cluster 'genBasic', data '{"65503":"\u0015��&\u0012\u0017��&\u0013"}' from endpoint 1 with groupID 0
debug 2020-07-30 17:17:26: No converter available for 'X711A' with cluster 'genBasic' and type 'attributeReport' and data '{"65503":"\u0015��&\u0012\u0017��&\u0013"}'

galzil, 2020-07-31:
@Tontze just to let you know, I just found out there's a way to turn the beep off. 
Click 4 times, keep pressing on the forth click. There will be a beep after a few seconds, keep holding until you hear 2 beeps. That's it. No more beeping!

Tontze, 2020-07-31:
@galzil awesome, thnx for sharing!

Koenkk, 2020-07-31:
Does it work as expected for you both now?

Tontze, 2020-07-31:
Cant get it working :/ Log is few post up

galzil, 2020-08-01:
Mine's working. 
The re-flashing was due to something else (not sure what).

Tontze, 2020-08-08:
@galzil do you see those same "no convertor" messages as i do ? Im starting to think that i have received defective switch, i cannot get it to respond mqtt payloads :|

galzil, 2020-08-09:
@Tontze sorry, it's working fine for me, no errors. 
Try ordering [X711A](https://a.aliexpress.com/_d7huMfK), 
although it's an End Device and not a Router like this one

Tontze, 2020-08-09:
I have three X711A, they do work nicely :) one without neutral and 2 with neutral. Neutral ones are not installed yet tho. I think the ones with neutral wire do work as router ? Ones without obviously dont :)

Tontze, 2020-08-09:
btw, is yours moeshouse switch with, or without neutral ?

galzil, 2020-08-09:
Currently all my zigbee devices are without neutral.
I actually think my X711A and X712A are recognized wrongly. I'll look for a similar issue as they should be Routers

Tontze, 2020-08-09:
Ok, now, my moes switch is with neutral, so after all, it is different switch than yours.

closed 2020-09-16 by github-actions[bot]