Koenkk/zigbee2mqtt #3990: E1743 now also returns action_group and action

sjorge, 2020-07-27:
# Bug Report
## What happened
```
zigbee2mqtt:info  2020-07-27 13:59:21: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"linkquality":47,"update_available":false,"battery":60,"action":"off","action_group":10112,"click":"off"}'
```

Looks like the E1743 is now also returning an action and action_group, this was introduce by the change for some remote that seem to have multiple groups. However for this device this doesn't make any sense and 'click' should be sufficient.

## What did you expect to happen
Not have action_group/action for remotes that only support one group.

## How to reproduce it (minimal and precise)
Pair a E1743 remote and click a button

## Debug Info
Zigbee2mqtt version: 1.14.2-dev
Adapter hardware: CC2531
Adapter firmware version: `{"type":"zStack12","meta":{"transportrev":2,"product":0,"majorrel":2,"minorrel":6,"maintrel":3,"revision":20200720}},"log_level":"info","permit_join":true}`

Koenkk, 2020-07-27:
Should be fixed in latest dev branch

sjorge, 2020-07-27:
Maybe I was too quick? I just updated but it's still included
```
zigbee2mqtt:info  2020-07-27 20:52:28: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"linkquality":31,"update_available":false,"battery":60,"action":"off","action_group":10112,"click":"off"}'
``` ?

Koenkk, 2020-07-27:
Yes too quick, just merged https://github.com/Koenkk/zigbee2mqtt/pull/3994

sjorge, 2020-07-27:
Ha I figured, I was just winding down for the day after pairing most of my new lamps. And saw the mail and hit update. The noticed the mail was less than 5min old.

sjorge, 2020-07-27:
All good now, thanks!
```
zigbee2mqtt:info  2020-07-27 21:01:12: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"linkquality":60,"update_available":false,"battery":74,"action":"on","click":"on"}'
```

sjorge, 2020-08-20:
While debugging some other stuff, I noticed that although action_group is now gone.
It still publishes duplicate data... both click and action are set to either on or off

```
Zigbee2MQTT:info  2020-08-20 14:32:39: MQTT publish: topic 'zigbee2mqtt/office/desk_light/remote', payload '{"action":"off","battery":87,"click":"off","linkquality":47,"update_available":false}'
```

We should probably deprecate 'click' and remove it in the future as it has been in a few releases now with both.

Koenkk, 2020-08-20:
It's already: https://www.zigbee2mqtt.io/devices/E1743.html#deprecated-click-event

closed 2020-08-20 by sjorge